kriswiner/MPU9250 #80: Stupid mag calibration question

frdfsnlght, 2016-10-24:
I've repeatedly read about calibrating the magnetometer here and in all cases I'm supposed to move the device in a "figure 8 pattern". What does that mean? In my mind, a figure 8 is a 2 dimensional shape, so what plane should the movement be in? I somehow doubt that's what meant though. Is there a video of someone moving the sensor for this calibration so I can see what's supposed to be happening?

kriswiner, 2016-11-23:
See:

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration

frdfsnlght, 2016-11-26:
I read that. It helped a bit, but didn't answer my original question. I've since found some videos like [this one](https://www.youtube.com/watch?v=3XVmdCtJO10) that illustrates a "figure 8 pattern".