kriswiner/MPU9250 #83: Possible error in setting USER_CTRL in accelgyrocalMPU9250

gjwo, 2016-11-01:
in MPU9250_MS5637_AHRS_t3.ino line 939 the register value doesn't look right as per the comment.
you have  "writeByte(MPU9250_ADDRESS, USER_CTRL, 0x0C);    // Reset FIFO and DMP"

Comparing this to the DMP-9250 register map (v1.4) table 1 p9
6A 106 USER_CTRL R/W - FIFO_EN I2C_MST_EN I2C_IF_DIS - **FIFO_RST** **I2C_MST_RST** SIG_COND_RST

bit2 and bit 1 seem to the the relevant ones not bit3 and bit2, so I think the value should be 0x06 not 0x0C

... I am still trying to get my java version working so I am going through line by line!

gjwo, 2016-11-01:
given the comment in the manual it might be better to leave I2C_MST_RST alone!
"1 – Reset I2C Master module. Reset is asynchronous. This bit auto clears after one clock cycle.
NOTE: This bit should only be set when the I2C master has hung. If this bit is set during an active I2C master transaction, the I2C slave will hang, which will require the host to reset the slave."

So a value of 0x04 might be better - This is effectively what you are doing since bit3 is reserved, but your experience of this device is much better than mine, so I will see what you recommend

kriswiner, 2016-11-01:
For USER_CNTL, 0x0C resets the DMP (bit 3, marked reserved in the data sheet) and the FIFO (bit 2). Resetting the DMP isn't really necessary since it is not being used here.

gjwo, 2016-11-01:
Thanks for the info. So where can I find a more up to date guide to the registers?

kriswiner, 2016-11-01:
The register map is available on the Invensense site.

They won't list the DMP registers though.

accelgyrocalMPU9250 (#83)

<https://github.com/kriswiner/MPU-9250/issues/83#issuecomment-257680909> ,
zn3tyx-ks5q55zdgaJpZM4KmIdQ> .
<https://github.com/notifications/beacon/AGY1qjBO32SgIMVF7WfM-H3YsUG6I0JFks5
q55zdgaJpZM4KmIdQ.gif>

gjwo, 2016-11-02:
Thanks Kris, that document is slightly more up to date than the one I had, but still has pin 3 as reserved.
Do you happen to know why they don't publish DMP info, or who does?

kriswiner, 2016-11-02:
Invensense doesn't publish it since, I assume, it is proprietary. You can
look at their API where they write to these hidden registers and figure out
what most of them are.

accelgyrocalMPU9250 (#83)

<https://github.com/kriswiner/MPU-9250/issues/83#issuecomment-257810198> ,
eMSy17yks5q6FQagaJpZM4KmIdQ> .
<https://github.com/notifications/beacon/AGY1qqG2zIUVQ_vhZNnypsM5dTWlMZ0Cks5
q6FQagaJpZM4KmIdQ.gif>

closed 2016-11-10 by gjwo