kriswiner/MPU9250 #87: Need your advice for using of this lib...

KhArtNJava, 2016-11-16:
Hello.

Can you help me with adaptation of https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS.ino for ESP-8266? )

I've connected 

```
MPU-9250                            ESP-12F
SDA                                      GPIO0
SCL                                       GPIO2
GND                                      GND
VCC                                      VCC +3 V
```

And changed code to 

```
***
// Pin definitions
int intPin = 12; //12  // These can be changed, 2 and 3 are the Arduinos ext int pins
int myLed = 16; //D0 // Set up pin 13 led for toggling
***
  Wire.begin(0, 2);
 // Wire.setClock(400000);
***
```

Here is pinout of ESP-12F:
http://tech.scargill.net/wp-content/uploads/2015/09/ESP-12F.jpg

After start, I got output like

```
Scanning...
I2C device found at address 0x68  !
done

MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF
```

I can't understand, what is intPin in your code?
I configured it 
```
// Pin definitions
int intPin = 12;
```

Ie, it's GPIO12. How should I connect GPIO12 port? To the INT port of the MPU-9250?

My code is:

```
/* MPU9250 Basic Example Code
  by: Kris Winer
  date: April 1, 2014
  license: Beerware - Use this code however you'd like. If you
  find it useful you can buy me a beer some time.

  Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor,
  getting properly scaled accelerometer, gyroscope, and magnetometer data out. Added display functions to
  allow display to on breadboard monitor. Addition of 9 DoF sensor fusion using open source Madgwick and
  Mahony filter algorithms. Sketch runs on the 3.3 V 8 MHz Pro Mini and the Teensy 3.1.

  SDA and SCL should have external pull-up resistors (to 3.3V).
  10k resistors are on the EMSENSR-9250 breakout board.

  Hardware setup:
  MPU9250 Breakout --------- Arduino
  VDD ---------------------- 3.3V
[... 1036 lines cut ...]
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address < 16)
       Serial.print("0");
    Serial.print(address, HEX);
      Serial.println("  !");
      nDevices++;
    }
    else if (error == 4)
    {
      Serial.print("Unknow error at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.println(address, HEX);
    }
  }

  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

}
```

kriswiner, 2016-11-17:
I wouldn't use this sketch, try looking here:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

KhArtNJava, 2016-11-17:
Thank you very much!
Seems, like they are very similar.
Have you some code, to read DMP data from MPU-9250 with ESP-8266?
Like in https://github.com/rpicopter/ArduinoMotionSensorExample/blob/master/AvrCopter.ino .

KhArtNJava, 2016-11-17:
I can't understand why you use SPI.h in your code https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_BasicAHRS2_ESP8266.ino#L35 , whereas MPU sensors connected to ESP with I2C ( https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_BasicAHRS2_ESP8266.ino#L21 ) ?

kriswiner, 2016-11-17:
For the display.

(#87)

https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_Basi
cAHRS2_ESP8266.ino#L35 , whereas MPU sensors connected to ESP with I2C (
https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_Basi
cAHRS2_ESP8266.ino#L21 ) ?

-
<https://github.com/kriswiner/MPU-9250/issues/87#issuecomment-261181147> ,
yE2CS7hks5q_AsTgaJpZM4KzvhR> .
<https://github.com/notifications/beacon/AGY1qm3ioaQ7PC5M4WxGb2_w-HA8ayBPks5
q_AsTgaJpZM4KzvhR.gif>

kriswiner, 2016-11-17:
I don't use the DMP.

(#87)

https://github.com/rpicopter/ArduinoMotionSensorExample/blob/master/AvrCopte
r.ino .

-
<https://github.com/kriswiner/MPU-9250/issues/87#issuecomment-261179347> ,
1wF_0CVks5q_AiagaJpZM4KzvhR> .
<https://github.com/notifications/beacon/AGY1qnWuhpjBEm-ymWos3kWOmC_1PslBks5
q_AiagaJpZM4KzvhR.gif>

KhArtNJava, 2016-11-18:
Kris, I am sorry for my silly question, but why you don't use DMP? 
As I understand (but I can be wrong), DMP makes some calc-s, and it's can reduce calc-s on the receiver side (ie, on ESP-8266).
What's bad in DMP? Is it slower, than direct values reading from Magnetometer, Gyro and Acc-r?

kriswiner, 2016-11-18:
Hi Arthur,

The DMP is difficult to use well, poorly documented, is limited to a fixed
100 Hz fusion rate, does not use the magnetometer in the fusion (limited to
6 DoF) so the heading drifts at 1 or 2 degrees per minute. Really not a good
solution.

I like the MPU6500 and AK8963C sensors and use them with this, which is
easier to use, much more accurate, can run the fusion rate at the rate of
the gyro up to 400 Hz, and provides 2 degree heading accuracy with no drift.

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/

Kris

(#87)

<https://github.com/kriswiner/MPU-9250/issues/87#issuecomment-261468488> ,
B2Enkj3ks5q_Vb9gaJpZM4KzvhR> .
<https://github.com/notifications/beacon/AGY1qsa-13QKUsOcarenyibJ8yvz1u0Wks5
q_Vb9gaJpZM4KzvhR.gif>

KhArtNJava, 2016-11-18:
Kris, what is price of the each chip EM7180? Without any board, just a chip.
And what is price of 100pcs of EM7180?

kriswiner, 2016-11-18:
The EM7180 are $1.50 each.

(#87)

<https://github.com/kriswiner/MPU-9250/issues/87#issuecomment-261607001> ,
8OuaLnqks5q_fBJgaJpZM4KzvhR> .
<https://github.com/notifications/beacon/AGY1qhXfsxMUGcYjc6UEiS_2nnPD528_ks5
q_fBJgaJpZM4KzvhR.gif>

KhArtNJava, 2016-11-18:
I am sorry, Kris... 
Very sorry...
But why so expensive board costs?
I think, it should be $10-12 per board.
OK, + $3 your profit +about $5 for shipping.
Ie, $18-20 per board.
I think, it's possible to make it for $15, if to make it in China...
May be I am wrong, it's just rough calcs...
In any case, $36 per board - it's very expensive.
Can you lower price? I am from Russia, where Roubles fell twice in last 2 years and $36 is approximate 1/18 of my monthly salary :)...

kriswiner, 2016-11-18:
Sorry you cannot afford it; I put a lot of work into designing and making
the product useful. Maybe you can use open source fusion and your MPU9250
instead...

(#87)

<https://github.com/kriswiner/MPU-9250/issues/87#issuecomment-261626084> ,
N-MX9kJks5q_gI7gaJpZM4KzvhR> .
<https://github.com/notifications/beacon/AGY1qn36a2bJ1M7QBaauvEVPXIBvxlBQks5
q_gI7gaJpZM4KzvhR.gif>

KhArtNJava, 2016-11-18:
I respect your position, but it's very pity for me :).

KhArtNJava, 2016-11-18:
I've sent you email with some questions and I will be thankful to your answer by email :) .

KhArtNJava, 2016-11-18:
Cleaned some my posts.

Desoky01, 2019-10-12:
@KhArtNJava  
Did you finish the adaptation of https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS.ino for ESP-8266? 

I want to use the code of @kriswiner  for mpu9250 with ESP8266 Nodemcu v3.

Desoky01, 2019-10-12:
@KhArtNJava  Did you finish the adaptation of @kriswiner  mpu9250 library to esp8266?

@kriswiner  I want to use  mpu9250 library to esp8266 nodemcu V3.

kriswiner, 2019-10-12:
OK, you have my permission to do so.

You could try this:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

If it doesn't work, find a way to modify it so it does.

Desoky01, 2019-10-12:
.

catchrow1, 2019-11-16:
Hi, 
Another newby- thanks forthe code!  
I've got it up and running, but the Nokia 5110 screen does not display- what am i missing please?

kriswiner, 2019-11-16:
Hard to say without more info, have you ever gotten data to display using
any sensor on the Nokia 5110?

catchrow1, 2019-11-17:
A bit of research found me the answer: had to add  ' #define LCD'  to the code! seems to be working reasonably well now.

Thanks for getting back to me.
Cheers,
R