kriswiner/MPU9250 #92: Control multiple MPU9250 to start collecting data at the same time

Hysenzhao, 2016-11-28:
Everyone,
Is it  possible to use multiple MPU9250s for data acquisition at the same time?
I want to use 10 MPU9250s in human motion capture, but how make all MPU9250 work at the same time so that make the data collected meaningful. The MCU what i use is RPi Zero, and transfer data using one I2C.

---Hysen

alexozer, 2016-12-25:
Right now it's only really possible to use one I believe; adding support for multiple sensors would at least require implementing multiple interrupt functions, but this is difficult because standard interrupt handlers only accept standalone or static functions and not C++ methods.

Defragster, 2016-12-25:
It would take a unique pin to catch each interrupt - and that would trigger a unique function.

But with 10 active devices the time it would take to poll all 10 would probably not be too fast to start over - unless  they were grouped onto unique i2c bus subsets.