kriswiner/MPU9250 #93: Is the singularity going from Quaternion to Tait-Bryan angles dealt with?

gjwo, 2016-11-30:
Kris,
in you code around line 625, do you deal with the singularity at +-90% pitch as described in your reference here [http://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles](url)? if not dealt with this could explain the glitches I am getting, I am guessing this isn't normally an issue with flying model planes or quadcopters as they shouldn't be pointing straight up or down! I am trying to get a cube to mimic the orientation of the sensor, so this is an issue for me.

gjwo, 2016-12-21:
and issue #45

UnTraDe, 2017-01-10:
I experience the exact same problem using the latest code for Arduino with the MPU-9250. I tried looking both directly at the values that the Arduino code outputs (yaw, pitch and roll) and using the quaternion directly to rotate a 3D cube in OpenGL. In both cases when Pitch gets to around -90/90 degrees the Yaw flips. Have you managed to find any solution to this issue?

gjwo, 2017-01-11:
@UnTraDe  thanks for commenting, I was beginning to think it was something I had done in translatng the code to java, but if you are getting the same to the original code it can't be that. If you are driving a cube directly off the quaternion, then it  has to be  the inputs to the caculation or the calculation itself not the Tait-Bryan conversion. Can you post a plot of yaw pitch and roll against time when doing a 360 degree rotation on pitch, to compare with mine?

UnTraDe, 2017-01-11:
I recorded 3 graphs, each with a 360 degrees rotation around different axis:
Sorry for the graphs quality didn't have enough time to fine tune everything...

graph Y axis is degress
graph X axis is sample count (I send sample over the com port each 20 ms (50 Hz) so the time is in units of 20 ms)

BLUE - pitch
GREEN - yaw
RED - roll

Pitch:
[image]

Yaw:
[image]

Roll:
[image]

gjwo, 2017-01-11:
@UnTraDe  see my graphs on #90 (near the  bottom), I would have expected yaw rotate and roll rotate to show a one sawtooth and two stable lines,  with the pitch one one showing a sawtooth with a  problem on  yaw. So your yaw rotation above looks a bit odd. I got a clean one by rotating on  the desk.

UnTraDe, 2017-01-11:
@gjwo Oh I see. You're graph also using 0 to 360 degrees and mine -180 to 180. But I think there is still something wrong with mine, let me try a few things and I'll post new graphs.

gjwo, 2017-01-11:
@UnTraDe  I have 360 for yaw because  I also have a compass display.

UnTraDe, 2017-01-12:
Better graphs:

BLUE - pitch
GREEN - yaw
RED - roll

Here you can see I got the exact same result as yours

Pitch:
[image]

Yaw:
[image]

Roll:
[image]

kriswiner, 2017-01-12:
If you want to avoid these "discontinuities you need to manage the
transitions in software. The yaw goes from -180 to +180, so if you want 0
to 360 like a compass with a smooth transition you have to construct this
in the sketch. This is what I do. These curves all look normal to me.

UnTraDe, 2017-01-12:
@kriswiner The Yaw and Roll graphs are ok, they are very precise and responsive. The problem is with the Pitch (the first graph). The pitch angle (the blue line) rotatets between +90 to -90, while the in the others (yaw, roll) rotates from +180 to -180 (as seen in the second and third graphs. And it also changes direction instead of rolling around like yaw and roll do.

But the bigger problem is that the pitch rotation affects yaw and roll heavily (like the example in the first graph).

Is there any reason for this?

kriswiner, 2017-01-12:
Yes, you will see this kind of cross talk when you are not feeding the
Madgwick filter the sensor data properly. It expects the data in a NED
format.

As far as pitch going from -90 to +90, this is what the asin does. If you
want some other values you will have to manage this in the sketch. For
example, if you wan it to go from 0 to 180 instead, you will have to add
the 90 degree delta appropriately. But it can only span 180 degrees.

gjwo, 2017-01-12:
@UnTraDe  am I right in thinking you are using Kris's sketch?

UnTraDe, 2017-01-13:
@gjwo Yes, I'm using the latest code from the master branch

gjwo, 2017-01-13:
@kriswiner  Kris, does this mean your code is not feeding the Madgeick filter the sensor data properly?

kriswiner, 2017-01-13:
Depends on which sketch, but in general I wouldn't assume so. The sensors
values have to passed into the Madgwick or Mahony function in a NED
configuration, this will change depending on the delais of your sensor
breakout board and specific application.