kriswiner/MPU9250 #95: Cant get it to work

AkshayGupta94, 2016-12-05:
Quite new to arduino , please excuse my lack of knowledge on certain subjects!!

I recently purchased an IMU, namely the MPU-9250 breakout board from SparkFun. Data sheet and other specs can be found here:

https://learn.sparkfun.com/tutorials/mpu-9250-hookup-guide

All I want to do is obtain the raw data from accelerometer, gyroscope and magnetometer from the sensor so I can record it/process it/etc in matlab.... However I am having a great deal of trouble knowing how to configure the code.. I tried using the arduino library by Kris Winer, but every time I try and run the code I get this error:

Code: [Select]

MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF

I also tried running the code on this site here:
http://www.lucidarme.me/?p=5057

and with that, I am getting 7 columns of data on the serial monitor, however, none of them change except the iteration column (the first column)... 

If someone could please help me, I would greatly appreciate it. Thank you so much.

kriswiner, 2016-12-05:
Do you have 3v3 showing on sda and scl?
 What addresses show in an i2c scan?
Try using 0x69 for the mpu9250 address. How are you connecting to the mcu?

AkshayGupta94, 2016-12-05:
Thanks for prompt reply
I tried your i2c scanning code as give here https://github.com/kriswiner/MPU-9250/issues/37
it shows no i2c device connected.
I checked sda and scl on chip shows 3 v when connected to voltmeter.. both show 3v individually when other end of voltmeter is grounded

AkshayGupta94, 2016-12-05:
I am using arduino Mega 2560 board

kriswiner, 2016-12-05:
Likely somethingnot configured properly. Did you pull up #CS on Sparkfun's board? You should probably go through their tutorial again...

Are you using the correct SDA and SCL pins on you MCU?

AkshayGupta94, 2016-12-06:
I configured it according to the two tutorials.. In first case I connected SDA & SCL to a4 &a5 and in second case I connected them to SDA and SCL of MCU respectively.
Rest I didn't do any connections with the chip.
Do I specifically need to configure the CS pin as well?

kriswiner, 2016-12-07:
I don't know much about Sparkfun's board; the I2C bus has to be enabled on the MPU9250 by pulling up #CS; does their board do this or do you have to do this?

AkshayGupta94, 2016-12-07:
Their hook up guide doesn't mention this. I have ordered another board now.. I feel my board is defective maybe..

kriswiner, 2016-12-07:
Maybe try one of these?

https://www.tindie.com/products/onehorse/mpu9250-teensy-3x-add-on-shields/

AkshayGupta94, 2016-12-07:
Actually I need a module.. And my board is Arduino mega.. But anyways thanks for the help. Will update you when I get a new board!

AkshayGupta94, 2016-12-10:
okay so got a new module.. Connected it to sda and scl of my mega.. and run the i2c code here
https://github.com/kriswiner/MPU-9250/issues/37
it shows port 0x69.. my AD0 is high so yeah it should be 69

but the program still says 
MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF
Why is it so? am i missing something?
Do i need to modify some part in code?
I am very beginner in ardunio

AkshayGupta94, 2016-12-10:
it worked.. i figured it out and it works just fine!! thanks a lot!
So AD0 was set to low and it worked.
For mega one thing.. always connect SDA to SDA and not a4 or a5.. that was my main issue

closed 2017-02-28 by AkshayGupta94