kriswiner/MPU9250 #97: Altering code for use with different sensors

evinramadeen, 2016-12-13:
Hi Kris,
I am currently going through the process of attempting to change what i think is necessary to use an AltIMU 10 V5 sensor instead of the MPU9250 and MPU9150. I was forced to use the AltIMU sensor because it was readily available at the university where I am currently doing my BSc. 
My question to use is a rather strange one, do you think that by simply making changes in the main cpp file to include the necessary information to open and read my sensor values that I can then use the filter and all of the different calibration data present in the cpp file?

kriswiner, 2016-12-14:
As long as your sensor generates accel/gyro/mad data you can use the Madgwick/Mahony filters, but not the MPU9250-specific calibration data or functions. You will have to adapt the rest of the sketch to suit your particular sensor.

evinramadeen, 2016-12-21:
Hi Kris, 
I was finally able to complete the changes required to use the AltIMU 10 V5, [(at](https://www.pololu.com/product/2739) least the changes I believe was required) and I am now able to get the pitch roll and yaw readings out after using the open source Madgwick quaternion filter.
However, I have noticed that when I shift the sensor, there is a slight delay in the response to change the Serial display. As it stands, I have my magnetometer set up at 40Hz, in my eyes, by choosing this frequency which is higher than the 10Hz that you used for your mag, I should be able to get data faster and thereby update the attitude faster, am I wrong in thinking this?

Question number 2, I noticed that the MPU9250 has preset calibration values in ROM, a luxury that does not exist with my LIS3MDL magnetometer(at least based on the datasheet), as such I attempted to find a way to calibrate it. While searching, I came across another post of yours located at: (https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration) . However, this code also makes reference to magCalibration values that I can only think is the same as those stored in the ROM. Therefore, I altered the code to simply ignore the mag calibration values and subtracted the magbias from all new mag values. Is this a wise approach or should I look for another way to calibrate my magnetometer to get proper values to use in the algorithm which you used in the link given above?
If you desire to see the code that I made in my attempt to port your implementation, I shall post it immediately, I simply did not want to make this post seem longer than it already is.

Regards.

kriswiner, 2016-12-21:
> Effective-Magnetometer-Calibration) . However, this code also makes

The fusion rate is going to be dominated by the processor speed of your
host IMU, not the mag sample rate.

If you use an MCU with sufficient power to run the fusion at > 1 kHz, then
the next constraint will be the gyro sample rate. For most human motion
somewhere between 200 and 1000 Hz is where you want to be with the gyro
sample rate, if your processor can keep up.

If you mag doesn;t have factory calibration values then yes, you should set
these = 1or just remove these from the algorithm.

There is really only one way to calibrate a magnetometer. The simple method
in the github post is good enough for a lot of applications.

evinramadeen, 2016-12-21:
I'm using and Arduino Mega 2560 and I'm making the AHRS for a quadcopter flight controller that I am building. Next step will be to incorporate a Barometer. Thank you for you speedy reply. I will go and try to increase my gyro sample rate to see if that reduces the lag between updates. Thanks alot for your assistance. Knowing that the method is sufficient for the magnetometer cal is very helpful indeed, it had me wondering. 
Thanks again.
As a side note, I noticed that the 10DOF example you created was for a Teensy, can this code be used with an Arduino Mega 2560? I am thinking about purchasing a Grove 10DOF IMU (https://www.seeedstudio.com/Grove---IMU-10DOF-v2.0-p-2691.html#) to use for my quadcopter instead. I noticed that this uses a different barometer to the LPS25H that you used but I will be able to adjust the code to incorporate the barometer that comes with the Grove, but I digress, can the code be used with an Arduino Atmega or will alot of changes be required?

kriswiner, 2016-12-21:
> Grove---IMU-10DOF-v2.0-p-2691.html#) to use for my quadcopter instead. I

Some changes would be required. I would go for the Teensy if I were you.