kriswiner/MPU9250 #98: Can't get it to work...

crosspatch, 2016-12-21:
Not sure why it won't work...I'm using this board(https://www.amazon.com/Diymall%C2%AE-Mpu-9250-Nine-axis-Attitude-Acceleration/dp/B00OPNUO9U)

Hooked it up
VCC-3.3V
GND-GND
SCL-A5
SDA-A4
INT-2

"MPU92509-DOF 16-bitmotion sensor60 ug LSBMPU9250 
I AM FF I should be 71
MPU9250
I AM FF I Should Be 71Could not connect to MPU9250: 0xFF"

Would someone be able to assist me with this?

kriswiner, 2016-12-21:
> Diymall%C2%AE-Mpu-9250-Nine-axis-Attitude-Acceleration/dp/B00OPNUO9U)

What I2C devices show up when you do a scan?

crosspatch, 2016-12-21:
@kriswiner I'm sorry, I'm new to arduino, how would I perform a scan?

kriswiner, 2016-12-21:
void I2Cscan()
{
// scan for i2c devices
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    error = Wire.transfer(address, NULL, 0, NULL, 0);

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
}

crosspatch, 2016-12-21:
@kriswiner it's saying 'class TwoWire' has no member named 'transfer'

kriswiner, 2016-12-21:
wrong device. Try:

// I2C scan function

void I2Cscan()
{
// scan for i2c devices
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknow error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

}

crosspatch, 2016-12-21:
okay...thank you.

This is what I get
Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
I2C device found at address 0x77  !
done

crosspatch, 2016-12-21:
@kriswiner does this mean something needs to be changed in the library?

kriswiner, 2016-12-21:
Well, if you use 0x68 as the MU9250 address you should be able to read the
WHO_AM_I register.

crosspatch, 2016-12-21:
@kriswiner 
Changed it to byte c = readByte(0x68, WHO_AM_I_MPU9250);

and this is what happened...I'm not sure why the first device found before has disappeared now.

Scanning...
I2C device found at address 0x68  !
I2C device found at address 0x77  !
done

MPU92509-DOF 16-bitmotion sensor60 ug LSBMPU9250 
I AM 73 I should be 71
MPU9250
I AM 73 I Should Be 71Could not connect to MPU9250: 0x73

kriswiner, 2016-12-21:
I am 73 means you have an MPU9255. More or less the same as MPU9250 but
with some extra functionality. 0x0C is the AM8963C magnetometer; you have
to enable the I2C bypass to see it.

crosspatch, 2016-12-21:
@kriswiner okay so I should change this line[if (c == 0x71)] to [if (c == 0x73)] then?

kriswiner, 2016-12-21:
yes

crosspatch, 2016-12-21:
@kriswiner it works but where it says pitch, it's actually the roll and vice versa...the pitch is absolutely perfect though...i'll have to subtract the base value of the roll to get that one to be perfect and the yaw is just super large for the base value. Any advice on how to get these to be more accurate?

kriswiner, 2016-12-21:
calibrate your sensors

crosspatch, 2016-12-21:
@kriswiner also...I notice that you have the acceleration values in millig's....shouldn't the z value be 0.001mg just resting on the table...but I'm getting values like ax=-111mg, ay= 0mg and az=889mg...would this be solved with calibration as well or is this another issue?

kriswiner, 2016-12-21:
at rest accel values should by 0, 0, 1000 mg

crosspatch, 2016-12-21:
@kriswiner right...rookie mistake...okay thank you very much! Also...is there some sort of machine learning algo in here? I'm not really doing much other than making a few formatting changes but it seems to get better in it's resting pitch and roll whenever I unplug and plug it back in

kriswiner, 2016-12-21:
It will be random until you calibrate it; no machine learning but what you
program in.

crosspatch, 2016-12-21:
@kriswiner gotcha...thank you very much!!

crosspatch, 2016-12-21:
@kriswiner I have been using the Basic AHRS with the WIRE library because I can't find i2c_t3.h...is there a location to get this particular library? and will it work with the MPU9255 and arduino uno?

kriswiner, 2016-12-21:
That is a Teensy 3.1-only library.

crosspatch, 2016-12-23:
@kriswiner ah okay. Thank you!

evinramadeen, 2017-01-03:
Hi Kris, my sensor was working fine then my arduino somehow disconnected briefly from my laptop(heard the disconnection sound that Windows makes). Now when I start the sensor it is saying I am FF instead of the desired 71, does that mean my sensor is no longer working?

kriswiner, 2017-01-03:
What does the I2C scan show?

evinramadeen, 2017-01-03:
Scanning...
I2C device found at address 0x68  !
I2C device found at address 0x77  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF

I am starting to look into the possibility of there being a short either by two wires touching(although I have them covered with heat shrink) or the connecting cable between the arduino and my laptop. But above shows what comes up when I run the code and the scan is executed

kriswiner, 2017-01-03:
And what address are you using for the MPU9250?

evinramadeen, 2017-01-03:
0x68 so the problem is just the MPU is sending back that it is FF instead of the desired 71. The sensor is not blown because sometimes it is located and other times it returns FF. So I guess that must be a hardware problem on my side.

kriswiner, 2017-01-03:
Not enough power delivered to the MPU9250 I'd guess; faulty wiring
probably...

evinramadeen, 2017-01-03:
Thanks, will change my wires and try to separate my connections and see if that solves the issue. I saw on another thread a possible idea of how to get consistency of my readings when I restart the Arduino, hopefully I can solve that issue easily as well. thanks alot again.

crosspatch, 2017-01-05:
@kriswiner can you explain the timing of the program? It's giving me values every second?

kriswiner, 2017-01-05:
Yes since that is what it is asked to do; not sure what you question is...

crosspatch, 2017-01-05:
I added this bit of code at the end of the loop

File dataFile = SD.open("data4.txt",FILE_WRITE);

    if(dataFile){
      dataFile.println((int)yaw);
      dataFile.println((int)pitch);
      dataFile.println((int)roll);
      dataFile.println((float)altitude,2);
      dataFile.println("\n");
      dataFile.close();
    }
    else{
      Serial.println("Couldn't write to file");
    }

but it's giving me more values than the serial gives me so I'm not sure how to edit that to give me the same values as serial because I'm not sure I understand how it was set to one second. @kriswiner

kriswiner, 2017-01-05:
There is an if statement that checks when the time elapsed reached 1000 milliseconds and then outputs serial.

crosspatch, 2017-01-05:
@kriswiner this ?

if (delt_t > 500)

kriswiner, 2017-01-05:
yup

crosspatch, 2017-01-05:
@kriswiner thank you

NickMortimer, 2017-07-19:
Hi I'm having problems getting this going on pro mini, been searching around I have wired it all up at per sparkfun guide but get this MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF

basically I'm getting 0xFF for all reads do I need to connect the chip select in SPI mode?

Thanks

Nick

kriswiner, 2017-07-19:
"do I need to connect the chip select in SPI mode?"

yes, please read the data sheet

NickMortimer, 2017-07-20:
Thanks I'm new to Github and was trying to delete my question after posting it (read the manual). I think the terminology that is being used is confusing (SPI and I2C). My problem was having multiple libraries installed and them not playing well together. Thanks for you  kind and moderated reply.

kriswiner, 2017-07-20:
Glad to be of some help...

avsingh999, 2018-06-12:
@kriswiner  I have same error
````
MPU9250 I AM 73 I should be 71
Could not connect to MPU9250: 0x73
````
I want to know only I have to change only c == 0x73 in if loop line no. 323 
if I change c 0x71 to 0x73 then it's work fine
 
can I change it or not??

kriswiner, 2018-06-12:
Yes, go ahead and change it, your board is using an MPU9255, which has
WHO_AM_I of 0x73.

avsingh999, 2018-06-12:
I have MPU9250.

[image]

[image]

avsingh999, 2018-06-12:
@kriswiner  As you say when I change WHO_AM_I to 0x73 and c = 0x73 in if loop then I got this error
````
MPU9250 I AM 0 I should be 71
Could not connect to MPU9250: 0x0
````

If I Only change in if loop c == 0x73 then it's work fine 
 should I change  WHO_AM_I to 0x73  or not?

Please help me?

kriswiner, 2018-06-12:
Nope, your IC is an MPU9255.

kriswiner, 2018-06-12:
Just change c = 0x73.

avsingh999, 2018-06-12:
thank you @kriswiner

iFrostizz, 2020-04-12:
Won't work if I write c = 0x73, same error message

kriswiner, 2020-04-12:
What is the WHO_AM_I register returning?