kriswiner/MPU9250 #99: Yaw angle dependent on Pitch angle

przemekBielak, 2016-12-21:
Hi Kris,
I have a quick question regarding the yaw/pitch/roll angles.
The angles are converted to pitch and roll correctly, so if I move the sensor, the values of angles are updated correctly. I have a problem with yaw angle, because it is dependent on pitch angle, so if I rotate the sensor in pitch axis, the value of pitch is changed correctly, but it also changes the yaw angle (the range of change is big, in resting position is for example 30, than when I pitch the sensor 90 degrees, the yaw angle is changed to about 90 or - 30). 
Do you know where is this error is comming from? Is my sensor faulty, or do I have some errors in my code?
I also checked the raw measurements from the magnetometer and when I move the sensor in one axis, the readings are changed in all of the axes.
I would really appreciate your help.

Thanks,
Przemek

kriswiner, 2016-12-21:
The magnetometer is behaving correctly. You might need to calibrate the
sensors. You also need to pay attention to how you are passing the sensor
data to the mdwick algorithm. The algorithm expects the data as NED, which
is not necessarily x,y,z.

przemekBielak, 2016-12-21:
Thanks for the reply.
I am passing the correct arguments to the mdwick algorithm , I compared my version of code with yours and it looks the same. Can it be caused by wrongly calibrated mag sensor? I used your code for calibrating the mag sensor to get the values and than I passed them to my mag reading function. 

` 
  myIMU.readMagData(myIMU.magCount);
  myIMU.getMres();
  myIMU.magbias[0] = 294.0;
  myIMU.magbias[1] = 105.0;
  myIMU.magbias[2] = -149.0;
  
    myIMU.mx = (float)myIMU.magCount[0]*myIMU.mRes*myIMU.magCalibration[0] -
               myIMU.magbias[0];
    myIMU.my = (float)myIMU.magCount[1]*myIMU.mRes*myIMU.magCalibration[1] -
               myIMU.magbias[1];
    myIMU.mz = (float)myIMU.magCount[2]*myIMU.mRes*myIMU.magCalibration[2] -
               myIMU.magbias[2];

    myIMU.mx *= 0.80;
    myIMU.my *= 0.99;
    myIMU.mz *= 1.35;`

Is something missing from that function?

kriswiner, 2016-12-21:
Did you plot Mx vs Mz? Is it a sphere?

gjwo, 2016-12-21:
This isn't a calibration issue, it is the same issue as I have #93 The yaw changes because the magnetometer flips direction as it passes 90 degrees. I can demonstrate the same effect in Kris's code converted to java  here https://github.com/gjwo/RPISensors. It may be a feature of Tait-Bryan angles but I can't be sure as I haven't found a mathematician who can explain it to me. 
Kris, does your set up do the same thing?

gjwo, 2016-12-21:
See also issue #45

przemekBielak, 2016-12-21:
Here is the Mx/Mz plot:
[Values during calibration](https://i.imgur.com/a49H7b4.png)
[Values after calibration, during program runtime](https://i.imgur.com/ThZRWE5.png)
Does it look correct?

kriswiner, 2016-12-21:
Is it a symmetric circle centered on the orgin? No....

When you calibrate the mag, you have to mode the device all aorund in order
to sample the whole response surface.

przemekBielak, 2016-12-21:
Ok, thank you, I really appreciate your help.
I will to try to callibrate it one more time tomorrow, and than I will post my results.

gjwo, 2016-12-21:
Kris,
please could you post plots of Bryan-Tait angles during a complete 360 degrees rotation of your device in each of the X,Y and Z planes for comparison to my plots in #90 this would really help to clear up if there is a common problem here.
@przemekBielak and @mohit5037 if you could do the same on your set ups, that would nail if the implementations and problems are the same or not.

kriswiner, 2016-12-21:
I have some of that here:

https://github.com/kriswiner/MPU-6050/wiki/Hardware-Sensor-Fusion-Solutions

gjwo, 2016-12-22:
Hi Kris,
I think those graphs are only showing heading (yaw) performance against time for comparing different sensors, it is important to have yaw pitch and roll as 3 lines simultaneously to see how yaw changes when pitch does a 360 degree rotation. Unless I am misinterpreting what you are showing that isn't what you have on that page,

kriswiner, 2016-12-22:
Try this one then: https://forum.pjrc.com/threads/25965-9-and-10-DoF-LSM9DS0-shield-for-Teensy-3-1?p=117999&viewfull=1#post117999

gjwo, 2016-12-23:
Hi Kris, yes that looks like it doesn't have the problem but it is a different device the  LSM9DS0, and  the signs in the call to Madgwick:
MadgwickQuaternionUpdate(ax, ay, -az, gx*PI/180.0f, gy*PI/180.0f, -gz*PI/180.0f, mx, my, mz);
are different to the ones for the MPU-9250:
MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);
So given your description in the text of "wicked crosstalk between pitch and yaw" which is what we are seeing I think a graph from similar your MCU-9250 would still be helpful.

kriswiner, 2016-12-23:
The point of the post is that you have to send the Madgwick filter the
propoer sensor data; it expects the data in a NED coordinate system. If you
don't do it correctly, you will get all kinds of wierd behavior.

przemekBielak, 2016-12-23:
Hi Kris, I was able to properly calibrate the mag sensor, here are the results:
[x/z](https://i.imgur.com/nLG5dOn.png)
[x/y](https://i.imgur.com/seLZ5qQ.png)
[y/z](https://i.imgur.com/guyrSmS.png)

When the sensor is flat on the table and I twist it , the yaw reading finally is correct, so you were right about the calibration. However when I add pitch motion, the yaw angle is lost and outputs strange values same as before.

I will try to modify the Madgwick filter inputs in my code, maybe that is the problem.

gjwo, 2016-12-23:
Hi Kris,
I have double checked my use of signs and the switch of "my" with "mx" so my code is using the same NED coordinate system as yours. Given that 2 other people have observed the same behaviour on yaw when the 90 degree point is reached on pitch, it is possible that your code has the same problem, which is why I am asking for the graph from your code on the MCU-9250. If that doesn't have the problem there must be some way I have failed to reproduce your algorithm in java, but if it does show it, we can all work on solving the same problem.

przemekBielak, 2016-12-24:
Hi Kris, it is working now!
The NED coordinate system was the key to my problem. Arduino code provided by the manufacturer was wrong in one place, because the magnetometer axes were adjusted to accel coordinate system, not the other way round as it should be:
[old, wrong coordinates](https://i.imgur.com/dGI3Obh.png)
[new, ajdusted to mag coordinates](https://i.imgur.com/zfoUo7B.png)

Also, the mag calibration played a big role, because now it is working good even with slow Arduino, the response is pretty quick and values are correct.

Thank you very much for your help!

clhne, 2018-04-12:
@przemekBielak 
what's your order in MahonyQuaternionUpdate( )?
Thank  you.

przemekBielak, 2018-04-12:
@clhne 
For the function parameters I have:
(myIMU.ay, myIMU.ax, -myIMU.az, myIMU.gy * DEG_TO_RAD,
myIMU.gx * DEG_TO_RAD, -myIMU.gz * DEG_TO_RAD, myIMU.mx,
myIMU.my, myIMU.mz, myIMU.deltat)

Hope that helps.

kriswiner, 2018-04-12:
You are going to have to figure this out for yourself, but I will get you
started.

1) decide which edge of your board is going to be North.
2) figure out which sensor axes face North
3) repeat for East and Down.

clhne, 2018-04-13:
@kriswiner 
Yes, I have figure out my board and sensor axes. What to do next ?
I also found when Pitch become 90 degree,  the Yaw and roll angle abrupt change.
Thanks and Regards,
clhne.