kriswiner/MPU9250 #102: MPU9250 tilt compensated compass and Y-Axis inclination

Lymkin, 2016-12-30:
Is it possible to convert the quaternions calculated from your libray into a tilt compensated compass reading?  I also would like to calculate Y-Axis inclination angle.  I've tried every equation I could find on the internet for a couple of weeks now with no luck.  Any insight if what I am trying to get is even possible?  My lack of a PhD in mathematics just leaves me confused!

kriswiner, 2016-12-30:
It is possible to convert the quats into yaw (heading) pitch and roll. The
heading is a tilt-compensated compass reading. Is this what you mean? Pitch
and roll are x- and y-axis inclination angles. What's the problem exactly?

Lymkin, 2016-12-30:
So, is yaw from your code suppose to already give the heading?
    yaw   = atan2(2.0f * (q[1] * q[2] + q[0] * q[3]), q[0] * q[0] + q[1] * q[1] - q[2] * q[2] - q[3] * q[3]);   
    pitch = -asin(2.0f * (q[1] * q[3] - q[0] * q[2]));
    roll  = atan2(2.0f * (q[0] * q[1] + q[2] * q[3]), q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3]);
    pitch *= 180.0f / PI;
    yaw   *= 180.0f / PI; 
    yaw   -= 1.15; // Declination at my location
    roll  *= 180.0f / PI;

Do I need to convert those values to get actual degrees? (0-360)?  When my MPU is flat on a surface and rotated my yaw only shows a range from about 15.25 to 89.88.

I'm trying to measure azimuth and elevation to track the sun at a static location.  So, I basically want to attach the MPU to a solar panel so I can feed azimuth and elevation to my controller and have it move the panel to that location based on MPU input.

kriswiner, 2016-12-30:
Yaw pitch and roll are in degrees.

Did you calibrate your sensors?

How?

Lymkin, 2016-12-30:
Good point....I did not calibrate, but I do not see how to perform a calibration within the code.  I'm using MPU9250BasicAHRS.ino example

I think I was editing my previous post after you replied.

Are these static values for all 9250's or something I have to derive?  Is this where the calibration values come into play?
magbias[0] = +470.; // User environmental x-axis correction in milliGauss, should be automatically calculated
magbias[1] = +120.; // User environmental x-axis correction in milliGauss
magbias[2] = +125.; // User environmental x-axis correction in milliGauss

kriswiner, 2016-12-30:
Don't use those static values, that is for a particular device, not yours.
Use this sketch:

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

It has calibration functions in it.

Plot Mx vs Mz to see if calibrated. You shoukd see a circle centered on the
origin if properly calibrated. See:

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration

Lymkin, 2016-12-30:
Calibration definitely helped for pitch as it seems pretty accurate, but yaw still not reliable.  I can put the MPU on a flat surface and rotate it 180 degrees and the values are only changing about 90 degrees.
Unfortunately I cannot plot anything as I'm running this off of an ESP8266 and I don't have a display other than serial output at the moment.

Here is some output:
MPU9250 9-axis motion sensor...
MPU9250 I AM 71 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : 0.3% of factory value
y-axis self test: acceleration trim within : 0.9% of factory value
z-axis self test: acceleration trim within : 1.2% of factory value
x-axis self test: gyration trim within : -2.5% of factory value
y-axis self test: gyration trim within : 0.2% of factory value
z-axis self test: gyration trim within : 0.0% of factory value
 Calibrate gyro and accel
accel biases (mg)
-3.54
-9.83
5.55
gyro biases (dps)
-0.73
-0.22
0.82
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
mag biases (mG)
89.90
124.47
140.43

Here is rotation of what should be 90 degrees around zero, but it shows only about 45 degrees either way:
ax = -86.36 ay = 23.62 az = 995.24 mg
gx = 0.07 gy = -0.05 gz = -0.70 deg/s
mx = -17 my = 550 mz = 105 mG
q0 = -1.00 qx = -0.01 qy = -0.04 qz = -0.02
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 0.95, 4.96, 1.36
rate = 629.44 Hz
ax = -86.18 ay = 23.25 az = 998.35 mg
gx = 0.03 gy = -0.08 gz = -0.74 deg/s
mx = -8 my = 555 mz = 138 mG
q0 = -1.00 qx = -0.01 qy = -0.04 qz = -0.01
Gyro temperature is 29.2 degrees C
Yaw, Pitch, Roll: 0.09, 4.93, 1.34
rate = 633.74 Hz
ax = -86.61 ay = 21.61 az = 997.31 mg
gx = 0.08 gy = -0.02 gz = -0.89 deg/s
mx = -23 my = 552 mz = 149 mG
q0 = -1.00 qx = -0.01 qy = -0.04 qz = -0.02
Gyro temperature is 29.2 degrees C
Yaw, Pitch, Roll: 1.55, 4.96, 1.24
rate = 635.80 Hz
ax = -100.65 ay = 28.87 az = 1000.61 mg
gx = -2.37 gy = -0.98 gz = 26.52 deg/s
mx = -70 my = 548 mz = 48 mG
q0 = -1.00 qx = -0.01 qy = -0.04 qz = -0.07
Gyro temperature is 29.2 degrees C
Yaw, Pitch, Roll: 6.89, 4.83, 1.44
rate = 634.39 Hz
ax = -144.29 ay = 33.81 az = 994.45 mg
gx = -2.75 gy = -0.71 gz = 35.93 deg/s
mx = -122 my = 514 mz = 57 mG
q0 = -0.99 qx = -0.01 qy = -0.04 qz = -0.12
Gyro temperature is 29.2 degrees C
Yaw, Pitch, Roll: 12.20, 4.75, 1.49
rate = 634.14 Hz
ax = -42.30 ay = 5.43 az = 1003.23 mg
gx = -1.75 gy = -1.54 gz = 23.83 deg/s
[... 136 lines cut ...]
Yaw, Pitch, Roll: -40.75, 5.89, 0.99
rate = 634.98 Hz
ax = -102.29 ay = 17.03 az = 995.00 mg
gx = -0.26 gy = -0.12 gz = 0.07 deg/s
mx = 273 my = 326 mz = 22 mG
q0 = -0.94 qx = -0.03 qy = -0.05 qz = 0.34
Gyro temperature is 29.8 degrees C
Yaw, Pitch, Roll: -40.46, 5.92, 0.95
rate = 634.85 Hz
ax = -105.22 ay = 16.48 az = 993.04 mg
gx = 0.35 gy = 0.08 gz = -7.43 deg/s
mx = 262 my = 308 mz = 46 mG
q0 = -0.94 qx = -0.03 qy = -0.05 qz = 0.33
Gyro temperature is 29.8 degrees C
Yaw, Pitch, Roll: -40.25, 5.94, 0.97
rate = 633.52 Hz
ax = -104.31 ay = 16.05 az = 994.20 mg
gx = -0.13 gy = -0.19 gz = -0.12 deg/s
mx = 273 my = 283 mz = 74 mG
q0 = -0.93 qx = -0.03 qy = -0.05 qz = 0.36
Gyro temperature is 29.8 degrees C
Yaw, Pitch, Roll: -43.76, 5.98, 0.90
rate = 633.12 Hz

What orientation should the sensor be in to calibrate? What axis of movement for the figure 8? Could my sensor be bad as calibrations seem wildly different everytime.

matherp, 2017-01-21:
The critical thing is to get the relative orientations correct in the inputs to the sensor fusion

For a MPU-9250 it should be

ax, ay, az, gx, gy, gz, **my, mx, -mz**

note the negative sign on mz. This should be applied after all the corrections for scale and offset are applied.

Here is my code in Basic for this demo
https://www.youtube.com/watch?v=4eBtP6cpqU4&feature=youtu.be

Note the use of sine, cos and atan2 to get a rolling average of readings whilst solving the 359-0 problem

Hope this helps

DO`  
readMPU9250(0)

'  SensorFusion Madgwick ax, ay, az, gx, gy, gz, my, mx, -mz, pitch, roll, yaw', beta 
  SensorFusion Mahony ax, ay, az, gx, gy, gz, my, mx, -mz,  pitch, roll, yaw', Kp, Ki
'
'rolling average of yaw angles (0-359)
'
  if yaw<0.0 then yaw=yaw+ 2*pi
  ctemp=cos(yaw)
  stemp=SIN(yaw)
  yawMAc=yawMAc-yawSMAcos(temp%)/MA
  yawMAs=yawMAs-yawSMAsine(temp%)/MA
  yawMAc=yawMAc+ctemp/MA
  yawMAs=yawMAs+stemp/MA
  yawSMAcos(temp%)=ctemp
  yawSMAsine(temp%)=stemp
  yawMA=deg(atan2(yawMAs,yawMAc))
  heading   = yawMA - declination
  gui startline origin%
  origin% = (origin% +272) mod 544
  drawpointer(heading, origin%)
  temp%= (temp% +1) mod MA
LOOP`