kriswiner/MPU9250 #103: Using with Arduino Atmega 2560

evinramadeen, 2016-12-31:
Good day Kris,
I was able to obtain an MPU9250 but unfortunately, the barometer it comes packaged with is a BMP280. I understand that for use with your 10DOF code, I will have to alter the necessary files and address to change  from the LPS25H that you used to accomodate my BMP280( I have the GROVE - IMU 10DOF V2.0).
My main question to you is; do I have to make alot of changes to the code to get it to work with the Atmega 2560 as I don't have a teensy 3.1?

kriswiner, 2016-12-31:
Shouldn't have to, a few changes in the I2C calls.

Here is an MPU9250+BMP280 sketch for the Teensy, I think the wire calls are
commented out. Should be an easy change:

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_BMP280_BasicAHRS_t3.ino

evinramadeen, 2016-12-31:
Thanks alot Kris. This will help me greatly.