kriswiner/MPU9250 #105: Hardcode calibration

chardur, 2017-01-03:
Hi Kris, thanks for all your work on the 9250!

I have ran the calibration sketch and I am having trouble hardcoding the calibration data into the basic example sketch. After adding the calibration data, the yaw does not seem to work, it will only move slightly, even when fully rotating the IMU. One problem may be the magScale data. I'm not sure how to add these figures to the example sketch. 

Here is the data from calibration: 

x-axis self test: acceleration trim within : -0.8% of factory value
y-axis self test: acceleration trim within : 1.0% of factory value
z-axis self test: acceleration trim within : 1.2% of factory value
x-axis self test: gyration trim within : -0.4% of factory value
y-axis self test: gyration trim within : -0.2% of factory value
z-axis self test: gyration trim within : 0.9% of factory value
MPU9250 bias
 x   y   z  
accelBias[0] : -38
accelBias[1] : -4
accelBias[2] : 41
mg
gyroBias[0] : -3.8
gyroBias[1] : 0.5
gyroBias[2] : -1.3
o/s
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
-2257.71
-1102.20
592.35
AK8963 mag scale (mG)
0.87
0.79
1.69
X-Axis sensitivity adjustment value 1.19
Y-Axis sensitivity adjustment value 1.19
Z-Axis sensitivity adjustment value 1.15

In the basic example sketch I have added the following data: 

  //hardcoded calibration
  myIMU.accelBias[0] = -38;
  myIMU.accelBias[1] = -4;
  myIMU.accelBias[2] = 40;

  myIMU.gyroBias[0] = -3.8;
  myIMU.gyroBias[1] = 0.6;
  myIMU.gyroBias[2] = -1.3;

  myIMU.magCalibration[0] = 1.19;
  myIMU.magCalibration[1] = 1.19;
  myIMU.magCalibration[2] = 1.15;

    // User environmental x-axis correction in milliGauss, should be
    // automatically calculated
    myIMU.magbias[0] = -2257.;
    // User environmental x-axis correction in milliGauss TODO axis??
    myIMU.magbias[1] = -1102.;
    // User environmental x-axis correction in milliGauss
    myIMU.magbias[2] = +592.;

What am I doing wrong?

Thanks! 

-Charles

chardur, 2017-01-03:
The yaw seems to work if I keep the following line in the example sketch:

myIMU.initAK8963(myIMU.magCalibration);

Previously I had commented this line out, thinking that the card coded values would be used instead.

kriswiner, 2017-01-03:
Not sure, I don't really understand what you are trying to do here.

closed 2017-01-20 by chardur