kriswiner/MPU9250 #106: Reading Accelerometer @ 4khz problem

Hesquibet, 2017-01-03:
Hello,

I would like to read acceleration data at 4khz, so i should get 1 new value every 0,25ms.
So far, i can only get a new value every 1ms which is only 1khz

What i have tried : 
As nicely indicated in the library, the bit accel_fchoice_b has been set to 1.
As indicated in the datasheet, SMPLRT_DIV has been set to 0 ( as it divides the sampling rate ) ( page 12 in register map  doc)

Did anyone got 4khz output reading from the MPU9250 ?
At which frequency do we withdraw new data from the sensor ? at Sampling Rate (4khz) or at Bandwith rate ( 1khz) ?

Any help is welcomed :)

Thanks

kriswiner, 2017-01-03:
For 4 kHzaccel data rate looks like you need to set bit 3 (accel_fchoice_b)
in the Register 29 – Accelerometer Configuration 2 to 1, otherwise the rate
is controlled by DLPF and is always 1 kHz or less depending on
sampleratedivide.

kriswiner, 2017-01-03:
Can you verify that bit 3 is 1 then?

Hesquibet, 2017-01-03:
Hey Kris,

Thank you very much for your quick answer !

I have checked the register 29 by reading it, the output is  : 0000 1000. So it looks OK.

Are you saying that in any cases (even with accel_fchoice_b set to 1), the maximum output data rate is limited by the DLPF and will never exceed 1Khz ? 
My understanding was that accel_fchoice_b set to 1 could bypass the DLPF and have an Output data rate of 4Khz. But i can't get it to work, so i might be missing something....

I need a 4Khz output data rate to apply my own band-pass filter on MCU.

Do you think there is any way to get an Output Data Rate of 4Khz ?

thanks again

kriswiner, 2017-01-03:
If accel_config_2 is 0b00001000 then you should be at 4 kHz independent of
DLPF, this bit disables the DLFP. How are you determining the update rate?
If by interrupt, do you have a gyro running at 1 kHz? I would turn off the
gyro and mag and check the interrupt rate on accel data ready only. It
should come in at 4 kHz with this bit set and 1 kHz with this bot set to
zero and sampleratedivide = 0. If not, I would send a note to Invensense
because their Register map would be wrong.

Hesquibet, 2017-01-04:
Good guess, disabling the Gyro did the trick !
i have added writeByte(MPU9250_ADDRESS, PWR_MGMT_2, 0b00000111); in the init sequence and now i get interrupt every 0.23ms.
When you think about it, it seems logic to rise an interrupt only when the whole data ( Gyro + Accel ) is available. I should have though about it by myself... 
Anyways, thanks a lot for you help Kris ! 
i owe you a beer !

closed 2017-01-04 by Hesquibet

mojtabahdpr, 2017-11-12:
Hi Hesquibet
can i have 4kH sample rate by i2c serial with your library ?

kriswiner, 2017-11-12:
The accel max rate is 1 kHz IIRC, consult the data sheet.

mojtabahdpr, 2017-11-12:
no in data sheet mpu9250 have 4khz sample rate

kriswiner, 2017-11-12:
Then follow the instructions in the data sheet?

mojtabahdpr, 2017-11-12:
this is possible with i2c ?

kriswiner, 2017-11-12:
yes

mojtabahdpr, 2017-11-12:
with hardware can i suggest 4khz ?

shirish47, 2018-05-31:
Hi, I think and I am also trying this 4k Sample thing. F_choice is inverse as from some libraries I have seen all keep that bit to 0 and set the first 3 bits for different LPF bandwidth. if you set it to 1 it will disable and you can get 4K sample rate. a fifo implementation would be a good choice. I have had troubles with I2C implementation so I moved on to I2C I used a library and modified it for my use case. that library does not have the functionality to give 4K samples. here it is https://github.com/bolderflight/MPU9250

5Priem, 2018-08-20:
Hi,
@Hesquibet You said you were able to receive the data at 4 kHz (every 0,25 ms), is this with I2C or SPI? Thank you for your answer
@kriswiner Sorry to bother you with my question so late after the topic, but maybe you can help me. I am using I2C to get a 4 kHz rate. However, I see on my oscilloscope that there is a moment of 130 µs where nothing happens. This moment is in between reading 2 bytes. So it goes like this:

1. read byte (160µs) 
2. wait (130 µs)
3. read next byte (160 µs)
4. wait (130 µs)
5. and so on...

It seems that my main computer (a Beaglebone Black Wireless) just randomly waits 130 µs. The process of reading 1 byte takes 160 µs. I am trying to read data from the MPU-9250. In my code, I wrote nothing else than "read register of sensor" in a while loop (I didn't manage to write a tab infront of the second line here): 
`while True: `
    `data = bus.read_i2c_block_data(0x68, 0x3B, 1)`
My question is: why is it waiting 130 µs? I read online ([here](https://www.intersil.com/content/dam/intersil/documents/an17/an1757.pdf), page 2, section **Byte read**) that the data should come immediately after the previous one so I don't understand this delay.
Any help would greatly be appreciated!
Thanks in advance.

kriswiner, 2018-08-20:
Well, I don't know but it might have to do with the bandwidth you have
selected.

But 1/(160+130 us) = 3448 Hz so maybe since you are asking for 4 kHz you
are getting data every ~250 us?

5Priem, 2018-08-21:
Thank you for your fast answer!
If I understand correctly, it is the Beaglebone that is waiting 160 µs everytime, but he has no way of knowing that the data is only ready every 250 µs so he should not be waiting 160 µs to read from the I2C bus. Maybe it is just the I2C protocol that waits a moment before reading the next byte to avoid bytes "overlapping"?

I also set the accel_fchoice_b bit to 1 so the output rate should be 4 kHz, is that what you mean with bandwidth?

Thank you very much!

kriswiner, 2018-08-21:
Are you not using a data ready interrupt to tell the MCU to read the data?

5Priem, 2018-08-21:
No I am not, I am just reading the data constantly since I thought this would be faster than using an interrupt. You think it should work better when using an interrupt?

kriswiner, 2018-08-21:
I am just trying to understand your timing.

It might be that it takes 160 us for the I2C transaction, just don't know.

5Priem, 2018-08-21:
Yes that sounds weird but maybe that is it.. Thanks for your help and time!

bkdeepak95, 2018-09-12:
I am concentrating only on Accelerometer data from MPU9250, I have disabled Gyro and Magneto for faster communication. I am working on STM32L072 Microcontroller. I am able to capture 177 samples per second (1 sample has 6 bytes). I have followed your blog strictly(configured register acc to 4KHz ) but did not get expected result. I am targeting to achieve 4KHz sampling rate  but i have confusion. Could you give clarification on Sampling Rate ..? Let say if my Sampling rate is 4 KHz, is it means following: 4000 samples per second or 4000 bytes per second or 4000 bits per second ? 

main config i am enclosing..
uint8_t buf[2];
		// Clear sleep mode bit (6), enable ONLY ACCELEROMETER
		buf[0] = PWR_MGMT_2;
		buf[1] = 0x07;
		HAL_I2C_Master_Transmit(&hi2c1,MPU9250_I2C_SLAVE_ADDRESS,buf,2,I2C_TIMEOUT);
		DelayMs(100); // Wait for all registers to reset
	
		// Get stable time source
		// Auto select clock source to be PLL gyroscope reference if ready else
		buf[0] = PWR_MGMT_1;
		buf[1] = 0x01;
		HAL_I2C_Master_Transmit(&hi2c1,MPU9250_I2C_SLAVE_ADDRESS,buf,2,I2C_TIMEOUT);

Kindly help.

kriswiner, 2018-09-12:
4 kHz sample rate means 4000 6-byte (2 for each axis) sample per second.

How are you selecting the sample rate?

bkdeepak95, 2018-09-12:
I am writing 0 in register 29 bit[3], DLPF will suppose to be bypassed and rate will be 4KHz. I am finding some ambiguity on Output Data Rate vs sampling rate. I am not able to fully decrypt the Invense Datasheet MPU9250 (Register map , V 1.1). I want to create efficient algorithm for FALL DETECTION as i need some graphical approach on accelero data (should be at maximum sampling rate i.e 4K).

5Priem, 2018-09-12:
hi @bkdeepak95, when I tried to do it it did not work simply because the I²C protocol implements too many delays. It is not possible to remove these delays, and because of this I could not get 4 kHz, just because of I²C.
If you know how to use SPI, this would be faster to use with the mpu