kriswiner/MPU9250 #107: MPU9250 Initial Position and Orientation

memkabdelaziz, 2017-01-04:
Hi Kris, 

Thank you very much for your efforts. 

Any idea on how to ensure that the sensor gives me proper readings regardless of it's starting position and orientation ? The problem is, when I place on the desk orthogonal to the ground, the readings are fine. On the other hand, when I place it on a body floating in the air (not necessarily orthogonal to the ground), the readings are wrong, they seem to have a wrong starting value. 

Best Regards,
Mo

kriswiner, 2017-01-04:
How did you calibrate the sensors?

memkabdelaziz, 2017-01-04:
Well I simply followed your tutorial (upto the hard iron correction)
https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration

kriswiner, 2017-01-04:
> Effective-Magnetometer-Calibration

How do you know the values are wrong then?

If you plot Mx vs. Mz do you get a circle centered on the origin?

If yes, then the problem could be the way you are passing the sensor data
into the Madgwick fusion algorithm.

memkabdelaziz, 2017-01-04:
I print out the roll, pitch and yaw. 
Yes, Figure 1 (before) vs Figure 2 (after)
[image]

This is how I pass the data to the Mahony fusion algorithm
magbias[0] = 71;
magbias[1] = 39; 
magbias[2] = -130; 

  while(1) {

    if(readByte(MPU9250_ADDRESS, INT_STATUS) & 0x01) {
      
      readMagData(magCount); 
      mx = (float)magCount[0]*mRes*magCalibration[0] - magbias[0]; 
      my = (float)magCount[1]*mRes*magCalibration[1] - magbias[1];
      mz = (float)magCount[2]*mRes*magCalibration[2] - magbias[2];
    }

kriswiner, 2017-01-04:
You need to pass the sensor data in a NED reference frame, that is in the
first slot goes the data pointining in the North direction, second slot
East, and third slot Down. If the mag +Mz  is pointing up, for example, the
-mz goes in the third slot for the mag input, etc. The MPU9250 sensor axes
are not aligned.

evinramadeen, 2017-01-05:
Hi Kris, speaking of NED reference frame, I was going through your sketch for the MPU9250 10DOF sensor with the BMP280. There was a point when you were lining up the axis as the gyro/accel and magnetometer axis are different. You made a statement where you said that the magnetometer z-axis was +down and the accel/gyro z- axis was positive up. I was trying to figure out if my board followed that same setup so I attempted to research that board that i believe you used and found the MPU 9250 Teensy ad on shield but in the picture, it seemed to me that magnetometer z-axis was positive up and the the accel/gyro was positive down . I tried to find images that illustrate exactly which side of the z-axis references the positive side but have not located any. Did i locate a different board to what you used?
This is the board i found that I think you use 
[image]

and this is the board that I am currently using http://wiki.seeed.cc/Grove-IMU_10DOF/

Thanks in advance.

kriswiner, 2017-01-05:
This is the board I am using. But the NED depends on what you choose for
North, etc.

evinramadeen, 2017-01-05:
Okay, I get what you are saying, but I don't follow how you chose the direction for the Z-axis on the accel. From my understanding, if I take +X as north, -Y will be East so doesn't that mean that +Z is down as the arrow is going into the board?

kriswiner, 2017-01-05:
On my board, magnetometer +z axis is Down, accel/gyro +z axis is UP or
-DOWN, so I would load (  ,  ,  -Az,  ,  , -Gz,, ,,   Mz) into the Madgwick
function, etc.

evinramadeen, 2017-01-05:
Ok, I understand what you are saying. Thank you very much.

memkabdelaziz, 2017-01-05:
[image]

I use this board. In my case I picked 
Ax/Gx/My as my North 
Accordingly,
-Ay/-Gy/-Mx --> East
-Az/-Gz/ Mz  --> Down
MahonyQuaternionUpdate(ax, -ay, -az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f, my, -mx, mz);

Do I understand you correctly ?

kriswiner, 2017-01-05:
Yes, this should work now.

memkabdelaziz, 2017-01-06:
Well I noticed something weird. In the MPU9250 calibration funciton, you assumed that the gravity compensation is always in the z-axis, and this shouldn't be the case everytime because my starting position and orientation of the sensor might be:

[image]

In this case the gravity compensation should be in the y-axis not the z-axis. 

if(accel_bias[2] > 0L) {accel_bias[2] -= (int32_t) accelsensitivity;}  // Remove gravity from the z-axis accelerometer bias calculation
  else {accel_bias[2] += (int32_t) accelsensitivity;}
 
Unfortunately, even after entering everything correctly to the function, my yaw, pitch and roll are incorrect.

kriswiner, 2017-01-06:
Yes, this is correct. These sketches are examples, not one-size-fits-all
solutions and you have to adapt them to your particular situation.

memkabdelaziz, 2017-01-06:
Concerning the magnetometer AK8963, the expected raw data range is +-4800 microTeslas....when I print the raw data prior to calibraiton,bias,etc....my values only range from -110 to 220...shouldn't the raw data be larger in value ? I mean range from -4800 to 4800 ?

kriswiner, 2017-01-06:
If you are using my sketch the mag values are in milliGauss. Earth mag
field is ~500 milliGauss.

Dim-P, 2017-03-08:
> But because gravity is by convention positive down, we need to invert the accel data, so we pass -Ax, Ay, Az, Gx, -Gy, -Gz, My, -Mx, and Mz into the Madgwick

@kriswiner This bit is a comment in your code. Can you elaborate? Why do we need to change sings for the acceleration?

Also, just for the acceleration, it seems like I am getting +1g when the board is flat, which seems right, but shouldn't this be -1g, since +z points upwards?

Thanks!

kriswiner, 2017-03-08:
By convention gravity is +1 g when z-axis is pointing up.

You don't have to invert acceleration if you are getting good results. We
do it to match the results from the EM7180 in some cases.

cadcam18, 2018-06-14:
Hi
I'm not familiar with the MPU-9250 yet.
in my project I have a servo carrying another servo carrying a camera. I need a feed back on the position of the camera so I thought I will use MPU-9250. how can I get an absolute position as if I connect 2 absolute encoders to each servo. Is it possible to get the correct position after cold start.
how do I get your reply to my email without checking on github 
zuhair@alswayeh.com
your help is highly appreciated.
best regards