kriswiner/MPU9250 #109: Altitude drifts over time

evinramadeen, 2017-01-06:
Hey Kris, 
I have read in a lot of posts that the yaw tends to drift over time but this has not been an issue for me using your code, I have however noticed a slow drift upward in the altitude. I can't figure out what would cause that. The only thing I could think of is I am sending my different accel gyro and magnetometer values into the quarternion update function with the wrong polarity but I have not seen anything to suggest that I am correct in thinking this.
Do you have an idea of what causes the altitude to drift?

kriswiner, 2017-01-06:
Yaw won't drift for a 9DoF solution (that's what the magnetometer is for)
as long as your sensors are properly calibrated.

By altitude I assume you mean the altitude derived from the barometer. of
course, this will drift due to changes in the weather or even someone
opening a door or window. If altitude estimation stability is important to
your application you can use a Kalman filter to combine the accel and
barometer to drift stabilize the altitude. This amounts to rejecting
changes in altitude if the accel doesn't also detect z-axis accelerations.
This is a more advanced topic not the subject of open-source solutions
AFAIK.

evinramadeen, 2017-01-06:
Yes, I am currently required to do a project for my Bsc that needs this. I am required to create an Open Source Quadcopter control platform that must be able to perform indoor navigation based on hard coded inputs. That is, I am to hard code a flight plan that the quadcopter must follow so the altitude being read by the sensor needs to be accurate to ensure that unnecessary adjustments are not made by my flight controller and sent to my ESCs. 
Thanks for the suggestion, I will look into Kalman filtering using the accel and barometer data. or possibly even try to just accept change in altitude only if accelerometer z axis changes as well. That was a great idea. Thanks