kriswiner/MPU9250 #111: Calibration for acceleration biases

mon1020, 2017-01-24:
Hi Kriss,

We are working on MPU9250. We have referred you code at below link:
**https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS.ino**
We use this code calibrate the MPU and then read acceleration values to calculate angle.
The values are not coming properly. 
**-1.5205078125,-2,-1.453125,-142.75594569821885**
Please advise on the same.

Regards,
Monica

kriswiner, 2017-01-24:
> *-1.5205078125 <(520)%20507-8125>,-2,-1.453125,-142.75594569821885*

You will need more than accelerations to get angles.

Anyway, what does this mean?

"The values are not coming properly."

mon1020, 2017-01-24:
Hi Kriss,

Thanks for your quick reply. 

When we take accel values with default offset registers(not calibrated) the values are as below:
-0.04931640625,0.96923828125,0.24755859375 ,
Angle here is calculated as:
angle=atan2(x/y)*(180/pi);
For above example: angle= -2.9127896829355713
Through out all the rotation of IMU I get angle between -180 to +180. Which is correct.
I want to calibrate IMU for more stable values.

Please correct me if angle calculation is wrong here.

And after calibration, It is giving following values throughout all positions of IMU which give the same angle:
-1.520507813,-2,-1.453125
-1.567871094,-2,-1.434082031
-1.566894531,-2,-1.473632813
-1.59375,-2,-1.615722656
-1.568359375,-2,-1.477539063
-1.556640625,-2,-1.330078125
-1.539550781,-2,-1.464355469
-1.6015625,-2,-1.57421875
-1.515136719,-2,-1.39453125
-1.560546875,-2,-1.531738281
-1.594238281,-2,-1.504882813
-1.58984375,-2,-1.546386719
-1.577148438,-2,-1.453125
-1.641113281,-2,-1.483886719
-1.687011719,-2,-1.337890625
-1.538574219,-2,-1.475097656
-1.5859375,-2,-1.513183594
-1.543457031,-2,-1.415039063
-1.627929688,-2,-1.434570313
-1.522949219,-2,-1.408203125
-1.629882813,-2,-1.493164063
-1.612792969,-2,-1.511230469

It is same all around the rotation of IMU. Not sure if calibration is correct.
Also, in your code you have mentioned this comment:
"Apparently this is not working for the acceleration biases in the MPU-9250"

I am very new to IMU calibration.
Your help will be much appreciated.

Thanks,
Monica

kriswiner, 2017-01-24:
Yes, I wouldmn't use the accelerometer offset bias register to perfrom the
calibration, do it in the main loop instead.

mon1020, 2017-01-26:
Hi Kriss,

I am not sure I understand what you are saying here. How to use gyro values for angle calculation?
How can I fix the issue of incorrect acceleration values?

Regards,
Monica

TediumRemedy, 2017-01-26:
In theory the accelerometer gives you the acceleration (d2x/dt2, d2y/dt2, d2z/dt2, not the actual speed dx/dt, dy/dt, dz/dt, and not the actual position x,y,z). So you cannot use just the accelerometer to calculate the angle.
With gyro the story is similar: it returns you the **speed** of change of the angle (d(alpha)/dt, d(beta)/dt, d(theta)/dt), not the actual angle (alpha, beta, theta). [where alpha beta and theta - are the angles of rotation around axes x y and z]. So To get the actual angle, you must perform an integration of the gyroscopic measurements by time (i.e., summing the values of measured gyroscopic angle speed (say, d(alpha)/dt) times delta t (this is a small period of time upon which you can consider the angle to be constant)). See http://stackoverflow.com/questions/31278784/integrate-readings-from-a-3-axis-gyroscope

To my understanding, you additionally use the accelerometer and the magnetometer to improve precision of the initial gyro integration

kriswiner, 2017-01-26:
> questions/31278784/integrate-readings-from-a-3-axis-gyroscope

If you want yaw, pitch, and roll why not just use the open-source sensor
fusion algorithms.

For accel biases, do not use the hardware bias registers,just subtract the
biases in the main loop.

There are several examples of how to do both in the github repository.

What else is missing?

TediumRemedy, 2017-01-31:
> For accel biases, do not use the hardware bias registers,just subtract the
biases in the main loop.

@kriswiner could you tell why not? (Sorry if you already answered this question)