kriswiner/MPU9250 #112: Magnetometer Calibration, Mahoney Orientation, Breakout quality

chadwyoung, 2017-01-25:
Hi Mr. Winer, 

Your documentation on the MPU9250 chip has been incredibly helpful to me. Thanks for being so thorough. I am having trouble getting correct yaw angles despite calibrating the magnetometer and checking for NED orientation for the Mahony algorithm. 

http://imgur.com/a/XERxg

I can successfully see what I believe are correct roll and pitch values. They are generally within at least 5 degrees when I place the sensor vertical for +/-90 pitch and roll.

My issue is with the yaw angle. I tried calibrating the magnetometer (By waving in a figure eight in several orientations of my hand). I calibrated it once, and the point clouds went from picture 1 to 2 in the attached imgur album. I believe it worked at least once with that calibration. 

I then decided it might be the information fed to the Mahony algorithm. I experimentally determined the axes of the accelerometer and gyroscope, and found that they did not coincide. I did not understand well enough the magnetometer to determine its orientation. Basically, I found that the X and Y axes of the gyroscope were in opposite directions than their counterparts on the accelerometer. After modifying the signs of some of the gyroscope, and accelerometer axes (so that Z was +1g down), the code would still not supply accurate yaw angles. This diagram is visible in the imgur album at the end.  

I then calibrated it again (Similar to above, but also turning 360deg while waving the device) , and got an even worse set of clouds (but more full!), despite not changing the biases in the code since the first calibration. This iteration is visible in photos 3 and 4 of the album. This made the sensor read correctly and I had a chance to record the yaw angles.

The yaw angle graph is visible in photo 5. The main takeaway from this is that the accuracy is not that great. I turned 90deg and paused, holding the MPU9250, but the yaw graph only indicates I turned about 70deg. 

After resetting the chip with the reset button, the yaw angle no longer calculated correctly. This leads me to believe it has to do with the breakout I am using…

I saw that it is essential that Pin 1 of the MPU9250 is attached to the power supply, and that some cheap Chinese chips have not done this. I have the breakout in photo 6 of the album. I can’t see clearly if pin 1 connects to VCC in the image, but I am not familiar enough with PCBs to know for sure.  For what it’s worth, my acceleration values are pretty noisy, up to +/-10 mg, with the magnitude of all three typically being over 1000mg. The gyro is pretty steady, within about 0.2 deg/s. 

Thanks for your help, please let me know if I could explain anything more, or in a different way.

kriswiner, 2017-01-25:
Hi Chad,

I am not sure your breakout is to blame since the mag calibration plots you
show look pretty good to me.

But looking at the pictures, I don't see where pin 1 is connected to
anything, so this will be a problem. Even though this is not purple, the
CJMCU breakouts tend to have this problem of not connecting pin 1. They
also have traces running under the chip, which is not going to give very
good magnetometer stability, etc.

It is unlikely that the accel x/y axes are oriented differently than the
gyro ones. If you look at the data sheet, you will see they are the same.
But these are not the same as the magnetometer. Ax/Gx is usually the same
direction as My, Ay/G7 as -Mx and Az, Gz as - Mz. So for Madgwick or
Mahiony you should have:

MadgwickQuaternionUpdate(ax, -ay, -az, gx*pi/180.0f, -gy*pi/180.0f, -gz*pi/
180.0f, my, -mx, mz);

But check the data sheet to confirm.

I would recommend that if you want better accuracy for your heading
estimation, try a different breakout like this one:

https://www.tindie.com/products/onehorse/mpu9250-teensy-3x-add-on-shields/

I would try the Mini version.

Kris

chadwyoung, 2017-01-26:
Kris, 

Thanks for your fast response!

I think I did the magnetometer calibration incorrectly the last two times, adding biases together in the wrong way. I recalibrated it and with modifying the inputs to the algorithm, it appears to consistently give me good data! It takes a little while for the yaw to stabilize initially, but after that it is pretty quick and responsive. A little noisy though. It's annoying that they would put wires near the magnetometer. 

I am using this as an IMU for a drone. I am also attempting to code the flight controller. 

If it ends up being too noisy or inaccurate for yaw, I will head to your store to pick up one of those shields. The form factor is pretty sweet. 

Again, thanks for your help. You might hear from me again but hopefully not! 

Chad

wewillfindyou, 2017-01-27:
Hy,

i am making a drone control, but cant get a correct value for the yaw. I did everything. configuration from kris to get the bias and scale values. the yaw value is changing if spin my quadrocopter on the y and x . 

Please help. thanks

kriswiner, 2017-01-28:
Are you using a magnetometer? Is it calibrated? How? and How do you know it
is properly calibrated? Your symptom sounds like a calibration problem.

wewillfindyou, 2017-01-28:
hy,
i have mag.bias values: -59 260 -336, mag.scale: 0.88 1.1 1.05 made with your calibrationmethode for the 9250. 

the NED is double checked.  what do i miss?

wewillfindyou, 2017-01-28:
i am using the magentometer from the 9250 package, i calibrated with an 8 or plain the axes, i got 15 different values. the one i wrote here are pretty right, from my experience. no big numbers like 30000.
i have no point plot yet.

wewillfindyou, 2017-01-28:
i did also the  Declination at Graz, Austria is 3 degrees 46 minutes on 2017-01-01 for my approach

wewillfindyou, 2017-01-28:
i think its maybe  the noise from the sensor  ?????
its about 20 out of the scale of 200. so its 10%  anyone better values?

kriswiner, 2017-01-28:
OK, it sounds like you are doing everything correctly. It might help to
check the point plot just to be sure. The mag calibration values could be
OK, the best way to check is to look for the response surfaces centered on
the origin with your calibration and reasonable spherical shape, as you
probably know. If you have a well-calibrated magnetometer and accelerometer
you should be getting stable yaw values if you are using the Madgwick
fusion filter as long as you are able to run the filter at ~1 kHz or higher
rate. What MCU are you using?

kriswiner, 2017-01-28:
What MPU9250 breakout are you using?

When you ay noise, do you mean the jitter in the magnetometer reading,
accelerometer reading, or what?

wewillfindyou, 2017-01-28:
Thank you for your commitment. 
-i have to make more tests with the calibration of the magnetometer.

my goal is to reset the magnetometer with every start, so i am always at the zero point, but not reseting the acc&gyro, if possible (because that would be calibrated values, to not having problems due to bad calibration). But as i see it, i have to reset the magnetometer everytime i start, so i have a zero point. is that true? is that the right approach?

I am running with 300 Hhz on a mbed arm m3. recommended would be a ticker with 1,6kH but i need about 3 micro second for a loop, so I am not getting the best results and problems with Yaw

wewillfindyou, 2017-01-28:
do i have to multiply the mx, my, mz with the magScale? recommended or obligated?

kriswiner, 2017-01-28:
If you just want quaternions and don't care about the magnetic field then
you can use raw counts, but the axes scale factors from the AK8963C fuze
ROM should be applied.

If you want to calibrate the mag very well once, you can simply use the
calculated bias offsets in the main loop thereafter without calibrating
every time, same with the accel.