kriswiner/MPU9250 #113: A few questions about the MPU9250 Arduino Code

TediumRemedy, 2017-02-02:
Hi Kris,

I am currently in the process of writing (or rather porting) an stm32 code for getting yaw/pitch/roll based from this board and I have few questions.
1) As I understand, the MPU9250SelfTest routine's values are not used, and it is not necessary have it to be able to calculate yaw/pitch/roll, and the call to MPU9250SelfTest can be skipped.
2) whatever is done in calibrateMPU9250 function can be done manually. So XG_OFFSET_H...ZG_OFFSET_L and XA_OFFSET_H... ZA_OFFSET_L registers can be loaded with zeros, and constant values can be subtracted/added to the final raw calculated values before submitting them to the madgwick/mahoney filters. These constants can be found by putting the board on a flat surface and adjusting the offsets so when gyro measurements are (0;0;0) and accelerometer's are (0;0;1). Is that correct?

Thanks

kriswiner, 2017-02-02:
yes, and yes. For the mag you will have to move the devcie in a figure
eight or however to sample as much of the 3D response surface as you can.
But you can similarly calculate the bias offsets and apply manually
(calibrate once).

TediumRemedy, 2017-02-04:
Hi Kris,
Thank you. That is what I did. 
I recorded mag data, added offsets to make it centered on a plot, also applied sensitivity factors, that I read from the fuse ROM of the mag, and here's what I eventually have with everything applied: http://imgur.com/a/0KkR0
This is mag data x,y,z drawn (green being x,y coordinates, red - x, z coordinates, yellow - y,z ones). I also tried to make sure max(x)+min(x)/2 are as close to zero as possible. The same I did for y and z.

The problem is when I supply this data into the magdwick algorithm, it behaves strangely. While Yaw seem to respond correctly, both pitch and roll also change the yaw (I rotate the board only around 1 "roll" axis, but both roll and yaw change). Do you have any idea why that might happen? I am currently delving into the madgwick algorithm, but I sense the problem is with the (wrong?) data I supply but I cant figure out why. For instance, the accelerometer correctly shows bottom direction with 1 on the corresponding axis and 0 on others when I rotate the board, leaving the board's side parallel to the ground.

Thanks for your help!

kriswiner, 2017-02-04:
You need to supplu the Madgwick algorithm with data in the correct order;
it expects NED ordering. Which direction is North? Note the accel/gyro have
a different z-axis orientation than the mag.

TediumRemedy, 2017-02-04:
Thanks, I missed that there are negative signs near some of the coordinates:
    MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);
Now it seems to be working properly!

Thank you for your great support of your code and your help!