kriswiner/MPU9250 #116: Magnetometer Calibration

baonguyen2604, 2017-02-20:
Hey Kris,

Thank you for your code. I'm trying to implement it on my IMU. However, I have some questions about the magnetometer calibration and NED reference.

I read through the code for the calibratioin on your seperate article. The instruction was wave the device in figure eight but I don't know which plane should I move the device in. Because each time I calibrate, it gives different values for magbias.

Also I'm not really sure how to determine the NED reference for the IMU. For example, gz is the rotation in the horizontal plane so how can we know if up or down is positive for gz. I hope you can elaborate more on this.

kriswiner, 2017-02-20:
"but I don't know which plane should I move the device"

 all of them, the idea is to fully sample the response surface.

Choose your North. Identify which sensor axes are pointed North. Then East and then Down. Then put these in the Madgwick or Mahony filters like so: (N, E, D, N, E, D, N, E, D) for accel, gyro, and mag, respectively.

baonguyen2604, 2017-02-21:
I tried to wave the IMU in figure in all three planes (with sample count as 384) and I got my bias as
magbias[0] = 87.25;
magbias[1] = 102.16;
magbias[2] = -269.54;
and
magScale[0] = 1.09;
magScale[1] = 1.08;
magScale[2] = 0.86;

I waved the IMU in figure in every plane and orientation. However my mx,my,mz graph is still not a sphere center at the origin

I figured out the NED reference according to the datasheet. I think for my chip it should be passed as (ax, ay, -az, gx, gy, -gz, my, mx, mz) if I take ax/gx/my as north.

So my only problem should be the calibration and I hope you can help me with that

kriswiner, 2017-02-21:
A sphere centered on the origin is the mag calibration goal. If not there, avergae the data for a longer time and move the sensor to sample more of the response surface.

baonguyen2604, 2017-02-22:
Also I want to ask do we need to add the environment magnetic field to magbias or is it already been accounted for during the calibration?

baonguyen2604, 2017-02-22:
[image]
This is my uncalibrated mx, my, mz
[image]
And this is after calibration. I know my data is not really well-distributed but is it good enough?

kriswiner, 2017-02-22:
Not if you want accurate orientation estimation.

Is it a sphere centered on the origin for all three axes? When the answer is yes, then it will be good enough for accurate orientation estimation.

If all you want is a general idea of where you are why bother calibrating at all?

baonguyen2604, 2017-02-23:
I was just confused because I read somewhere saying that you need to account for the environmental magnetic field. I just want the rotation in the horizontal plane (yaw angle) to control a robot.

With the calibrated magbias, I'm able to get pretty accurate measurements now but it takes quite a bit of time for the data to be stable (~ 5 seconds). How can I improve the stability while maintaining reasonable accuracy? Should I use Mahony or Madgwick filter and what should be free parameters be for optimal performance?

kriswiner, 2017-02-23:
" need to account for the environmental magnetic field"

Only if you want absolute orientation wrt True North.

It won't be stable without accurate calibration, and you need to make sure you pass the sensor data into the Madgwick or Mahnt filter properly. Either canl work well with the default parameters.

baonguyen2604, 2017-02-23:
I think my calibration is pretty good at the moment as the magnetometer values form a sphere centered at the origin
I also checked the datasheet for my IMU
[image]
And according to it, I pass the paremeters as (ax, ay, -az, gx, gy, -gz, my, mx, mz) if I take ax/gx/my as north.

kriswiner, 2017-02-24:
Looks right, you should be able to get ~5 degree heading accuracy if your MCU can do the fusion fast enough.