kriswiner/MPU9250 #117: Need Clarification on Orientation setting for filtering.

Prasadrs3, 2017-02-21:
hi Kris,
              we are new to IMU in our system we are using MPU-9250 connected to STM32L15x, We have given NED orientation as follows,
 MadgwickQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f,  my,  mx, mz);  
And in the example deltat was initialised with 0, we changed to 0.1, 
when deltat was 0 ,our quaternion, where 1,0,0,0 and after changing to 0.1 it was changing too much,
we need some assistance on how to set above two parameters.
[image]

kriswiner, 2017-02-21:
delta t is not set by the user, it is the iteration rate.

Try: MadgwickQuaternionUpdate(ax, ay, az, gxPI/180.0f, gyPI/180.0f, gz*PI/180.0f, my, mx, -mz);

Prasadrs3, 2017-02-22:
Thank you for your replay..
we made the corrections back to delta t as it was, and now the Quaternion are getting updated.
still the values are changed continuously mainly on the yaw, can you direct as how to do calibration and set initial bias,
note : we also tried with un-commenting the "Push gyro and accel biases to hardware registers".