kriswiner/MPU9250 #119: errors in readings of accelerometer

AkshayGupta94, 2017-02-28:
Hello Kris!
It's a fantastic code . I have applied some low pass filter and now I am able to get a 0,0,0 reading of accelerometer at rest.. Biasing removed the gravity component in z..
Now if I move my sensor in one direction.. Like along y or x or z. I am very particular that it moves in straight line.. Even then I see reading in all the axis.. Though reading about that axis has higher spikes.. Still other axis are showing reading too big to be ignored by my filter.. 
Can you suggest any solution? Thanks in advance!

kriswiner, 2017-02-28:
Not sure how to help you here. You might be better off doing the full 9DoF fusion and then extracting linear acceleration if that is what you want.

AkshayGupta94, 2017-02-28:
Thank you sir!

closed 2017-02-28 by AkshayGupta94