kriswiner/MPU9250 #120: Calibrate Accelerometer

Stips123, 2017-03-01:
Dear Kris,

thank you so much for this beautiful library! 
Anyhow, I have a small question. I only need acceleration data in my project, but I don't know how to calibrate the IMU.
my Code so far:
```
#include <Wire.h>
#include <MPU9250.h>

MPU9250 myIMU1;

void setup() {
	Serial.begin(115200);
	Wire.begin();

	myIMU1.initMPU9250();

	myIMU1.writeByte(myIMU1.sensorAdresse, SMPLRT_DIV, 0);

	myIMU1.calibrateMPU9250(myIMU1.gyroBias, myIMU1.accelBias);

	myIMU1.getAres();
	myIMU1.getGres();
}

void loop() {
	myIMU1.readAccelData(myIMU1.accelCount);
	myIMU1.ax = (float)myIMU1.accelCount[0] * myIMU1.aRes; // -myIMU1.accelBias[0];
	myIMU1.ay = (float)myIMU1.accelCount[1] * myIMU1.aRes; // -myIMU1.accelBias[1];
	myIMU1.az = (float)myIMU1.accelCount[2] * myIMU1.aRes; // -myIMU1.accelBias[2];

	Serial.print(myIMU1.ax);
	Serial.print('\t');
	Serial.print(myIMU1.ay);
	Serial.print('\t');
	Serial.println(myIMU1.az);
}
```

No the problem is that sqrt(ax^2 + ay^2 + az^2) is not 1 when I don't move the IMU. Can you please tell me, how to calibrate the Sensor? Thank you!!
Steffen

kriswiner, 2017-03-01:
Check here for an accel/gyro calibration function:

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

and adapt as needed.

Stips123, 2017-03-01:
Thanks for the fast reply! I really appreciate it.
Do you mean, that I have to adapt the code for the MPU9250.h library, or do you advice to use the library in your link (cause it's more up-to-date)?
btw you definitely get the beer if we'll ever meet =)

kriswiner, 2017-03-02:
I don't use Sparkfun's library even though they based it one one of my sketches. They chose a really old sketch for some reason that had some errors and omissions. The sketch I linked to is more up to date, has accel/gyro as well as mag calibration and I recommend you use it.

Well, virtual beer is OK too!

Stips123, 2017-03-02:
thanks for the link!
is the calibration permanently stored in the MPU, or do I have to calibrate every single time I (re-)start the Arduino? Or is it necessary to place the IMU on flat ground, at all?
I'm asking because I can't place the IMU on solid ground everytime in my application (it's rigidly attached to sth).
Cheers! 
[image]

kriswiner, 2017-03-02:
Thanks for the beer. You can calculate the offset bias calibrations and store them in your sketch to avoid calibrating each time.

For gyro and accel the device should be motionless and flat. Fir mag calibration, you have to move the device around to sample all of the 3D response surface. If you print out the calibration values on the serial monitor, you can just subtract these from your data stream on subsequent startup without having to calibrate anymore unless the environmental conditions (location and temperature, mostly) change.

Stips123, 2017-03-03:
hey,
thanks for your answer. Something strange is happening here: The x- and y-axes are working fine. Sometimes the z-axis is working correctly which means it is +1 g or -1 g, if the +z or -z-axis is parallel to gravity, respectively. But sometimes, the output is 0 g and -2 g at rest, and I don't know why (as you see in the graphs above). Have you ever observed such a behaviour?
Thanks!

kriswiner, 2017-03-04:
Yes, there is a bug in the calibration code. The calibration has to be done with the sensor facing up in order to get the correct  calibration.

Stips123, 2017-03-04:
lol, that's good to know.

NickMortimer, 2017-07-21:
Hi I'm looking through your excellent calibration information. I'm trying out the sketch from above. I have installed the <i2c_t3.h> library but I'm getting this compile errors. Mainly round wire not being defined. I see that the wire.h is commented out but un-commenting still gives errors. Am I missing something?

monstah, 2017-07-21:
@NickMortimer can you post the exact error message after uncommenting the header file?

kriswiner, 2017-07-21:
i2c_t3.h is only for the Teensy 3.X MCU. Is that what you are using?

NickMortimer, 2017-07-25:
Ah sorry for the delay I have been what we say here in Australia out in the bush! I'm using Pro mini but I'm ordering Teensy right now!

NickMortimer, 2017-07-25:
So I have un-commented the wire and taken out i2c_t3.h and get the following, seems to be a problem with I2C_MASTER not being defined. could be me?

C:\Users\mor582\Documents\Arduino\myIMU\myIMU.ino: In function 'void setup()':

myIMU:312: error: 'I2C_MASTER' was not declared in this scope

   Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

              ^

myIMU:312: error: 'I2C_PINS_16_17' was not declared in this scope

   Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

                                ^

myIMU:312: error: 'I2C_PULLUP_EXT' was not declared in this scope

   Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

                                                ^

myIMU:312: error: 'I2C_RATE_400' was not declared in this scope

   Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

                                                                ^

C:\Users\mor582\Documents\Arduino\myIMU\myIMU.ino: In function 'void loop()':

myIMU:521: error: 'MadgwickQuaternionUpdate' was not declared in this scope

     MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);

                                                                                                    ^

C:\Users\mor582\Documents\Arduino\myIMU\myIMU.ino: In function 'void MS5637PromRead(uint16_t*)':

myIMU:1219: error: 'I2C_NOSTOP' was not declared in this scope

           Wire.endTransmission(I2C_NOSTOP);        // Send the Tx buffer, but send a restart to keep connection alive

                                ^

C:\Users\mor582\Documents\Arduino\myIMU\myIMU.ino: In function 'uint32_t MS5637Read(uint8_t, uint8_t)':

myIMU:1233: error: 'I2C_NOSTOP' was not declared in this scope

         Wire.endTransmission(I2C_NOSTOP);        // Send the Tx buffer, but send a restart to keep connection alive

                              ^

C:\Users\mor582\Documents\Arduino\myIMU\myIMU.ino: In function 'uint8_t readByte(uint8_t, uint8_t)':

myIMU:1340: error: 'I2C_NOSTOP' was not declared in this scope

   Wire.endTransmission(I2C_NOSTOP);        // Send the Tx buffer, but send a restart to keep connection alive

                        ^

C:\Users\mor582\Documents\Arduino\myIMU\myIMU.ino: In function 'void readBytes(uint8_t, uint8_t, uint8_t, uint8_t*)':

myIMU:1352: error: 'I2C_NOSTOP' was not declared in this scope

   Wire.endTransmission(I2C_NOSTOP);  // Send the Tx buffer, but send a restart to keep connection alive

                        ^

exit status 1
'I2C_MASTER' was not declared in this scope

kriswiner, 2017-07-25:
You'll need i2c_t3.h if you use i2c_t3 function calls. This will only work
on a Teensy 3.X though.

Bobur23, 2017-08-16:
Hello Kris! I have been wondering how to calibrate MPU9250 on Unity. I am using Serial, Madgwick source codes

kriswiner, 2017-08-16:
Just use the calibration functions from one of my sketches.

NickMortimer, 2017-10-02:
Hi Kris, Sorry to bother you again I'm having real problems getting my calibration to work. I have a 9 DOF using on the inside of a large and very magnetic statue of the Moon. The 9 DOF is on the lighting system inside the moon.

I have attached some output. I would really value your opinion!

It's a standard ax,ay,az,gx,gy,gz,hx,hy,hz format

Thanks
[moon2.zip](https://github.com/kriswiner/MPU9250/files/1349147/moon2.zip)

Nick

NickMortimer, 2017-10-02:
Ok I think my problem is that the statue is fixed and my light assembly moves that is the worst possible combination

"It is also important to recognize that effective compensation of hard- and soft-iron distortions is dependent upon the distorting material(s) rotating/moving with the sensor. An example would be mounting the sensor in an aircraft; any materials that are part of the aircraft that exhibit a distorting effect would move as the aircraft and mounted sensor move, and it would generally be possible to compensate for the associated hard- and soft-iron effects. In contrast, it is much more difficult—if not impossible—to compensate for distorting effects exhibited by material external to the aircraft/sensor platform. Thus, it is important to understand not only how compensation may be applied, but also to recognize those conditions under which effective compensation techniques are not possible."

So that leaves me with a bit of a problem!

Nick

kriswiner, 2017-10-02:
I think you might want to try this then:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/

we use for small quadcopters with brushed dc motors pulling variable 0 - 2
Amps each so there is a stron fixed and variable magnetic field, yet we
maintain accurate orientation estimation.

NickMortimer, 2017-10-02:
[image]

Thanks once again your input. Here's a picture of the structure, I don't need to be too accurate but as you can see there is a lot of steel on the supports, the light system is inside the globe!

kriswiner, 2017-10-02:
Can you calibrate the mag inside the globe? There is auto mag calibration
on the EM7180 so might not even be an issue, but if you can manually
calibrate inside it would be easier. No guarantees, but if you can't do an
adequate job with the MPU9250+EM7180 them I would suppose an inexpensive
IMU solution is hopeless.

NickMortimer, 2017-10-02:
Yep I was not involved with the design, I wanted shaft encoders on the axis but it was built before I was on board! Yep The cal data I posted is the sensor mounted on the light system. Checking with a standard compass you can get needle reversals at points close to the bottom of the structure!

NickMortimer, 2017-10-04:
[image]
So this is the problem. This was a vertical circle!

kriswiner, 2017-10-04:
Yes, but is a fixed field so can be calibrated out

NickMortimer, 2017-10-05:
No at this time I can fix the calibration if I'm only interested in one plain as soon as I change the orientation of the sensor and sweep it a gain I get a completely different result mainly because the think causing the deviation is in a fixed location and thus it's distance to the sensor is varying. I'm going to tell them to get some shaft encoders! Thanks once again for your replies and information!

jenextech, 2018-01-09:
Hello kris,

      I have used your code with mpu9250 and testing data.
I had some doubt that how can verify data.So,i have verify with android application.

I have comparing data with SENSOR KINETICS (app),and its very different readings.

So,what should be issue?

kriswiner, 2018-01-09:
One or both are wrong?

Did you calibrate the sensors?

Are you using an MCU to give a fusion rate > 1 kHz?

Are you feeding the Madwick filter the data in the proper order?

jenextech, 2018-01-10:
I dont know which one is wrong but one of the two is wrong.
I have used your code as it is,no any additional configuration has done.

So,how can i verify raw data as well as all result of code.(yaw,pitch,roll 9 axis data with particular units)?

Thanks in advanced.

kriswiner, 2018-01-10:
You didn;t answer any of my questions, hard to help without this info.

jenextech, 2018-01-11:
Sorry if i had not respond your question.But i told you.

As you asked me about which one is wrong?
I told that one of two platform can wrong.

I want to verify IMU data.So,from android application we can measure 9-axis data.
And data which we getting from IMU and APPLICATION.Both are different.

So,how can i verify it?.

nanoTitan, 2018-02-27:
Hi Kris,

I went through your accel/gyro calibration with the device at rest, and the magnetometer calibration while performing figure eights. If I rotate the device parallel to the ground, the yaw seems to work well. However, I'm seeing that a change in the roll of the device also causes a large change in the yaw. 

Any ideas of what might be causing this or where I can look to fix it?

Thanks so much for putting this library together! I've modified it from your Stm32 Mbed code to use Stm's HAL library for the Stm32f401. If you would like, I can provide that code for you.

kriswiner, 2018-02-27:
I will assume you verified that your sensors are properly calibrated.

How are you passing the sensor data tot he Madgiwck or Mahony fusion
filter? NED?

nanoTitan, 2018-02-27:
Yes, I can verify my magnetometer data with a 2-D plot like you have in your magnetometer calibration article. I'm not sure of how I would verify accel/gyro data though after running the calibration on those. Any tips on that would be great. I do see that the accel/gyro trims are within roughly 1-2% of factory settings.

Yes, I believe the data is provided in the N.E.D. orientation. I'm using your example code from here:
https://os.mbed.com/users/onehorse/code/MPU9250AHRS/file/4e59a37182df/main.cpp/

nanoTitan, 2018-02-28:
Ok, I've remapped those ax, ay, az, etc to the order that the filter expects. It looks like it is working much better now. Thanks for that!

For the accel calibration, I'm getting 0.02, 0.01, 1.00 which looks about right. The gyro is giving me values around 0.2, 0.3, 0.01. Are those close enough to zero consider accurate? I'm trying to get this within the 2 degrees of accuracy.

kriswiner, 2018-02-28:
You might get 4 degrees rms  heading accuracy. In part it depends on the
fusion rate your MCU is capable of. What are you using again?

nanoTitan, 2018-02-28:
I'm using a stm32f401 running at 84Mhz. It looks like the Teensy runs at 180Mhz, and I'm guessing I need something close to that to get greater precision.

kriswiner, 2018-02-28:
F401 has an FPU so you will be able to get fusion rates >20 kHz which is
more than enough.

aabdulkadar, 2018-06-20:
I have problem with my mpu 9250 .I use this code( https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino) to calibrate .But the values are not changing and I used to get these in serial monitor 

Gyro temperature is 30.8 degrees C
Digital temperature value = 20.00 C
Digital temperature value = 68.00 F
Digital pressure value = 0.00 mbar
Altitude = 145366.45 feet
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 6978.08 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00

Please help me.I am new

aabdulkadar, 2018-06-20:
always zero irrespective of the movements !!

kriswiner, 2018-06-20:
Are you using the data ready interrupt?

aabdulkadar, 2018-06-21:
`
[MPU9250_calib.docx](https://github.com/kriswiner/MPU9250/files/2122137/MPU9250_calib.docx)
`This is the code what I am using

aabdulkadar, 2018-06-21:
Thank you for the fast response.

I am reading like what I written below

/ If intPin goes high, all data registers have new data
   if(newData == true) {  // On interrupt, read data
     newData = false;  // reset newData flag
     readMPU9250Data(MPU9250Data); // INT cleared on any read
 readAccelData(accelCount);  // Read the x/y/z adc values
    
    // Now we'll calculate the accleration value into actual g's
    ax = (float)MPU9250Data[0]*aRes - accelBias[0];  // get actual g value, this depends on scale being set
    ay = (float)MPU9250Data[1]*aRes - accelBias[1];   
    az = (float)MPU9250Data[2]*aRes - accelBias[2];  
   
  readGyroData(gyroCount);  // Read the x/y/z adc values

    // Calculate the gyro value into actual degrees per second
    gx = (float)MPU9250Data[4]*gRes;  // get actual gyro value, this depends on scale being set
    gy = (float)MPU9250Data[5]*gRes;  
    gz = (float)MPU9250Data[6]*gRes;   
  
    readMagData(magCount);  // Read the x/y/z adc values
   
    // Calculate the magnetometer values in milliGauss
    // Include factory calibration per data sheet and user environmental corrections
    if(newMagData == true) {
      newMagData = false; // reset newMagData flag
      mx = (float)magCount[0]*mRes*magCalibration[0] - magBias[0];  // get actual magnetometer value, this depends on scale being set
      my = (float)magCount[1]*mRes*magCalibration[1] - magBias[1];  
      mz = (float)magCount[2]*mRes*magCalibration[2] - magBias[2];  
      mx *= magScale[0];
      my *= magScale[1];
      mz *= magScale[2]; 
    } 
  }

kriswiner, 2018-06-21:
> here is my reading part

" int intPin = 8;"

Are you using interrupt pin on Teenst pin 8?

aabdulkadar, 2018-06-21:
yes..I checked with 2 and 12 also ..No change

aabdulkadar, 2018-06-21:
/* MPU9250_MS5637_t3 Basic Example Code
 by: Kris Winer
 date: April 1, 2014
 license: Beerware - Use this code however you'd like. If you 
 find it useful you can buy me a beer some time.
 
 Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. Added display functions to 
 allow display to on breadboard monitor. Addition of 9 DoF sensor fusion using open source Madgwick and 
 Mahony filter algorithms. Sketch runs on the 3.3 V 8 MHz Pro Mini and the Teensy 3.1.
 
 This sketch is intended specifically for the MPU9250+MS5637 Add-on shield for the Teensy 3.1.
 It uses SDA/SCL on pins 17/16, respectively, and it uses the Teensy 3.1-specific Wire library i2c_t3.h.
 The MS5637 is a simple but high resolution pressure sensor, which can be used in its high resolution
 mode but with power consumption of 20 microAmp, or in a lower resolution mode with power consumption of
 only 1 microAmp. The choice will depend on the application.
 
 SDA and SCL should have external pull-up resistors (to 3.3V).
 4K7 resistors are on the MPU9250+MS5637 breakout board.
 
 Hardware setup:
 MPU9250 Breakout --------- Arduino
 VDD ---------------------- 3.3V
 VDDI --------------------- 3.3V
 SDA ----------------------- A4
 SCL ----------------------- A5
 GND ---------------------- GND
 
 Note: The MPU9250 is an I2C sensor and uses the Arduino Wire library. 
 Because the sensor is not 5V tolerant, we are using a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1.
 We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.
 We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.
 */
#include "Wire.h"   
//#include <i2c_t3.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_PCD8544.h>

// Using NOKIA 5110 monochrome 84 x 48 pixel display
// pin 7 - Serial clock out (SCLK)
// pin 6 - Serial data out (DIN)
// pin 5 - Data/Command select (D/C)
// pin 3 - LCD chip select (SCE)
// pin 4 - LCD reset (RST)
Adafruit_PCD8544 display = Adafruit_PCD8544(7, 6, 5, 3, 4);

// See MS5637-02BA03 Low Voltage Barometric Pressure Sensor Data Sheet
#define MS5637_RESET      0x1E
#define MS5637_CONVERT_D1 0x40
#define MS5637_CONVERT_D2 0x50
#define MS5637_ADC_READ   0x00

// See also MPU-9250 Register Map and Descriptions, Revision 4.0, RM-MPU-9250A-00, Rev. 1.4, 9/9/2013 for registers not listed in 
// above document; the MPU9250 and MPU9150 are virtually identical but the latter has a different register map
//
//Magnetometer Registers
#define AK8963_ADDRESS   0x0C
#define AK8963_WHO_AM_I  0x00 // should return 0x48
#define AK8963_INFO      0x01
#define AK8963_ST1       0x02  // data ready status bit 0
#define AK8963_XOUT_L   0x03  // data
#define AK8963_XOUT_H  0x04
#define AK8963_YOUT_L  0x05
#define AK8963_YOUT_H  0x06
#define AK8963_ZOUT_L  0x07
#define AK8963_ZOUT_H  0x08
#define AK8963_ST2       0x09  // Data overflow bit 3 and data read error status bit 2
#define AK8963_CNTL      0x0A  // Power down (0000), single-measurement (0001), self-test (1000) and Fuse ROM (1111) modes on bits 3:0
#define AK8963_ASTC      0x0C  // Self test control
[... 1449 lines cut ...]
            }

            // Apply feedback terms
            gx = gx + Kp * ex + Ki * eInt[0];
            gy = gy + Kp * ey + Ki * eInt[1];
            gz = gz + Kp * ez + Ki * eInt[2];

            // Integrate rate of change of quaternion
            pa = q2;
            pb = q3;
            pc = q4;
            q1 = q1 + (-q2 * gx - q3 * gy - q4 * gz) * (0.5f * deltat);
            q2 = pa + (q1 * gx + pb * gz - pc * gy) * (0.5f * deltat);
            q3 = pb + (q1 * gy - pa * gz + pc * gx) * (0.5f * deltat);
            q4 = pc + (q1 * gz + pa * gy - pb * gx) * (0.5f * deltat);

            // Normalise quaternion
            norm = sqrt(q1 * q1 + q2 * q2 + q3 * q3 + q4 * q4);
            norm = 1.0f / norm;
            q[0] = q1 * norm;
            q[1] = q2 * norm;
            q[2] = q3 * norm;
            q[3] = q4 * norm;
 
        }

aabdulkadar, 2018-06-21:
I changed intpin to 3 and it works 

Digital temperature value = 20.00 C
Digital temperature value = 68.00 F
Digital pressure value = 0.00 mbar
Altitude = 145366.45 feet
Yaw, Pitch, Roll: 280.96, -24.25, -144.30
Grav_x, Grav_y, Grav_z: 532.09, -410.65, -740.44 mg
Lin_ax, Lin_ay, Lin_az: -480.34, 1387.70, -1310.83 mg
rate = 96.03 Hz

but even it is at rest,acceleration values are not zero..?
Thank you for your patience