kriswiner/MPU9250 #121: 9255 on STM32 F103RB

esePuente, 2017-03-02:
Hello Kris, I join choir here great work on keeping it up with the dev and support.

I've been trying to spot my issue somewhere along these threads, but can't seem to find someone that had already run into this. Hopefully, you can think of something ...

So I am running one of those 9250's from amazon that turned out to be a 9255 (whoami == 0x73).
I hooked this up to my SCA,SDL,3.3 and Gnd pins on a Nucleo F103 and then tried to use your code for the F401, I remember it running pretty much same as it was fro the F401 (I don;t see anything evident I should change).

Anyway, I keep getting really messy accelerometer data. for both x and z, curiously not terribly bad for y.
I've tried all of the low pass filter configurations and sample rate dividers, but none of them seem to do the trick. So comparing this data to what I was getting from a 9150 definitely seem odd (same STM, same setup, etc.). So I went ahead and got another board, maybe it just was a bad one. Well, I either got another bad one, or I am doing something clearly wrong and can't really find what it is... Have you heard of any issue similar to this?

I'm adding some output here, maybe you can spot something.

CPU SystemCoreClock is 72000000 Hz
I AM 0x73

I SHOULD BE 0x73

MPU9250 is online...

CPU SystemCoreClock is 72000000 Hz
I AM 0x73

I SHOULD BE 0x73

MPU9250 is online...

x gyro bias = -2.007634

y gyro bias = 0.358779

z gyro bias = -0.419847

x accel bias = 0.107544

y accel bias = 0.003601

z accel bias = -0.177002

MPU9250 initialized for active data mode....

AK8963 initialized for active data mode....

Accelerometer full-scale range = 2.000000  g

Gyroscope full-scale range = 250.000000  deg/s

Magnetometer resolution = 16  bits

Magnetometer ODR = 100 Hz

Accelerometer sensitivity is 16384.000000 LSB/g 

Gyroscope sensitivity is 131.072006 LSB/deg/s 

Magnetometer sensitivity is 0.666938 LSB/G 

[image]

Thanks a lot.

esePuente, 2017-03-13:
Thanks for the reply Kris, good feedback.
But, I still think I'm missing something... I am pretty new to STM's so I think there's something I still should do. So, when I hooked up my signal analyzer, as soon as I did that I noticed the data started making sense. Accel values when straight to what they should be... So my first though was the pull ups, I need to add some pull up resistors, but I noticed there are some 10K already in there. Anyway, I put another 10K on top of those (well, in parallel I guess) but nothing, still the noisy signal. Tried an 8K instead, nothing again. So I really don't know if it's a pull up thing.
As soon as I hook up the analyzer it gets fixed, oh and one thing. I also noticed the sample rate goes from 700 ish to 1800 ish... so definitely a timing thing. But what do you think could be causing it? Why would hooking up a signal analyzer could help?

Again any help would be greatly appreciated.

kriswiner, 2017-03-13:
No idea, sounds like a poor connection. Pullup resistance has nothing to do with it.