kriswiner/MPU9250 #122: MPU9250 self-test and calibration questions

aclex, 2017-03-07:
Hello, Kris! At first, thank you for so comprehensive code to interact with IMU and articles on the background!

I'm trying to port some pieces of "MPU9250_MS5637_AHRS_t3.ino" to plain Linux (on Raspberry Pi), connecting via SPI with spidev, I'm actually half-way with it, and have a couple of questions on the difficulties I faced.

1. Is the formula at the very end of `MPU9250SelfTest()` function (`100.0*((float)(aSTAvg[i] - aAvg[i]))/factoryTrim[i] - 100.`) supposed to return small numbers of percent of these deviations or something around 100%? I have the latter case when performing self-test, all the values are around 100%, and I saw similar code both with subtraction/sum of 100 and without it (e.g. Sparkfun's https://github.com/sparkfun/MPU-9250_Breakout/blob/master/Libraries/Arduino/src/MPU9250.cpp).

2. I'm trying to implement the same accel and gyro calibration routine, as in `accelgyrocalMPU9250()`, doing almost literally the same, but can't get MPU to fill FIFO buffer (line 958), values returned from `FIFO_COUNTH` (and `FIFO_COUNTL`) are all zeros. Maybe, you happened to see something similar or have a hint on what it can be?

kriswiner, 2017-03-07:
>    Libraries/Arduino/src/MPU9250.cpp

On the first problem, the self test documentation specifies that they use
only the LSByte for the self test, so the self test is incorrectly coded.
See attached.

On the second problem, I have never seen it, You might not be correclty
configuring the FIFO? I don't know it is pretty straightforward.

Kris

aclex, 2017-03-07:
Thank you for clarification on the self test.
As for second, yes, I guess, there's some incorrectness in the initialization of using FIFO, but can't find, what exactly. If I understand it right, it should fill the FIFO right after `USER_CTRL` and `FIFO_EN` are written, as it is done in lines 952, 953. I'll try to find the cause of it.
Thanks for your help!

aclex, 2017-03-08:
To keep it resolved here, in case of someone would search for similar problem, in my case the issue was with wrong implementation of writing operation to MPU register through SPI, so FIFO couldn't be properly configured. Another issue could be timing — the delays in the sketch are enough, but if you skip some of them, there could be lack of samples in the FIFO.

Unfortunately, now I suffer the problems with self-test again, the accel values for Y and Z axes are normal, but X is about 700% of factory values, the same issue for gyro. Would try to solve it as well, maybe some another error in my porting.

drankinatty, 2021-08-18:
I happened to have gone through this recently. The Self-Test documentation is ambiguous on several points, but I managed to sort through the issues. When taking the lower 8-bits of the `ST_OTP` calculation using the `ST_Code` value read from the register on-chip, you take the lower 8-bits of the *Integer Parr* of the floating-point value, not the lower 8-bits of the mantissa. Next, the `FS` value used in the `2 ^ FS` part of the calculation is the **FS Setting Value** of `0`, (not `250` for the gyro , `2 ^ 250` is `1.8e+75` way, way too big). 

With that the pass/fail criteria correlate to exactly what I see as axis values (I have had a number of bad MPU9250 chips lately -- which prompted the dive)

If your interpretation of the `ST_OTP` calculation and value is different, I would value your thoughts.

kriswiner, 2021-08-18:
Been a while since I fussed with all of this. The MPU9250 has been out of
production for 18 months now and I haven;t used this sensor for anything
for a bit longer than that. So not sure how relevant trying to parse
Invensense's unnecessarily convoluted self test procedure might be. I will
note that many of the self-test implementations in this repo are simply
wrong, but nothing close to a 1e75 error. I believe the later ones are
correct but fundamentally this does not matter. The proper test of whether
the MPU9250 ICs you have are sound or not is whether they can be
configured, calibrated, and then produce sensible data.