kriswiner/MPU9250 #123: No response from magnetometer

MaSkr09, 2017-03-08:
Hi Kris
I’m working on a open source tracking device “DroneID”, for trafic management of drones in Denmark [DroneID_report](http://droneid.dk/download/DroneID_Report_October_2016.pdf). I have som trouble with a MPU-9250 sensor. I’ve tried to implement your solution, but went back to basics just to get started. The problem is that I don’t get any response from the magnetometer. There is no problem getting data from the other sensors in the chip, but when i get to the last line of the code below, I get an error.

Inspired by [lucidarme](http://www.lucidarme.me/?p=5057) 

  bool error = false;
  // Set accelerometers low pass filter at 5Hz
  data_test = 0x06;
  error |= i2c_write(0x68,29,1,&data_test);
  // Set gyroscope low pass filter at 5Hz
  error |= i2c_write(0x68,26,1,&data_test);
  // Configure gyroscope range
  data_test = 0x10;
  error |= i2c_write(0x68,27,1,&data_test);
  // Configure accelerometers range
  data_test = 0x08;
  error |= i2c_write(0x68,28,1,&data_test);
  // Set by pass mode for the magnetometers
  data_test = 0x02;
  error |= i2c_write(0x68,0x37,1,&data_test);
  
  // Request continuous magnetometer measurements in 16 bits
  data_test = 0x01
  error |= i2c_write(0x0C,0x0A,1,&data_test);

Do you have any idea of what could be wrong? 

Kind regards
Martin Skriver

kriswiner, 2017-03-08:
0x01 is not a valie entry, please see the data sheet.

```
void initAK8963(float * destination)
{
  // First extract the factory calibration for each magnetometer axis
  uint8_t rawData[3];  // x/y/z gyro calibration data stored here
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00); // Power down magnetometer  
  delay(10);
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x0F); // Enter Fuse ROM access mode
  delay(10);
  readBytes(AK8963_ADDRESS, AK8963_ASAX, 3, &rawData[0]);  // Read the x-, y-, and z-axis calibration values
  destination[0] =  (float)(rawData[0] - 128)/256. + 1.;   // Return x-axis sensitivity adjustment values, etc.
  destination[1] =  (float)(rawData[1] - 128)/256. + 1.;  
  destination[2] =  (float)(rawData[2] - 128)/256. + 1.; 
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00); // Power down magnetometer  
  delay(10);
  // Configure the magnetometer for continuous read and highest resolution
  // set Mscale bit 4 to 1 (0) to enable 16 (14) bit resolution in CNTL register,
  // and enable continuous mode data acquisition Mmode (bits [3:0]), 0010 for 8 Hz and 0110 for 100 Hz sample rates
  writeByte(AK8963_ADDRESS, AK8963_CNTL, Mscale << 4 | Mmode); // Set magnetometer data resolution and sample ODR
  delay(10);
}
```

MaSkr09, 2017-03-08:
Thank you for your fast respond. 0x01 was my panic attempt to try something new :) I tried implementing the code, but it didn’t change anything. It seems like there is no node with addr 0x0C on the bus. 
Do you think it can it be a hardware problem? It is just strange since there is no problem getting data from node addr 0x68.

Kind regards

kriswiner, 2017-03-08:
If you set bypass by writing 0x02 to the INT_CFG register this should allow
discovery of the AK8963C by an I2C scanner.

Why don't you try a known good Arduino sketch?

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

MaSkr09, 2017-03-08:
Thanks Kris 
I'll try working with it tomorrow and get back with the result.

MaSkr09, 2017-03-10:
Still no response fra AK8963C. 
I attached the MPU9250 part of the diagram that we used in the DroneID. I have seen diagrams of the sensor where pin 1 is open. We added 3 V to the pin. Is that correct?

[image]

Kind regards

kriswiner, 2017-03-10:
The schematic looks OK except I would no connect the pad to anything and
would have no copper on any layer beneath the package. I doubt this is
causing the problem with the mag unless the pad is bridging one of the pins.

MaSkr09, 2017-03-21:
I had a problem with the I2C but it is working now.
Thank you very much, I owe you a beer ;)

closed 2017-03-21 by MaSkr09

vahidyazdnian13, 2021-11-09:
> 
> 
> I had a problem with the I2C but it is working now. Thank you very much, I owe you a beer ;)

hello ! i have a very similar problem to what you had an my I2C doesn't connect to 0x0C address to get data can you help me out and tell me what did you do to solve this please!

ethem4161, 2022-02-14:
> > I2C ile ilgili bir sorunum vardı ama şimdi çalışıyor. Çok teşekkür ederim, sana bir bira borcum var ;)
> 
> Merhaba ! sahip olduğunuza çok benzer bir sorunum var, I2C'm veri almak için 0x0C adresine bağlanmıyor, bana yardım edebilir ve bunu çözmek için ne yaptığınızı söyleyebilir misiniz lütfen!

+1

kriswiner, 2022-02-14:
English please...

kriswiner, 2022-05-25:
> I'm using a Raspberry Pi with a MPU9250, and I'm having similar problem.
> Everything works fine (if I add in python a try/except block) except for
> the magnetometer. Im running the command i2cdetect -y 1 to scan the
> addresses. The 0x68 of the accel/gyro is appearing, but the 0x0c is not
> showing up. Could someone help me? I know this is not talking about python,
> but...

You have to enable I2C pass through to allow the magnetometer to be
addressed by the host MCU.