kriswiner/MPU9250 #124: EMSENSR-9250 no longer sold

grep137, 2017-03-10:
I spent the last couple of hours searching for the EMSENSR-9250; unfortunately it looks like Embedded Masters may have gone out of business.  Does anyone have any information about Embedded Masters, or can anyone recommended a good replacement?

kriswiner, 2017-03-10:
Here it is under a different name:

https://www.cdiweb.com/ProductDetail/NWMOTMPU9250-notWired-co/605716/

And of course there are there:

https://www.tindie.com/products/onehorse/mpu9250-teensy-3x-add-on-shields/

grep137, 2017-03-10:
Thank you very much!  When I got the response from CDI saying, "We are no longer franchised for this line.  Sorry," I thought that Embedded Masters was no longer in existence.

Also, thank you for the tindie link.