kriswiner/MPU9250 #125: Quaternion Filter Drift

Jaqos, 2017-03-12:
Hello Kris,
Thanks for posting your code, examples and wikis online, they are awesome!

I am getting a small drift when the MPU9250 is put at rest after movements, it looks like a small 5-10° converge to magnetometer.

The filter converges from initial conditions just like the black cube on [Open source IMU and AHRS algorithms](http://x-io.co.uk/open-source-imu-and-ahrs-algorithms/)
In the video when the sensor is placed flat on the desk, there is no drift, where i do see drift when i place my sensor flat on the desk.

My setup:
Original code: [MPU9250BasicAHRS](https://github.com/sparkfun/MPU-9250_Breakout/tree/master/Libraries/Arduino)
Sketch uploaded to Arduino: [MPU9250_Jaqos_GitHub.zip](https://github.com/kriswiner/MPU-9250/files/836352/MPU9250_Jaqos_GitHub.zip)
Board: [Arduino Pro Mini 8mhz](https://www.sparkfun.com/products/11114)
Sensor: [Sparkfun MPU9250 breakout](https://www.sparkfun.com/products/13762)

I have learned a lot from reading the issues and questions on this GitHub, but haven't found anyone with the same results as me.

The filter is updated 5-6 times every new sensor data, Arduino sends the Quaternion to my computer which is applied visually to a cube in Unity.

The drifting effect is different with different NED orders (north, east, down).
By trial and error, the best order i found was: (ay, az, ax, gy, gz, gx, mx, -mz, my)
[image]

Drift graph: (you can see the blue line slowly moves up when sensor is at rest):
[image]

The mag calibration looks good:
[image]

A video showing the drift:
[cube drift](https://drive.google.com/file/d/0B4G1yoIaeMUFaEUtRTR4Qk5lMm8/view?usp=sharing)

The Questions:
1. Did you get good results like the x-io video, what board, what sensor?
2. How do you think i can eliminate the drift?
3. What is the correct NED order for the Sparkfun MPU9250 breakout?

Thanks for your time ;)

[image]

kriswiner, 2017-03-12:
Seems like you have everything set up properly, I think the problem is your
MCU speed. It is likely that 4 - 5 times for iteration isn't sufficient. So
I would recommned you step up to a Teensy 3.2 or a Butterfly.(
https://www.tindie.com/products/TleraCorp/butterfly-stm32l433-development-board/?pt=full_prod_search).
With the latter you can easily run the fusion at > 1 kHz, which will reduce
or eliminate the drift you are seeing.

kriswiner, 2017-03-12:
> development-board/?pt=full_prod_search). With the latter you can easily

One other thing, Sparkfun routed traces and vias undeneathe the MPU9250
package, a no-no. I have told them about this but I guess they have to sell
off their inventory before they can redesign it. Anyway, here are properly
designed MPU9250 breakouts:

https://www.tindie.com/products/onehorse/mpu9250-teensy-3x-add-on-shields/?pt=full_prod_search

Kris

Jaqos, 2017-03-12:
Thanks, i have changed the Arduino sketch to send raw accelerometer, gyro, magnetometer data to computer. With the Madgwick filter iteration running much faster on my computer, and the drifting looks to be a little smaller.

At some orientations it seems to be noisy, maybe it's the MPU9250 breakout. Should i use a filter to smooth gyro or mag noise?

kriswiner, 2017-03-12:
I usually run the accel and gyro at 200 - 400 Hz and use a 40 Hz low pass
filter for smooting.

Jaqos, 2017-03-13:
Thanks, I will try adding a low pass filter. 

Maybe the drift is because of the NED order, which order is the best for the MPU-9250?

How good are results with the Teensy 3.2 & MPU9250 Teensy 3.X add-on shield?

kriswiner, 2017-03-13:
The order you are using is correct.

I have done some testing just using the Madgwick filter and with the MPU9250 properly calibrated you should be able to hold +/-4 degree accuracy in heading and +/-1 accuracy in pitch and roll. I usually do not see this kind of drift.

I have always had good results with the MPU9250 using either a Teensy or a Butterfly/Ladybug.

Jaqos, 2017-05-16:
It seems to be some problem related to the magnetometer. And the drift i saw was due to the orientation estimation being incorrect.

I have changed the breadboard setup: replaced the Arduino with an esp8266 for faster speeds running the same Arduino code and a different mpu-9250 breakout board (gy-9250).

I am sending raw data (which looks good) from the sensor to my computer via serial (accelerometer, gyro, mag), doing the motion fusion on the computer and directly applying the quaternion to a cube.

Any IMU code works very well.
All AHRS/MARG code gives strange results.

The motion fusion works when the sensor is rotated on an axis that does not rotate around yaw. And doesn't rotate at all around the yaw axis (relative to earth). This leads me to think that the part of the correction related to gravity works but not compass.

Here is a video, i show how it should rotate, then i turn off the gyro and show how it solves orientation with gravity and compass.

https://youtu.be/GtdHHOppNCs

Results seem to be the same with or without mag calibration, how important is calibration?

what i pass into the function: (-ax, ay, az, gx, -gy, -gz, my, -mx, mz);

What do you think and do you have any suggestions?

Thanks

[mpu9250 quaternionFilters.zip](https://github.com/kriswiner/MPU-9250/files/1003293/mpu9250.quaternionFilters.zip)

kriswiner, 2017-05-16:
Calibration (especially of the mag) is pretty important to get reasonable
results. Plot Mx vs My, Mx vs. Mz and My vs. Mz while moving the sensor all
around. Are the clouds of points centered on the origin? Are the clouds
spherical slices or ellipsoidal? Until these data are spheres centered on
the origin, your mag is not calibrated and without a calibrated mag you
will get nonsense from AHRS.

Jaqos, 2017-05-16:
this is what the plots look like

[image]
[image]
[image]

the My vs Mz plot looks ellipsoidal

the AHRS seems to be the same with and without calibration.

kriswiner, 2017-05-16:
This doesn't look well calibrated, you will have crosstalk between My and
Mz, causeing headign drift when you make a change in roll (or pitch,
depending on your axes choices).

If you are using the ESP8266, you should be able to use this sketch to get
good AHRS, I do and don't have the kinds of problems you describe:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

Lastly, the Sparkfun MPU9250 design is flawed, and you could easily be
picking up false mag field reading since they poorly chose to rout power
traces under the MPU9250 IC.

Jaqos, 2017-05-21:
I some questions:

Do you think when calibration is perfect absolute orientation should be as well?

How to correct elliptical mag plot / cross talk?

How should the accelerometer be calibrated? 

I am using a different mpu-9250 board (the gy-9250).

Thanks

Jaqos, 2017-06-06:
I understand the concepts of mag and gyro calibration, how should the accelerometer be calibrated?

Gyro calibration: (rest on flat surface, read bias)
Mag calibration: (make a prefect sphere)
Accelerometer calibration ??

kriswiner, 2017-06-06:
Same principle for accel. Flat and motionless. Ideal values are 0, 0, 1.
Calculate bias as deviations from these. There are examples of how to do
this in th MPU9250 repository. For more sophiticated aceel calibration see
here:

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/F.--Magnetometer-and-Accelerometer-Calibration

myzb, 2017-06-14:
I'm encountering the same issue with my Sparkfun MPU, to bad I missed this post before ordering. Basically the sample "cube" I'm rotating on the PC screen slowly moves to kind of the opposite direction after putting the sensor to rest on the table. I assume this is related to the Sparkfun bug.

Before I order the other sensor @kriswiner suggested, I wanted to know if @Jaqos did find a workaround/solution to this.

Just for completion, I'm running this on a Teensy 3.6.

kriswiner, 2017-06-14:
Are you sure you are properly calibrating the sensor?

myzb, 2017-06-14:
I started with 'MPU9250_MS5637_AHRS_t3.ino' from your github and #ifdef'd the barometer code. Then I changed the "Serial.send" commands to just send yaw,pitch,roll in order to be able to display the data in the serial plotter. I Also tried with the Sparkfun version. (To rotated the cube on my PC app I added the necessary USB functions to send the quaternion.)

I followed the "figure 8 instructions" during startup so I guess the sensor is 'properly' calibrated?
I'll re-check everything on Friday as I don't have my development board with me atm.

I'll start with a sphere plot of the mag field to get out of doubts. I'll get back to you. Thanks for your time.

myzb, 2017-06-16:
The 'backwards spin' of my cube after putting the sensor to rest, was related to a copy paste error in my code. I still have some slight issue, but I think they are related to some code changes I did and are tide to the 'deltat' integration time, as after stopping any motion my cube rotates for a few more degrees.

These are my calibrated ellipsoids. I think they look quite decent within -400/+400 and centred in the origin. What do you think? 

btw, I was planing to use the IMU to do some AR Headtracking.

[image]
[image]
[image]

kriswiner, 2017-06-16:
The fusion method is iterative, it will get closer and closr to a stable answer each time the filter is iterated on a set of sample data.

Jaqos, 2017-06-21:
@myzb - check out MagMaster, it can calibrate soft-iron and its easy to use, if the mag plot looks distorted on the Sparkfun MPU-9250, give it a try.
http://diydrones.com/profiles/blogs/advanced-hard-and-soft-iron-magnetometer-calibration-for-dummies
https://github.com/YuriMat/MagMaster

Hi Kris,
i have been researching auto mag calibration, using ellipsoid fitting.

these articles look good:
https://skotagiri.wordpress.com/2010/06/19/c-implementation-for-fitting-an-ellipse-for-a-set-of-points/
http://diydrones.com/forum/topics/magnetometer-soft-and-hard-iron-calibration
https://sites.google.com/site/sailboatinstruments1/c-language-implementation

do you know of any open source ellipsoid fitting algorithms in c# ?

thanks.

kriswiner, 2017-06-21:
> iron-magnetometer-calibration-for-dummies
> implementation-for-fitting-an-ellipse-for-a-set-of-points/
> hard-iron-calibration
> language-implementation

Yes, freescale has one, ask about it at the Teensy forum.

myzb, 2017-06-27:
@Jaqos I managed to get a quite decent calibration with the current mag calibration method after a few tries. As I will be using this for head orientation tracking inside of vehicles an more advanced calibration may come in handy down the road. Thanks for the links.

arkhipenko, 2018-09-03:
I would recommend moving to Teensy 3.5 or 3.6.   These babies run at 120Mhz and have a math coprocessor...