kriswiner/MPU9250 #126: Fix for accelerometer calibration

aclex, 2017-03-15:
Hello, Kris,

I tried to port and learn the calibration code yesterday (`accelgyrocalMPU9250()` of `MPU9250_MS5637_AHRS_t3.ino` sketch), it worked fine for me, but I also wanted to sort out with the accelerometer calibration, which is obviously a little more tricky, and the writing the results to offset registers were even commented out in the sketch.

I looked at the register map document (`RM-MPU-9250A-00-v1.6.pdf`) and noticed, that offset registers seem to contain only 15 lower bits of the offset value, which is placed at 15 higher bits of `XA_OFFSET_H` - `XA_OFFSET_L` (considering X axis), while LSB of this word is undocumented (in the sketch comments you stated, that it's temperature bit). So, the idea was, that to create proper corrected value we should scale our bias value to 16 g (i.e. divide by 8, as it's in the code), and subtract the result from the value in 15 higher bits of the just-read factory offset register word (elements of `accel_bias_reg[]` in the sketch).

I tried to follow the existing way of manipulating `accel_bias_reg` value, but confused completely with it and found, that the easiest way is to just divide the bias value by 16 and shift it one bit left, like this:
```
accel_bias_reg[0] -= (accel_bias[0] >> 4) << 1;
```
Mask saving and restoring is not needed in this case, as LSB of `accel_bias_reg` is preserved. With this correction the readings of accelerometers become centered around (0, 0, g) value, so it seemed to work for me.

The only problem is left shift operation, which is apparently [UB, if `accel_bias` value is negative](http://en.cppreference.com/w/cpp/language/operator_arithmetic#Bitwise_shift_operators). Still it works as expected for me on Raspberry, but is obviously non-portable. To be on the light side, we could use some longer code:
```
int16_t temp = accel_bias[0] >> 4;
accel_bias_reg[0] -= temp * 2;
```

I hope, this little correction would also work for you, as the whole calibration process (including mag calibration) is quite effective and helped me to, for example, correct Yaw orientation by about +10°.

Thanks again for very useful sketch and detailed materials on that!

kriswiner, 2017-03-15:
Thanks Alexey,

I periodically go back and try to make this work myself but the results by
putting the offset biases in the main loop work well enough my motivation
is low. Maybe I'll give it another try!

Kris

aclex, 2017-03-15:
Yes, I noticed these biases are stored and then compensated on data reading, that's indeed probably more obvious and also less confusing way. For example, about the gyro offset registers the documentation reads: "The  values  in  these  registers  are  subtracted  from  the  gyro  sensor  values  before  going  into  the sensor registers." , while in fact they are added, as you mention in the code. But I just wanted to try to make it work out of interest and to use this chip functionality, since it's implemented there.

AntoinedeChassey, 2017-04-05:
Hey Aclex,

Would you be so kind to share your version of the accel calibration function you ended up writing ? I did not manage to have a suitable one with the hints you gave unfortunatly...

It is quite unfortunate that the MPU HAS to be flat at start-up in order for the accel calibration to be correct...
I am willing to initiate the MPU vertically instead of horizontaly. When doing this, the acceleration on Z axis at start-up is correct (1G) but after turning 90 degrees it ends up with 2G... :(

Thanks in advance for any reply!

kriswiner, 2017-04-05:
This is simply a matter of how you subtract off the expected 1 G in the
calibrate function. If you know you will calibrate with the device
vertical, just change to the appropriate axis.

aclex, 2017-04-05:
@AntoinedeChassey yes, no problem, the only issue is that my version is for Raspberry Pi and for the sake of the project consistency I've converted the calibration code to C++14, so it differs dramatically from the sketch code. However, I hope, it would be helpful at least a little for you.

```C++
// Construct the accelerometer biases for push to the hardware accelerometer bias registers. These registers contain
// factory trim values which must be added to the calculated accelerometer biases; on boot up these registers will hold
// non-zero values. In addition, bit 0 of the lower byte must be preserved since it is used for temperature
// compensation calculations. Accelerometer bias registers expect bias input as 2048 LSB per g, so that
// the accelerometer biases calculated above must be divided by 8.

std::array<uint8_t, 2> offset_data = { };
std::array<uint16_t, 3> accel_bias_reg = { }; // A place to hold the factory accelerometer trim biases

// Read factory accelerometer trim values
// *a_offset_*-registers aren't adjacent, need to read them separately, word by word
bus >> reg_addr(address::xa_offset_h) >> offset_data;
accel_bias_reg[0] = to_word<uint16_t, int>(offset_data[0], offset_data[1]);
bus >> reg_addr(address::ya_offset_h) >> offset_data;
accel_bias_reg[1] = to_word<uint16_t, int>(offset_data[0], offset_data[1]);
bus >> reg_addr(address::za_offset_h) >> offset_data;
accel_bias_reg[2] = to_word<uint16_t, int>(offset_data[0], offset_data[1]);

std::cout << "accel bias factory: " << accel_bias_reg[0] << ' ' << accel_bias_reg[1] << ' ' << accel_bias_reg[2] << std::endl;

// Subtract calculated averaged accelerometer bias scaled to 2048 LSB/g (16 g full scale)
const auto& shift_me_right = [](int a, int b) { return b - sgn(a) * ((static_cast<uint16_t>(abs(a)) >> 4) << 1); };

// Construct total accelerometer bias, including calculated average accelerometer bias from above
std::transform(std::cbegin(accel_bias), std::cend(accel_bias), std::cbegin(accel_bias_reg), std::begin(accel_bias_reg), shift_me_right);

{
	std::array<uint8_t, 6> data = { };
	std::tie(data[0], data[1]) = to_bytes(accel_bias_reg[0]);
	std::tie(data[2], data[3]) = to_bytes(accel_bias_reg[1]);
	std::tie(data[4], data[5]) = to_bytes(accel_bias_reg[2]);

	// Apparently this is not working for the acceleration biases in the MPU-9250
	// Are we handling the temperature correction bit properly?
	// Push accelerometer biases to hardware registers
	bus << reg_addr(address::xa_offset_h) << data[0];
	bus << reg_addr(address::xa_offset_l) << data[1];
	bus << reg_addr(address::ya_offset_h) << data[2];
	bus << reg_addr(address::ya_offset_l) << data[3];
	bus << reg_addr(address::za_offset_h) << data[4];
	bus << reg_addr(address::za_offset_l) << data[5];
}
```

`sgn()` here is just a signum function (following [this advice](http://stackoverflow.com/a/4609795) from StackOverflow):
```C++
template <typename T> int sgn(T val)
{
	return (T(0) < val) - (val < T(0));
}
```

And `to_word()` and `to_bytes()` just glues bytes together to a word and splits a word to bytes, correspondingly. The stream operators express reading from and writing to MPU registers via SPI (it's connected through SPI in my configuration).

Also, please, note, that the accelerometer offset values are written byte by byte, as far, as I know, that's due to that these registers are not contiguous on the MPU side, so we can't just put a whole byte array at the initial register address in one call.

Again, sorry, that it doesn't tally well with the sketch code, but I hope, it would help.

AntoinedeChassey, 2017-04-05:
Firstly, thank you very much for both of your very fast replies.

Kris, below is the code I changed in the MPU9250.cpp. I am detecting which axes has the greatest `accel_bias` (in absolute) and then removing or adding the `accelsensitivity `(16384 LSB/g) according to the sign of this `accel_bias`.
`/* Commented Kris code
        if(accel_bias[2] > 0L) {accel_bias[2] -= (int32_t) accelsensitivity;} // Remove gravity from the z-axis accelerometer bias calculation
        else {accel_bias[2] += (int32_t) accelsensitivity;}
 */

// The code below will detect which accelerometer axes is closest to need gravity correction (in other words this allows to start the MPU-9250 in another position than flat)
        int accelBias[3] = {abs(accel_bias[0]), abs(accel_bias[1]), abs(accel_bias[2])};
        int value = abs(accelsensitivity - accelBias[0]);
        int axesToRemoveGravity = 0;

        for(int i = 0; i<3; i++) {
                if(value > abs(accelsensitivity - accelBias[i])) {
                        value = abs(accelsensitivity - accelBias[i]);
                        axesToRemoveGravity = i;
                }
        }

        if(accel_bias[axesToRemoveGravity] > 0L) {
                accel_bias[axesToRemoveGravity] -= (int32_t) accelsensitivity;
        }
        else {
                accel_bias[axesToRemoveGravity] += (int32_t) accelsensitivity;
        }
`

My problem is now solved. This solution might not be perfect at all but it works as I desired it to work... (for now.. ;)).
Thanks again for your great work.

TonDuino, 2017-10-14:
Hello Kris and Alexey,

I have been working on Kris's code to adapt it for the SPI-interface and found this post with Alexey's solution for the accelerometer's offset registers. I have implemented this in Kris's code for Arduino, and ... it works! 
If I run the calibration routine twice, the calculated biases are near zero the second time. Here is the output for the first and second calibration:

```
Calibrate accelerometer. Put device flat on horizontal surface and do not move!
  Bias values, based on #samples = 10000
  Axis	Counts	g
  X	-100	-0.006
  Y	-77	-0.005
  Z	1693	0.103
Calibrate accelerometer. Put device flat on horizontal surface and do not move!
  Bias values, based on #samples = 10000
  Axis	Counts	g
  X	-5	-0.000
  Y	-13	-0.001
  Z	13	0.001
```

Here is the code, starting with reading the accelerometer's offset registers and ending with writing them. (Note: I have adapted the readBytes en writeByte procecdures for SPI and so they don't contain the I2C MPU9250_ADDRESS anymore)

```
  int32_t accel_bias_reg[3] = {0, 0, 0}; // A place to hold the factory accelerometer trim biases
  readBytes(XA_OFFSET_H, 2, &rawData[0]); // Read factory accelerometer trim values
  accel_bias_reg[0] = (int32_t) (((int16_t)rawData[0] << 8) | rawData[1]);
  readBytes(YA_OFFSET_H, 2, &rawData[0]);
  accel_bias_reg[1] = (int32_t) (((int16_t)rawData[0] << 8) | rawData[1]);
  readBytes(ZA_OFFSET_H, 2, &rawData[0]);
  accel_bias_reg[2] = (int32_t) (((int16_t)rawData[0] << 8) | rawData[1]);

  // this replaces Kris Winers code for calculating the bias values to put into offset registers
  int sign;
  for(byte ii = 0; ii < 3; ii++) {
    if (accel_bias[ii] >= 0) sign = +1;    // determine sign of accel_bias
    else sign = -1;
    // Shift right abs(accel_bias) four positions and shift left 1 postion; result is division by 8
    // with least significant bit cleared. Then restore sign and subtract from factory value
    accel_bias_reg[ii] -= sign * ( ( (uint16_t)abs(accel_bias[ii]) >> 4) << 1);
  }

  rawData[0] = (accel_bias_reg[0] >> 8) & 0xFF;
  rawData[1] = (accel_bias_reg[0])      & 0xFF;
  rawData[2] = (accel_bias_reg[1] >> 8) & 0xFF;
  rawData[3] = (accel_bias_reg[1])      & 0xFF;
  rawData[4] = (accel_bias_reg[2] >> 8) & 0xFF;
  rawData[5] = (accel_bias_reg[2])      & 0xFF;
  
  // Push accelerometer biases to hardware registers
  writeByte(XA_OFFSET_H, rawData[0]);
  writeByte(XA_OFFSET_L, rawData[1]);
  writeByte(YA_OFFSET_H, rawData[2]);
  writeByte(YA_OFFSET_L, rawData[3]);
  writeByte(ZA_OFFSET_H, rawData[4]);
  writeByte(ZA_OFFSET_L, rawData[5]);

```
Thanks Kris and Alexey!

aclex, 2017-10-14:
@TonDuino Hello, TonDuino, thank you, nice, that it works for you! I also keep using the hardware bias registers for accelerometer.

Please, however, note, that, according to my experiments, the "software" way of storing bias values, like it's done in original Kris's sketch, seems to be slightly more accurate in terms of zero bias drift of accelerometer signal, as we can store these bias values at the best possible precision for our hardware, while storing in registers implies rounding and shifting. Although it's not too important, when using the signal in fusion filter, this may increase accuracy, if you plan to do e.g. accelerometer and barometer fusion for altimeter signal.

TonDuino, 2017-10-15:
Hi Alexey,
You are right, Kris's method for accelerometer bias compensation by software is more accurate, because his bias correction is not divided by 8. I don't know what the impact in practice is.
I hoped the MPU925x handles bias compensation for temperature change by means of the temperature compensation bit. However I cannot find any documentation about this in the datasheet nor the register map.  So this is pure speculation.
I do know the difference between your and Kris's method is, that you always write a zero to bit0 and Kris preserves the value of bit0 from the factory setting. Apparently the MPU925x is not happy with that.