kriswiner/MPU9250 #127: Create MPU-9250 object from other class

XaviLC, 2017-03-15:
HEllo,

I am using the library for the MPU-9250 of this repository to interface with my MPU9250, however I am facing an issue when starting the device, another of the components of my board stalls and stop working. 

I think that I am not creating properly the MPU-9250 object; I am using arduino and I am not sure whether or not I should include a line like `extern MPU9250 mpu9250` at the end of the `MPU9250.h` definition file.  I have not do that, I create my MPU-9250 object as `MPU9250 mpu9250;` however I wonder if I should create it in a different way like `MPU9250 mpu9250 = MPU9250();`

thanks in advance,

REgards

kriswiner, 2017-03-15:
I have no idea what you are doing. Are you using an Arduino IDE?

XaviLC, 2017-03-15:
Yes I am using Arduino for nRF52 platform. 

I have just figured out what was happening,  in the constructor of MPU9250 there is a call to ´wire.begin()´ my other component was dependent of wire class as well,  so the re-initialisation of the class wire was making my other component to fail/stop. 

I have comented out that call in MPU9250 constructor and now it does work fine

Regards

XaviLC, 2017-03-15:
@kriswiner looking your code, there is something that I don´t understand, the use of  ADO is only happening with the MPU9250 mounted into a Teensy 3.1 rigth?

```
// Using the MPU9250Teensy 3.1 Add-On shield, ADO is set to 0 
// Seven-bit device address is 110100 for ADO = 0 and 110101 for ADO = 1
#define ADO 0
#if ADO
#define MPU9250_ADDRESS 0x69  // Device address when ADO = 1
#define AK8963_ADDRESS 0x0C   //  Address of magnetometer
#define MS5611_ADDRESS 0x76   // Address of altimeter
#else
#define MPU9250_ADDRESS 0x68  // Device address when ADO = 0
#define AK8963_ADDRESS 0x0C   //  Address of magnetometer
#define MS5611_ADDRESS 0x77   // Address of altimeter
#endif
```

kriswiner, 2017-03-15:
ADO is the MPU9250 address pin, HIGH for 0x69, LOW for 0x68 I2C address