kriswiner/MPU9250 #128: Error in MPU9250 magnetometer values

ashwinvkm, 2017-03-15:
I am using MPU9250 in my project. While reading the magnetometer readings, I find that the readings are not as expected. According to my understanding, if I rotate MPU about its magnetometer x-axis, the x-axis component of magnetometer output should remain constant (more or less). But when I do this, I see huge variations in the magnetometer readings. Same thing with y-axis.  However z-axis gives fine results. What could be the reason for this? I have done only the sensitivity adjustment as per AK8963 datasheet.

ashwinvkm, 2017-03-15:
Adding to the above query, I am using the MPU-9250 breakout board from sparkfun. One thing that I notice is that, when I rotate the breakout board keeping it horizontal (i.e. rotating about z-axis of magnetometer), the x-axis and y-axis magnetometer readings shows unexpected change. The values change maximum to zero for other than 90 degree rotation. Is it a known issue with MPU-9250

kriswiner, 2017-03-15:
Is your magnetometer calibrated?

ashwinvkm, 2017-03-15:
The calibration that I have done is only the sensitivity adjustment calibration. Other than that what other calibration needs to be done?

kriswiner, 2017-03-15:
bias offset

https://github.com/kriswiner/MPU-6050/wiki/Simple-and-Effective-Magnetometer-Calibration

ashwinvkm, 2017-03-15:
As far I understand the bias is a fixed value that I can subtract from the magnetometer readings. But still the issue of the magnetometer x (or y) readings changing  when I rotate mpu9250 about magnetometer x (or y) axis will persist right?

kriswiner, 2017-03-15:
yes, this is normal. It's how magnetometers work.

ashwinvkm, 2017-03-16:
The bias offset calibration method that you suggested in the above link solved the problem I was facing. Thanks

closed 2017-03-23 by ashwinvkm