kriswiner/MPU9250 #129: how to plot circle from magnetometer value

sexyday, 2017-03-26:
hi kris,
i found some magnetometer calibration issue from others website(includeing here),but i still can't figure out why i can't plot circle even ellipse like these
[image]
[image]

my result  is showed below.
[image]
if i want to plot circle  ,how to rotate my imu?whose coordinate should i refere to ?inertial z axis or imu?
what should i operate with it?here are two methods what i realize from other website,which is right?
or both are wrong?
[image]
[image]

kriswiner, 2017-03-26:
Move the sensor in every direction up dowm rotate left right etc. The idea
is to sample as much of the response surface as possible during the
collection of the data.

sexyday, 2017-03-27:
should i normalize mag value like this

			 magn=1/(sqrt(mX*mX+mY*mY+mZ*mZ));
			 mX=mX* magn*100;
			 mY=mY* magn*100;
			 mZ=mZ* magn*100;

kriswiner, 2017-03-27:
I would say no, but I don't know what you are trying to do here.

sexyday, 2017-05-08:
i have a new question.when there is pitch or roll angle occured ,yaw angle will be inaccurate. how to solve this problem?i find a solution the picture shows below.
[image]
but the problem still occure.

kriswiner, 2017-05-08:
I have no idea what you are trying to do.