kriswiner/MPU9250 #131: MPU-9250 Not Recognized anymore

mjs513, 2017-03-28:
Hi Kris, 
Me again.  I have been playing around the 9250 (your board) to try and get either MasterMode or FIFO working.  But I kept having a problem with after I changed the registers it would not recognize the 9250 when I did a who I am check.  So I decided to reload your code as a check to make sure it still worked.  Guess what it didn't this is what I get:
```
Scanning...
No I2C devices found
MPU9250 9-axis motion sensor...
MPU9250 I AM E3 I should be 71
Could not connect to MPU9250: 0xE3
```
Just to be sure I did this test on a Arduino 101 board then attached it a 3.5 and no luck.  Any ideas?
PS.  DId you ever get FIFO working on the 9250?

thanks
Mike

kriswiner, 2017-03-28:
E3 sounds like it is shifted by 1. What do you see when you do an I2C scan?

I use the FIFO for the calibration routine, not much else.

kriswiner, 2017-03-28:
Check that SDA and SCL are at ~3V when the board is powered on.

If it was working before, what could have changed? Surely software would
not break the board...

mjs513, 2017-03-28:
Ok.  Here we go. 

SDA and SCL are at 3.27v (multimeter).    I just ran a I2C scanner and I can see the 9250 at 0x68.  Note:  The scanner did not work before.  Since I saw it I reran the sketch and it still said "NO I2C Devices Found" and the Iam is still at E3.

For the heck of it I commented out the if test on whoami just to let it run and see what happens.  It did recognize the magnetometer at 0x48 and ran the tests but the accel/gyro as expected was zero.  As another  what the heck I ran Jeff Rowbergs MPU60x0 raw example.  The connection still failed but still got good accel and gyro data from the device.

This is giving me a headache.

kriswiner, 2017-03-28:
Sounds like a connection problem somewhere.

mjs513, 2017-03-28:
Ok.  Ain't many connections.  Will recheck everything and let you know.  Just strange.  Oh, the reason I was asking about the fifo is that I was getting a bunch of questions on implementing multiple (2) imu's with the FreeIMU lib.  Haven't much luck. Any good sources on anyone who got it working?  Did you ever get the chance to check out the esp32 9250 mods I made?

kriswiner, 2017-03-28:
Not sure which ESP32 mods you mean. You mean to the MPU9250 sketch I posted
at my repo for the ESP32?

Using the FIFO is straightforward with I2C, not sure how to use it with SPI.

For reading two MPU9250s it is far easier to toggle INT_CFG with I2C than
set up and read the embedded slave address registers.

mjs513, 2017-03-28:
I attached two files to the first message of [https://github.com/kriswiner/ESP32/issues/5](https://github.com/kriswiner/ESP32/issues/5).  The first was your 9250 sketch moded to use UDP to send IMU data over UDP.  The second was an associated Processing sketch to visualize the data.

What's got me confused was all the setup for with slave registers.  But now not sure if I had a code error or a connection problem.  Have no problem with setting up reading the accel/gyro data for separate imu's but whats got me is how to distinguish the AK8963 data since the address is the same.

Mike

kriswiner, 2017-03-28:
> I attached two files to the first message of kriswiner/ESP32#5

That's why I2C is better IMHO. If you address one MPU9250 at 0x68 and set
INT_CFG to 0x00, this should close the switch to the associated AK8963C.
The write 0x12 to INT_CFG of the MPU9250 at 0x69 and read the mag data.
Then toggle back and forth. Clunky, yes, but it can work. And if you use
the FIFO, you can at least store the data while the IMU is "offline" and
read it when its turn comes up.

mjs513, 2017-03-29:
Sorry for the delay.  Had some problems with my usb connections.  Found one connection error - it was on the protoshield I was using - go figure.  Anyway resolder the joints on the 9250 and still no luck.  The I2C scanner just hung.  I downloaded and installed the Arduino IDE 1.8.2 and ran Paul's version of the I2C scanner. The scanner returned no I2C devices found.  So I soldered up the new 9250 that I just got the other day from you but had the same results.  Have 3.2v on SDA and SCL.  To check if its the wires or the chip I attached a 6050 10DOf I have and it recognized it no problem.  

By the way for these tests I am using the Teensy 3.5.  I know the 101 has some problems with I2C so decided to stay away from it.  Have no idea what else to check.  By the way do you use pullups?

kriswiner, 2017-03-29:
Of course, one must use pullups with I2C and I wouldn't rely on so-called
internal GPIO pullups. This is not sufficient for proper I2C. There are 4K7
pullups on the MPU9250 board but you have to close the solder joint to
enable them. If you got a new board from me and you can't get an I2C ACK
there is something wrong with your MCU and/or wiring. I test each board
before I send it out and I can guarantee unless the post man stepped on it
that the board will work.

Are you sure you don't have SDA and SCL reversed? I mean, what could it be?

How are you connecting 3V3 and GND? if on the top two holes then this
should be fine. If on the boards edge, you have to close the solder jumpers
on the back to enable power and GND to be taken from the board edge.

mjs513, 2017-03-29:
Here's another one to confuse you.  I attached the 9250 up to your ESP32 board which already had the 9250 sketch loaded and it recognized the addresses just fine.  Think the only difference is SDA/SCL pins were specified.  Wanted to give it a try but when I tried compiling for the ESP32 I received the following error:
`gen_esp32part.exe": file does not exist`

kriswiner, 2017-03-29:
Sounds like many software errors or something then. This doesn't sound like
a hardware error. I have no idea what's going wrong...

mjs513, 2017-03-29:
Kris.  I am at a loss.  

I am using the power pins on the boards edge and closed the jumpers on the underside of the board.  I seems to work fine with the ESP32 but not the 3.5.  Yes I have 3.3v going to 3.3v on the teensy and gnd to gnd.  I tested the connections with the 6050 and the hardware seems to work fine.  I even reversed them just in case but that just froze the scanner and or the 9250 sketch.

The only thing left are the pullups resistors.  I will give it a test tomorrow.  I missed that part that I have to solder the jumpers on the top side.

Thanks for the help.

mjs513, 2017-03-29:
Hi Kris.  Ok last for tonight. I agree don't think its hardware, teensy 3.5, 9250 or the connections.  Think there is something with wire? 

I soldered up the jumper pads on top of the board. Running Arduino ide 1.8.2 with beta teensyduino.
Tests ran....
1. Attached to Teensy 3.5 and first ran Paul's scanner and it returned error on all addresses and finally no i2c devices found.  Tried the i2cscanner and it did not find anything just froze.
2. Attached to Arduino Curie ran the normal i2cscanner and it found the 9250 at 0x68 and the baro at 0x76.  Then I tried running Pauls scanner and it periodically found the 9250 but not the baro between no i2c devices found and then identified the 9250 on 0x69.

By the way tried with 1.8.1 ide and same thing happens.  Going to try tomorrow with the Invense 9250 breakout board (yes I have one of those as well - but I like yours more).

Again it works fine with your esp32 board.  I am now getting the following error on the 9250 sketch (this is the from the esp32 just with wire.begin(400000).
```
> 
>  MPU9250_MS5637_AHRSv2: In function 'uint8_t readByte(uint8_t, uint8_t)':
> C:\Users\CyberPalin\Documents\Arduino\MPU9250_MS5637_AHRSv2\MPU9250_MS5637_AHRSv2.ino:1200:30: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]
> 
>    Wire.requestFrom(address, 1);            // Read two bytes from slave register address on MPU9250 
>                                ^
> In file included from C:\Users\CyberPalin\Documents\Arduino\MPU9250_MS5637_AHRSv2\MPU9250_MS5637_AHRSv2.ino:29:0:
> 
> C:\Local Programs\arduino-1.8.1\hardware\teensy\avr\libraries\Wire/Wire.h:73:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)
> 
>      uint8_t requestFrom(int, int);
>               ^
>  C:\Local Programs\arduino-1.8.1\hardware\teensy\avr\libraries\Wire/Wire.h:71:13: note: candidate uint8_t TwoWire::requestFrom(uint8_t, uint8_t)
> 
>      uint8_t requestFrom(uint8_t, uint8_t);
```

Good night.

mjs513, 2017-03-29:
Ok. Kris.  Had a bad solder joint on the jumpers.  Checked this morning after attaching the invense 9250 and checking that out.  I ran the i2cscanners and it saw the ak8962, 9250 and the baro.  Ran your 9250 sketch (as posted with the ESP32 since the lcd stuff is removed) and still got the same warnings as above but this time at least all devices were found and calibrated.  However, the values for accel, gyro and mag were all zero.  You want I close this one and open a different issue.

kriswiner, 2017-03-29:
I don't care about proper issue management. There must be some software
problem then if you can get an I2C ACK but register reads are zero. Does
this include the WHO_AM_I registers?

The skecth for the ESP32 is probably using an interrupt. If you don't have
one the sketch will never read the data.

mjs513, 2017-03-29:
Got the WHO_I_AM registers read correctly.  That's probably whats happening.  Never getting the interrupt.  I will check later and keep you posted.  I tested the 9250 with the Freeimu library and that works fine so the HW is ok (I didn't mess it up :) ).