kriswiner/MPU9250 #132: Calculation of Angular Rate and Acceleration from Quaternion

AmirhosseinR, 2017-04-02:
Hello Kris,
Thanks for your code, and your topic on "Affordable 9 DoF Sensor Fusion". It is so useful. I have some question about your code: "MPU9250_MS5637_AHRS_t3.ino"
1- In line 638, "lin_ax = ax + a31;", why you add "ax" by"a31" to calculate lin_ax? I do not get it?
2- In my project, I need angular rate. Can I calculate it from Quaternion or just using gyro data?
3- Which algorithm is better, Madgwick or Mahony? 
Thanks for your time.

kriswiner, 2017-04-02:
1) this is just the definition. -a31 is the gravity component.

2) use calibrated gyro then

3) I prefer madgwick, but neither will give good results without good calibration.

AmirhosseinR, 2017-04-02:
Thank you for your prompt reply. 
I google "convert quaternion to angular rate" and [I found this formula](https://en.wikipedia.org/wiki/Rotation_formalisms_in_three_dimensions#Quaternion_.E2.86.94_angular_velocities):
[0 Wx Wy Wz] = 2 q_dot q_inv
so I added this code in the last line of "MadgwickQuaternionUpdate" function in "quaternionFilters.ino" file:
`	// Compute Angular Rate : w = 2 x qDot x qStar`
`	AngRat[0] = 2.0f * (qDot1*q[0] + qDot2*q[2] + qDot3*q[3] + qDot4*q[4]);`
`	AngRat[1] = 2.0f * (qDot1*q[2] - qDot2*q[1] - qDot3*q[4] + qDot4*q[3]);`
`	AngRat[2] = 2.0f * (qDot1*q[3] + qDot2*q[4] - qDot3*q[1] - qDot4*q[2]);`
`	AngRat[3] = 2.0f * (qDot1*q[4] - qDot2*q[3] + qDot3*q[2] - qDot4*q[1]);`
I do not know which one is more accurate, calibrated gyro output or stimated angulare rate by Quaternion.
What is your idea?

kriswiner, 2017-04-02:
No idea, I would use the gyro.

AmirhosseinR, 2017-04-02:
Thank you. I would compare results.