kriswiner/MPU9250 #134: Question?  9250 miniboard on arduino uno

mjs513, 2017-04-03:
Kris
Quick question if I may.  Can I run your mini-board on a uno with out level shifter on sda/scl? Or should I use level shifter?

Thanks
Mike

PS.  I extracted the code piece on master mode from Brian Taylor's library and it works really well you can actually use multiple 9250s on since i2c bus.  Just thought you would like to know.

kriswiner, 2017-04-03:
The board is 3V3 only so you should really use a level shifter.

Does this mean you have sorted out your problem(s)?

mjs513, 2017-04-03:
Thanks Kris.  Wanted to be sure.  

Yes problems all sorted out.  The initial issue with losing the address looked like it was due to pull-up pads not being soldered.  Then I found out some interesting things with Jeff's Rowbergs library when trying to use master mode or the fifo.

Anyway using Brian's master mode control code it works perfectly for using multiple sensors since the mag data is read into the external sensor register and then you read all data straight way.  Don't have to address the mag separately or don't have to deal with fifo.  You may want to check it out.  Also, it would probably work for the 9150 and my guess is maybe the 6050 with an external mag.

Mike