kriswiner/MPU9250 #135: I dont have VDDI pin on MPU9250

hemangjoshi37a, 2017-04-05:
I have only these pins on my mpu9250

VCC
GND
SCL/SCLK
SDA/SDI
EDA
ECL
ADO/SDO
INT
NCS
FSYNC

It shows device not found on serial monitor .

Please somebody help.

Thank you.

MaxZZZZZZ, 2017-04-16:
Same here. Was you able to solve it?

hemangjoshi37a, 2017-04-16:
Naah. I will ping you if I get it solved. And you also please leave comment if you get solved.

kriswiner, 2017-04-16:
What exactly is the problem?

kriswiner, 2017-04-17:
Pull NCS HIGH, this enables I2C. If you have pullups on your board, you should measure SDA and SCL at ~3V and be able to get an I2C ACK at address 0x68.

hemangjoshi37a, 2017-04-17:
Okay @kriswiner sir, I will try that.
Thank you for your support.🙏

ksmo7, 2017-05-05:
So you just replaced the VDDI with NCS?

hemangjoshi37a, 2017-05-05:
Yes exactly... @ksmo7