kriswiner/MPU9250 #138: Simple Magnetometer and Accelerometer readings

petkool, 2017-04-08:
Hi Kris,

I am building a land robot for a school project and bought the Sparkfun MPU-9250 IMU. My level of arduino knowledge is not the best, but I am having trouble understanding your library. I intend to do things:

1. Calculate a rough estimate of linear velocity using accelerometer values.

2. Using the magnetometer readings to get a compass heading.

Do let me know if you have any idea on how to achieve my tasks with your library.

Thanks.

kriswiner, 2017-04-08:
Did you look here?

https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

This sketch prodces quaternions from which heading (and pitch and roll) as
well as linear acceleration adn gravity are derived.

petkool, 2017-04-11:
Hi Kris, 

I tried the code and i seem to be getting alot of errors. I think it is because it is optimized for Teensy

kriswiner, 2017-04-11:
Probably the i2c_t3.h library instead of the Wire.h library.