kriswiner/MPU9250 #139: Most recent code?

mconsidine, 2017-04-22:
Hi,
Thanks as well for all the work you have done to provide this code.  It's very helpful.
I have ported one of the sketches to a NodeMCU/ESP8266 and noticed that some of the sketches are slightly different.  I began working off of the "BasicAHRS" sketch, but according to the list of commits the sketch for BMP180 is the most recently updated.
In particular, it looks like the sketch "MPU9250_BMP280_BasicAHRS_t3.ino" does not do anything with magScale, while "MPU9250_MS5637_AHRS_t3.ino" does.  Yet the former seems to be derived from the latter if the first commented line is any indications.
Similarly, it looks like the "magcal" routine is a bit different among the sketches.
Could you advise which one would be considered the most up to date to build off of?
Thank you in advance for any help.
Matt

kriswiner, 2017-04-22:
Actually this:

https://github.com/kriswiner/Butterfly/blob/master/MPU9250_MS5637_BasicAHRS2_Butterfly.ino

There is also this repository:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

is likely the most "up-to-date" sketch but remember, these are not intended
to be turnkey black box solutions. Rather, they are demonstrations of how
to use the MPU9250, etc. You will have to configure for your own use
case(s). I have put in enough comments that you should be able to
understand what I am doing so you can decide if it is 1) correct and 2)
something you also want to do.

Kris

mconsidine, 2017-04-22:
Much obliged.  Thank you.

mconsidine, 2017-04-22:
Actually, a quick follow-up question:  In your sample code you have
`MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);`
I'm assuming you have these examples written for your break out board shown here
  https://www.tindie.com/products/onehorse/mpu9250-teensy-3x-add-on-shields/
and was wondering if a) that is correct and b) if so, are you taking +x to be north and +y to be west with this  call?  Is this chip reading as a 9250 or a 9255?  I have a 9255 blue board from aliexpress and am wrestling with getting the NED correct.
Thanks again.

kriswiner, 2017-04-22:
For NED and accel x North, I would use ax, -ay, -az, gx,- gy, -gz, my, mx,
mz)

mconsidine, 2017-04-22:
Again, thank you.

closed 2017-04-22 by mconsidine