kriswiner/MPU9250 #140: can't calculate the heading with mpu9240

luisvelas, 2017-04-24:
i bougth a mpu9250 and im trying to calculate heading using the compass AK8963 included in the mpu9260. i've read that its calculating atan(my/mx) with raw values and also found that i have to consider "tilt" and i get the ecuations for that but in both cases my heading only vary between 130 and 60 degrees, i dont know what im doing wrong. Can anyone give me some advice??

kriswiner, 2017-04-24:
Try this instead:

https://github.com/kriswiner/Ladybug/blob/master/MPU9250_MS5637_BasicAHRS2_Ladybug.ino

luisvelas, 2017-04-24:
thanks kriswiner i will test with that code, but my issue stay in its really possible to calculate heading using only raw values of the compass???

kriswiner, 2017-04-24:
Not really.

mconsidine, 2017-04-24:
Hi Kris,
If I could just jump in with a question related to the sketch you reference above.  In this sketch
https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_BasicAHRS2_ESP8266.ino
starting on line 456 there is this code segment
`    readMagData(magCount);  // Read the x/y/z adc values
   
    // Calculate the magnetometer values in milliGauss
    // Include factory calibration per data sheet and user environmental corrections
    if(newMagData == true) {
      newMagData = false; // reset newMagData flag
      mx = (float)magCount[0]*mRes*magCalibration[0] - magBias[0];  // get actual magnetometer value, this depends on scale being set
      my = (float)magCount[1]*mRes*magCalibration[1] - magBias[1];  
      mz = (float)magCount[2]*mRes*magCalibration[2] - magBias[2];  
      mx *= magScale[0];
      my *= magScale[1];
      mz *= magScale[2]; 
    } `

while on the sketch noted above the similar codeblock looks like this
`    newMagData = (readByte(AK8963_ADDRESS, AK8963_ST1) & 0x01);
    if(newMagData == true) { // wait for magnetometer data ready bit to be set
      readMagData(magCount);  // Read the x/y/z adc values
  
    // Calculate the magnetometer values in milliGauss
    // Include factory calibration per data sheet and user environmental corrections
      mx = (float)magCount[0]*mRes*magCalibration[0] - magBias[0];  // get actual magnetometer value, this depends on scale being set
      my = (float)magCount[1]*mRes*magCalibration[1] - magBias[1];  
      mz = (float)magCount[2]*mRes*magCalibration[2] - magBias[2];  
      mx *= magScale[0];
      my *= magScale[1];
      mz *= magScale[2]; 
    }`

In other words, the readmagData call in the first case is outside of an if-then test, where in the second case it is included.  Also in the second case, the call the the Madgwick routine is inside a for 1-to-20 loop.

Can you explain if one implementation is preferred over the other in your implementation?  Is the difference related to the hardware each is written for, for some reason?

Thanks in advance,
Matt

kriswiner, 2017-04-24:
> MPU9250_MS5637_BasicAHRS2_ESP8266.ino

The difference in the mag call is that the status check is handled in the
readMag function in one case and explicitly in the other. Best would be a
dedicated mag interrupt, but lacking this one has to poll. I think the
better way to poll is to ask if there is a new data ready flag set after
each time there is new accel/gyro data. So the second method where the mad
read is a simple register read with no data ready status bit check.

On the second question, the Madwick function will be run at the rate of the
MCU if not regulated., For some MCUs like the poky AVR or even Teensy 3.2
the rates are ~1 kHz or less, which is about right. For the STM32L4
however, the rates can approach 60 kHz, which is way too much. So
regulating the number of Madwick (or Mahony) iterations to 10 or 20 or 50
or 100 is the better method.

mconsidine, 2017-04-24:
The ESP8266 is running at a good clip, so it looks like I should switch the code over to the looping version of the Madgwick.  And it sounds like the "second method" you refer to on the readMag function is that found in "Ladybug."  I appreciate the perspective - thank you again.

luisvelas, 2017-04-24:
from where can i get the "MadgwickQuaternionUpdate" function????

kriswiner, 2017-04-24:
In the same repo

mconsidine, 2017-04-27:
Hi again,
From above

> The difference in the mag call is that the status check is handled in the readMag function in one case and explicitly in the other. Best would be a dedicated mag interrupt, but lacking this one has [...]

In reading through some of the docs it seems that if one avoided "bypass" mode then magnetometer readings would be handled by the 9250 and then passed to the host.  Do I understand that correctly?  If so, then the docs suggest that an interrupt can be triggered by a couple of events, one of which is new data in the Data Register (as opposed to FIFO).  Wouldn't that in effect put the magnetometer on an interrupt and remove the need to poll?  I'm probably missing something subtle - or not so subtle - so apologies if the answer should be apparent.
Thanks,
Matt

kriswiner, 2017-04-27:
Here is an MPU9250 library that does this:

https://github.com/brianc118/MPU9250

You can then read all of the sensor data in a burst read from the fifo. It
is a bit more difficult to set up and a bit less transparent than the
method I use but the advantage is slight. That is, you are still
reduntantly reading mag data or have toset up an internal timer (via the
sample rate) so that the embedded API can manage the mag data read. This
doesn't really save anything in speed or efficiency but adds rather a lot
of complexity.

For the fast MCUs I use there is really no advantage over the simple
pollingof the status register. Still, an interrupt would be even better.