kriswiner/MPU9250 #141: Rotating the results to get proper roll, pitch, and yaw angles

skrick, 2017-04-24:
I apologize in advance if this question has been asked before, but I did search the issues and haven't seen my question.

I am able to calculate roll, pitch, and yaw for the sensor assuming it is installed face up.  But what if I install the sensor at a complicated angle(s) in my "box".  That is, what if I expect roll and pitch to be 0 degrees while the board is at a,b,c degrees in the x,y,z axis.  I assume I simply multiply my results by a rotation matrix of sorts but is this done before, during, or after the filter?  And what does that rotation matrix look like?  Thanks.

kriswiner, 2017-04-24:
You can also just calculate the biases at the intended angle of use and use
this as a reference for changes. No rotation matrix required.

skrick, 2017-05-15:
A "no rotation" method would be great (fewer computations). But when you say "use this as a reference for changes" what exactly are you implying. Do we replace the final bias with the difference between the original bias and the intended angle bias?

kriswiner, 2017-05-15:
Define NED wrt your preferred installation orientation, calculate offset biases in this orientation, pretty straightforward, Not sure what is tripping you up?

skrick, 2017-05-15:
I understand now, thank you. I had forgotten to redefine NED wrt my installed orientation so the re-calibrated bias wasn't helping me. Just so you know where I am coming from, I'm looking at complicated installations angles where no x,y,z axis is aligned with NED but all offset by some angle e.g. -120.25 z-axis and -90 degrees in the x-axis.  I guess I just need to write a function that will take the new angles and redefine the NED for me.

kriswiner, 2017-05-15:
You can define an NED aligned with x,y,z sensor axes but that is "close" to
you expected orientation. This way you can calibrate more easily on a flat
surface and then install with reasonable fidelity in the results. There
will always be at least one some "flat" orientation close enough to your
install orientation to make this work.

skrick, 2017-05-15:
Excellent! I wondered if that would work but never tested it. That would by far be the simplest approach.