kriswiner/MPU9250 #142: "Programming Sequence for DMP Hardware Functions"

mconsidine, 2017-04-26:
Hi Kris,
The above document is referenced in some of the existing MPU-9250 docs as - it would appear - enabling quaternion output from the MPU-9250.  Have you explored any of the "advanced" functions that are referenced in the existing docs?
Thanks,
Matt

kriswiner, 2017-04-26:
The MPU9250 DMP can only do 6 DoF fusion; 9 DoF fusion has to be dome on
the host MCU. I have used their 6 DoF fusion solution from the DMP but it
is not as good as even simple madwick fusion done on the host. See:

https://github.com/kriswiner/MPU-6050/wiki/Affordable-9-DoF-Sensor-Fusion

mconsidine, 2017-04-26:
Okay, thanks.  I managed to get a copy of the above mentioned docs and was wondering how useful the "advanced" functions would be.
Matt

kriswiner, 2017-04-26:
Can you send me a copy of the document(s) please?

mconsidine, 2017-04-26:
Sent a pm to you on the pjrc site.

kriswiner, 2017-04-26:
Please send it to me at tleracorp@gmail.com, thanks.

mconsidine, 2017-04-26:
Email sent.

kriswiner, 2017-04-26:
Thanks, I have seen most of these before but not the advanced features.

Too bad they don;t publish the firmware, I would like to try the tap
detection.

closed 2017-04-27 by mconsidine

phanhust09, 2017-05-26:
hi. i have project use MPU. so i want to find out more about DMP in MP 9250. so could you send me this document . thanks 
mail : lamhust09@gmail.com

kriswiner, 2017-05-26:
Don't have it, ask Invensense.

kriswiner, 2017-05-26:
Actually, I do.

SeanStayn, 2018-01-06:
SparkFun has a firmware in the git: https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/util/inv_mpu_dmp_motion_driver.c

At line 223.

Perhaps, it helps you.

SeanStayn, 2018-01-06:
Additionally: In the datasheet from the MPU9250 on the page 54 is a hint, that a 9-axis quat. output is possible. 

https://cdn.sparkfun.com/assets/learn_tutorials/5/5/0/MPU-9250-Register-Map.pdf

It would be very cool to use the direct quat values. :)

kriswiner, 2018-01-06:
> MPU-9250-Register-Map.pdf

Not with the MPU9250, not possible.

But with this, it is:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=ac_prod_search