kriswiner/MPU9250 #143: Madgwick for positional displacement

boskopetkovski, 2017-05-03:
Hello, 

I'm trying to implement the madgwick algorithm with MPU-9250 in order to determine the gravity component and subsequently the linear acceleration from which I'll try to calculate positional displacement. 

I intend to use this position in a loosely coupled Kalman filter with GNSS. 

However,  after initial testing I'm worried that the algorithm's response is too slow to be used in this scenario.

Have you had any experience with Madgwick for positional displacement/linear acceleration?

Thanks!

kriswiner, 2017-05-03:
We use the Madgwick algorithm frequently for gravity and linear
acceleration estimation. With the STM32L4 we can easily run the Madgwick
algorithm at 50 kHz with gyro sample rate at 1 kHz, way faster than
required for any sensible orientation estimation. Velocity and position
estimation will always be error prone and only accurate, if at all, for
very short times. We do not usually attempt this. if the response of the
algorithm is too slow, I suggest you move to a CORTEX M4F MCU.

boskopetkovski, 2017-05-04:
Hello Kris, 

Thank you for your answer. I'm working with Cortex M3 and moving to a different MCU is not an option. However I've started optimizing the algorithm to use fixed point arithmetics and the initial results are interesting.

I think I'll get at least 40Hz this way which should be more than enough for the application I'm working on.

kriswiner, 2017-05-04:
You might consider using the EM7180, it runs the fusion at the rate of the
gyro up to 400 Hz and can provide a pretty accurate orientation estimation
as well as gravity and linear acceleration with all calculation for these
off the host.

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=full_prod_search

Kris