kriswiner/MPU9250 #149: Without LCD code

reycreator17, 2017-05-24:
Hi. @kriswiner  Thank you so much for your work for this library, also I've read your comparison of IMU's which is really informative. Well, I'm developing navigational system for mobile robot using DWM1000 by thotro and I'd like to use MPU9250 for absolute orientation. 

Is there available code here with Quartenion filter that doesn't have the LCD code in it? It seems I find it hard to remove all those codes which are for the LCD itself.

Thanks a lot

Rey

kriswiner, 2017-05-24:
Maybe this one?

https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_BasicAHRS2_ESP8266.ino

reycreator17, 2017-05-25:
Thanks @kriswiner

BTW, I managed to remove all the display.(Something) functions. and I've changed the code 

< #define ADO 1
#if ADO
#define MPU9250_ADDRESS 0x68  // Device address when ADO = 1
#else
#define MPU9250_ADDRESS 0x68  // Device address when ADO = 0 
#define AK8963_ADDRESS 0x0C   //  Address of magnetometer
#endif  >

I don't understand what "ADO" means, but its working when MPU9250_ADDRESS is 0x68.

Overall, Thanks you so much, its working now, quite stable. I just need to calibrate it. 

Regards

Rey

closed 2017-05-25 by reycreator17