kriswiner/MPU9250 #150: WHO_AM_I returns zero

ibudnikov, 2017-05-26:
Hello, Kris!
I have a problem  -- WHO_AM_I returning 0.
I have GY-91 (pink one) and Arduino UNO R3. Connections:
VCC - 5V
Gnd - Gnd
SCL - SCL
SDA - SDA
Tried to connect 3.3V-NCS -- with no effect.
I2C scan finds 2 devices with addresses 0x68 and 0x76
Tried to launch with old library for MPU6050 -- all works (without magnetometer, obviously), WHO_AM_I returns 56. Do you have any ideas what's wrong?
P.S.: SCL and SDA levels are ~4V

monstah, 2017-07-11:
Have you tried to set the "ADO" to "0" instead of "1" at the beginning of the sketch, to switch the WHO_AM_I registers? Its above these lines and worked for me:
`#define AHRS true`
`#define SerialDebug true`

ibudnikov, 2017-07-11:
I've already solved this problem. For some reason, low-level functions like "writeRegister", "readRegister" and so on written by Kris didn't work well, so I rewrote whole sketch, and now it uses I2C libary. Now it works perfectly.