kriswiner/MPU9250 #151: Calculate angle from any starting position

tim77777, 2017-06-02:
Hi, 

is it possible to perform a calculation such that an angle from 0 - 90 degrees can be calculated from any starting position? For example the accelerometer has an real-world pitch of 45 degrees, but a software function has the pitch calculation beginning again at zero degrees. And so as pitch continues to increase to the real-world value of say 90 degrees, the software function reports 45 degrees, etc.

kriswiner, 2017-06-02:
Yes, this is trivial.

tim77777, 2017-06-02:
OK, not trivial to me. This is the code I use for my pitch and roll. What do I do?

 

pitch = atan(accele_x / sqrt(accele_y * accele_y + accele_z * accele_z)) * 180 / 3.14; 

roll = atan(accele_y / sqrt(accele_x * accele_x + accele_z * accele_z)) * 180 / 3.14;

kriswiner, 2017-06-02:
> auth/AXcUIRXe2Ez_hBRjLLCMALXSKNjTh0K7ks5sAFkagaJpZM4NupdT> .Description:

store the pitch as a variable and compare with the pitch from a later time.

tim77777, 2017-06-02:
I see. It doesn’t seem to  work well when pitch is over real-world 90 degrees though. That is to say that the adjusted pitch only gets up to about 60 degrees.

kriswiner, 2017-06-02:
Well first of all why not use real sensor fusion to estimate roll and pitch?

Secondly, what you want requires just basic arithmetic and the results will
depend on how this is implemented; for example, does pitch abruptly change
at 90 degrees to -89 degrees? This could be a problem.

tim77777, 2017-06-02:
Oh I see I think. 

 

Stored_pitch = pitch;

//Then later do this

Adj_pitch = pitch – stored_pitch.

 

But when the real-world pitch is over 90 degrees, the X acceleration begins to decrease, and the Z acceleration turns negative, and so the real world angle decreases, and so I don’t see how the stored pitch formula can work at that point.

kriswiner, 2017-06-02:
Add 90 degrees to your pitch or something to prevent negative vales

tim77777, 2017-06-02:
Basic arithmetic, yes I thought that was the case, but having worked on it for some time now, and not getting the results I expected, I began to wonder if it was possible.

Thanks for your confirmation, I’ll keep trying.

 

Fusion, I’m not capable.

 

thanks

kriswiner, 2017-06-02:
If you are using the MPU9250, the fusion is straightforward, see:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

for example.

tim77777, 2017-06-03:
Ok, I am looking at that and running into problems. I am using MPU9250BasicAHRS.ino, on an Arduino. Is that OK?

 

Compiles ok, but 0x71 is not read back from WHOAMI, and so it goes no further. I am wondering about these comments in the code

//Note: The MPU9250 is an I2C sensor and uses the Arduino Wire library. 

//Because the sensor is not 5V tolerant, we are using a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1./

//We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.

//We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.

 

Are you saying that change, has been written into MPU9250BasicAHR.ino, or are you saying that the user of this file, needs to disable the pull-ups.

 

My MPU9250 board has built in pull-ups. Where are the micro’s internal pull-ups disabled.

I changed the following in twi.c

 

  digitalWrite(SDA, 1);
  digitalWrite(SCL, 1);

 

to

 

  digitalWrite(SDA, 0);
  digitalWrite(SCL, 0);

 

but still no good.

 

What do you think?

 

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

tim77777, 2017-06-03:
Is the following possible.  Regarding using the magnetometer. I want to measure 0 – 360 degrees, but it does not need to be relative to the earth’s magnetic poles. The application is used in different environments and the time and space required to do a calibration is not viable in those environments.

So is there a formula to use whereby I could set things to zero at any time or position, and me able to measure a 360 degree rotation?

 

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

kriswiner, 2017-06-03:
> auth/AXcUIYZ_nL7wjZuRlub2OhGE2iCsQziRks5sAHtqgaJpZM4NupdT> .Description:

Why don;t you use sensor fusion, calibrate once and there you go?

tim77777, 2017-06-03:
Because, the same device is used in different environments. However, I am looking at sensor fusion now, but I have a problem with your code.

 

I am using MPU9250BasicAHRS.ino, on an Arduino. Is that OK?

 

Compiles ok, but 0x71 is not read back from WHOAMI, and so it goes no further. I am wondering about these comments in the code

//Note: The MPU9250 is an I2C sensor and uses the Arduino Wire library. 

//Because the sensor is not 5V tolerant, we are using a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1./

//We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.

//We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.

 

Are you saying the pull-up change, has been written into MPU9250BasicAHR.ino, or are you saying that the user of this file, needs to disable the pull-ups.

 

My MPU9250 board has built in pull-ups. Where are the micro’s internal pull-ups disabled.

I changed the following in twi.c

 

  digitalWrite(SDA, 1);
  digitalWrite(SCL, 1);

 

to

 

  digitalWrite(SDA, 0);
  digitalWrite(SCL, 0);

 

but still no good.

 

What do you think?

kriswiner, 2017-06-03:
> auth/AXcUIfbH-v05viw83hriqdzFTMR44dycks5sAYn3gaJpZM4NupdT> .Description:

How were you reading the sensor before?

tim77777, 2017-06-03:
I am using the Serial monitor and commented out anything starting with ‘display’.

I have a different version of your code form sparkfun

https://github.com/sparkfun/SparkFun_MPU-9250_Breakout_Arduino_Library

 

and it works ok.

 

But the latest version of yours doesn’t work for me

https://github.com/kriswiner/MPU9250 

 

 

Compiles ok, but 0x71 is not read back from WHOAMI, and so it goes no
further. I am wondering about these comments in the code

//Note: The MPU9250 is an I2C sensor and uses the Arduino Wire library.
//Because the sensor is not 5V tolerant, we are using a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1./
//We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.
//We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ to 400000L /twi.h utility file.

Are you saying the pull-up change, has been written into MPU9250BasicAHR.ino, or are you saying that the user of this file, needs to
disable the pull-ups.

kriswiner, 2017-06-03:
> issuecomment-305956054
> auth/AXcUIQagAZBsNtBCAf6O-IpvTgtJFYvFks5sAePGgaJpZM4NupdT> .Description:

Use the sparkfun version then, the useful part of the code you chose to try
is the fusion filter, which you can use with Sparkfun's code too.

tim77777, 2017-06-03:
Fair enough. thanks