kriswiner/MPU9250 #153: yaw slowly go back to one specify value after rotating

AlexHaijiaZhu, 2017-06-21:
Hi Kris,
First of all, thank you for sharing the code. It helps a lot so far.
I met this weird issues, and I tried google it but can't find any answer, I hope you can give me some idea on how to solve this problem.

System setup:
OS: windows 10
IIC lib: i2cdevlib
Sensor: SparkFun 9DoF Razor IMU M0

After calibrate the magnetometer (http://diydrones.com/profiles/blogs/advanced-hard-and-soft-iron-magnetometer-calibration-for-dummies), every thing look fine, the output is quite stable. However, when I rotate the sensor alone yaw axis, the output value did change, and then slowly go back to the original value.  
Attached is the out put:
[image]
As you can see, the blue line represents yaw, I stopped rotating (marked in red circle in the picture) but the output value is not stable after that.

[image]
As for the sensor, I place this side up.

Thanks,
Alex

kriswiner, 2017-06-21:
Well there are two things assuming you mag and accel are well calibrated.

You need to make sure you pass the madgwick function the sensor data in NED coordinates. So in your case,  ax, ay, az, gx, gy, gx, my -mx -mz. I usually also multiply the a-data by -1 but this won't affect the drift.

The other possible problem is the processor speed. What kind of fusion rate are you able to achieve with this AVR pocessor? If you use accel/gyro sample rates of 200 Hz or so, then you will need to run the fusion at > 1 kHz to get reasonable responsiveness.

AlexHaijiaZhu, 2017-06-23:
Hi Kris,

Thanks you for your response. 
I have rewrite my programming completely, and the yaw is working properly.  I didn't know what's wrong with my code, but I will figure it out later.

This is a ARM m0 processor and is very powerful compare to atmega series.  I can achieve around 450 hz fusion rate. 

In you code (MPU9250BasicAHRS.ino), you said:
> "With these settings the filter is updating at a ~145 Hz rate using the Madgwick scheme and >200 Hz using the Mahony scheme" and "This filter update rate should be fast enough to maintain accurate platform orientation for stabilization control of a fast-moving robot or quadcopter." 

But here you said: 
> If you use accel/gyro sample rates of 200 Hz or so, then you will need to run the fusion at > 1 kHz to get reasonable responsiveness.
Can please explain more about that?

Thanks,
Alex

kriswiner, 2017-06-23:
You really 5-10x the sample rate for accurate orientation estimation

Patrickyp, 2018-05-11:
can you attach code you used to fix yaw alex?

Patrickyp, 2018-05-21:
HI kRIS,

Is the fusion rate the number that is printing out in debug mode as "rate = 1301.90 hz"?  When I turn the magnetometer 90 degrees it takes 3-5 seconds for the serial monitor to get to the correct value is this delay normal for 1300 hz rate?

kriswiner, 2018-05-21:
If you are running the sensors at ~200 Hz using the Madgwick algorithm then
yes, this seems about normal to me.

PhoenixZh, 2019-01-11:
> Hi Kris,
>
> Thanks you for your response.
> I have rewrite my programming completely, and the yaw is working properly. I didn't know what's wrong with my code, but I will figure it out later.
>
> This is a [...]

hi, I got the same problem now, can you please tell me how you solved your problem?