kriswiner/MPU9250 #154: Are the orientation angles relative to earth frame or initial IMU frame?

xiaoxuesu14, 2017-06-26:
Hi Kris,

first of all, thank you so much for this amazing library! 

I have a question about the Euler angles: are they relative to the earth frame (gravity and north, absolute) or to the IMU's initial frame (relative)? Using your code the roll and pitch converges to zero at the initial stage, no matter what orientation the IMU is. So I assume it is relative to the IMU frame? However, the matlab implementation of Madgwick gives absolute orientation to the earth frame. I cannot tell a difference between the two implementation thought. Could you enlighten me how this happens?

Thanks
Xuesu

kriswiner, 2017-06-26:
The orientation is relative to Earth frame. How are you calibrating your
sensors?

xiaoxuesu14, 2017-06-26:
I only calibrated the magnetometer, nothing else. The roll and pitch converges to 0 no matter how the IMU configuration is. After roll and pitch settle down, if I rotate the IMU, the euler angles make sense, but it's just relative to the initial IMU configuration, not the earth frame.

kriswiner, 2017-06-26:
Strange, not sure why. Not this way for me.

xiaoxuesu14, 2017-06-26:
Orientation: 177.17 52.85 46.14
Orientation: 176.56 54.16 46.35
Orientation: 175.85 55.53 46.52
Orientation: 175.05 56.98 46.65
Orientation: 174.22 58.48 46.74
Orientation: 173.25 60.04 46.75
Orientation: 172.03 61.66 46.59
Orientation: 170.70 63.40 46.37
Orientation: 169.07 65.18 45.91
Orientation: 167.14 67.03 45.19
Orientation: 164.71 68.98 44.07
Orientation: 161.77 70.98 42.50
Orientation: 157.99 73.02 40.15
Orientation: 153.14 75.06 36.78
Orientation: 146.45 77.13 31.65
Orientation: 136.98 79.08 23.79
Orientation: 123.22 80.74 11.70
Orientation: 104.18 81.75 -5.57
Orientation: 81.51 81.74 -26.39
Orientation: 60.81 80.49 -45.18
Orientation: 45.21 78.25 -58.79
Orientation: 34.42 75.35 -67.50
Orientation: 26.69 71.95 -73.10
Orientation: 20.92 68.18 -76.67
Orientation: 16.41 64.02 -78.91
Orientation: 12.82 59.52 -80.17
Orientation: 9.80 54.70 -80.80
Orientation: 7.30 49.46 -80.89
Orientation: 5.01 43.91 -80.67
Orientation: 2.95 37.91 -80.17
Orientation: 1.14 31.51 -79.44
Orientation: -0.31 24.67 -78.43
Orientation: -1.37 17.33 -77.19
Orientation: -1.88 9.59 -75.87
Orientation: -1.09 1.77 -74.70
Orientation: -0.08 -0.24 -74.27
Orientation: -0.08 -0.53 -73.96
Orientation: 0.01 -0.39 -73.93
Orientation: -0.02 -0.48 -73.53
Orientation: 0.07 -0.27 -73.66
Orientation: -0.07 -0.39 -73.81
Orientation: -0.05 -0.33 -73.66
Orientation: 0.03 -0.24 -73.68
Orientation: -0.06 -0.39 -73.98
Orientation: -0.05 -0.51 -73.77
Orientation: 0.07 -0.46 -74.10
Orientation: -0.02 -0.19 -74.28
Orientation: -0.07 -0.26 -74.60
Orientation: -0.09 -0.22 -74.38
Orientation: -0.09 -0.46 -74.41
Orientation: 0.04 -0.35 -74.38
Orientation: -0.09 -0.50 -74.46
Orientation: -0.05 -0.38 -74.47
Orientation: -0.05 -0.43 -74.51
Orientation: 0.07 -0.09 -74.74
Orientation: -0.04 -0.57 -74.91
Orientation: -0.09 -0.44 -74.71

This looks like the output no matter how I put the IMU. Roll and pitch converges to zero anyways...

kriswiner, 2017-06-26:
Looks like an error in the sensor data to me.

markussen145, 2017-07-14:
When you say roll and pitch converge, do you mean they change as you move the mpu but once you keep it steady, they decrease to zero? If that's the case, it may be cause by inserting raw data in madgwick/Mahoney functions with the wrong signs.

Check the data sheet of your device and see their oriention (check where x,y and z are positive).

xiaoxuesu14, 2017-07-14:
Hi,

no, that was not what I meant. I meant, at the initial stage after I
started the algorithm, I kept the IMU steady, and then the roll and pitch
converges to zero. It really looks like an initialization stage. After
that, if I rotate the IMU, the three Euler angles all make sense. So the
algorithm output is with respect to the initial configuration of the IMU
(how I started with the IMU, since it initializes roll and pitch to zero),
not with respect to the earth frame (gravity and north).

Thanks
Xuesu

superviruz, 2017-10-03:
Hi Xuesu,

I facing the same issue as you. When I laid IMU (9250) in any orientation then power it up, it will start roll and pitch to nearly zero (vary from -10 to 10).  But after that, I rotate to another orientation and it send data correctly (relative to initial orientation).

Below are "some portion" of my mod code (download from sparkfun that modified from original Kris code). 

I have to shorten code as most as I can to fit in Arduino mini pro (for minimize cost and size)

I know that it may be too short and make not stable as original code.

#include "quaternionFilters.h"
#include "MPU9250.h"

#include <Arduino.h>
#include <Wire.h>

 
#define AHRS true         // Set to false for basic data read
#define SerialDebug false  // Set to true to get Serial output for debugging

int intPin = 3;
MPU9250 myIMU;

void setup()
{
  
  Wire.begin();
  // TWBR = 12;  // 400 kbit/sec I2C speed
  if (SerialDebug) 
  {
  Serial.begin(38400);
  }
  // Set up the interrupt pin, its set as active high, push-pull
  pinMode(intPin, INPUT);
  digitalWrite(intPin, LOW);

  // Read the WHO_AM_I register, this is a good test of communication
  byte c = myIMU.readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250);
  Serial.print("MPU9250 "); Serial.print("I AM "); Serial.print(c, HEX);
  Serial.print(" I should be "); Serial.println(0x71, HEX);

  {
    Serial.println("MPU9250 is online...");

    // Start by performing self test and reporting values
    myIMU.MPU9250SelfTest(myIMU.SelfTest);
    Serial.print("x-axis self test: acceleration trim within : ");
    Serial.print(myIMU.SelfTest[0],1); Serial.println("% of factory value");
    Serial.print("y-axis self test: acceleration trim within : ");
    Serial.print(myIMU.SelfTest[1],1); Serial.println("% of factory value");
    Serial.print("z-axis self test: acceleration trim within : ");
    Serial.print(myIMU.SelfTest[2],1); Serial.println("% of factory value");
    Serial.print("x-axis self test: gyration trim within : ");
    Serial.print(myIMU.SelfTest[3],1); Serial.println("% of factory value");
    Serial.print("y-axis self test: gyration trim within : ");
    Serial.print(myIMU.SelfTest[4],1); Serial.println("% of factory value");
    Serial.print("z-axis self test: gyration trim within : ");
    Serial.print(myIMU.SelfTest[5],1); Serial.println("% of factory value");

    // Calibrate gyro and accelerometers, load biases in bias registers
    myIMU.calibrateMPU9250(myIMU.gyroBias, myIMU.accelBias);

    myIMU.initMPU9250();
    // Initialize device for active mode read of acclerometer, gyroscope, and
    // temperature
    Serial.println("MPU9250 initialized for active data mode....");

    // Read the WHO_AM_I register of the magnetometer, this is a good test of
    // communication
    byte d = myIMU.readByte(AK8963_ADDRESS, WHO_AM_I_AK8963);
[... 196 lines cut ...]
        Serial.print(", ");
        Serial.println(myIMU.roll, 2);

        Serial.print("rate = ");
        Serial.print((float)myIMU.sumCount/myIMU.sum, 2);
        Serial.println(" Hz");
        
        Serial.print("count = ");
        Serial.print(count);
        Serial.println(" times");
      }
     
      myIMU.count = millis();
      myIMU.sumCount = 0;
      myIMU.sum = 0;
    } // if (myIMU.delt_t > 100)
  } // if (AHRS)
  //Take temp out off AHRS
  myIMU.tempCount = myIMU.readTempData();  // Read the adc values
  // Temperature in degrees Centigrade (-6000,-3),(4500,29)
  // http://www.webmath.com/equline1.html
  myIMU.temperature = ((float) myIMU.tempCount)*8/2625+107/7;
  // Print temperature in degrees Centigrade
  temp = myIMU.temperature;
}

kriswiner, 2017-10-03:
> quaternions_and_Euler_angles

I don;t think this is right:

// MadgwickQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz
*PI/180.0f, my, mx, mz); MahonyQuaternionUpdate(myIMU.ax, myIMU.ay,
myIMU.az, myIMU.gx*DEG_TO_RAD,
myIMU.gy*DEG_TO_RAD, myIMU.gz*DEG_TO_RAD, myIMU.my,
myIMU.mx, myIMU.mz, myIMU.deltat);

For one thing, Az is opposite to Mz.

If you want to make Ax == North, then feed in the data like this:

Ax, -Ay, -Az, Gx, -Gy, -Gz, My, -Mx, Mz

then all sensor axes are NED configuration. I ususally negate the A values,
but this is not necessary for consistent results.