kriswiner/MPU9250 #156: accelerometer cross axis interference

sageblue, 2017-06-30:
Hi, 
I am hoping to use MPU9250 for dead reckoning type of application. In my test accelerometer showed quite large cross axis interference in its raw measurement (hopefully this is the right terminology more details to come on this).  

Come cross your project, wonder if you have tried this type of test and would appreciate if you can comment whether you made similar observation on MPU9250's accelerometer performance. Want to find out if this is MPU9250's  limitation or my configuration, use issue. 

More detail on my observation:

Basically, I moved device along y axis (of accelerometer of MPU9250), once along +Y, stop and rest a bit, then along -Y, think I did a pretty good job align the movement with axis, since I pushed device along a guiding rail that is parallel to the board. Then I read the raw accelerometer reading from DMP FIFO, and they are print below,  readings are converted to mg, and formatted as := sample index, reading on x, y, z axis 
Note reading around sample 74-100 and 300-320.
The accelerometer can allow idle and motion to be distinguished, but  I would expect most acceleration during motion occurs on Y axis, but the portion on X is large enough to compare with Y axis portion.

acce(mg)=	1	-2.441406	-1.953125	-1008.300781
acce(mg)=	2	-8.789062	4.882812	-999.023438
acce(mg)=	3	6.347656	4.882812	-986.328125
acce(mg)=	4	4.882812	0.976562	-1012.207031
acce(mg)=	5	0.976562	-2.929688	-991.210938
acce(mg)=	6	9.277344	0.488281	-1000.976562
acce(mg)=	7	2.929688	0.976562	-1006.835938
acce(mg)=	8	-0.488281	7.324219	-1008.300781
acce(mg)=	9	2.441406	10.253906	-1007.8125
acce(mg)=	10	-5.859375	3.90625	-996.582031
acce(mg)=	11	-0.976562	7.324219	-1020.996094
acce(mg)=	12	6.347656	0	-1002.441406
acce(mg)=	13	6.347656	-0.488281	-999.511719
acce(mg)=	14	-5.859375	-4.394531	-1000.488281
acce(mg)=	15	0.976562	3.417969	-999.511719
acce(mg)=	16	2.929688	-2.441406	-997.558594
acce(mg)=	17	-2.929688	0	-1007.324219
acce(mg)=	18	7.8125	1.953125	-1012.695312
acce(mg)=	19	-0.488281	6.347656	-1006.835938
acce(mg)=	20	9.765625	2.929688	-1000.976562
acce(mg)=	21	2.929688	0.976562	-1007.324219
acce(mg)=	22	-0.488281	1.953125	-1000.976562
acce(mg)=	23	-7.8125	8.789062	-1019.042969
acce(mg)=	24	-4.394531	1.953125	-1023.4375
acce(mg)=	25	-2.929688	-3.417969	-996.09375
acce(mg)=	26	0	8.789062	-1003.417969
acce(mg)=	27	1.953125	-6.347656	-1008.300781
acce(mg)=	28	4.882812	-1.464844	-998.535156
acce(mg)=	29	4.882812	-4.394531	-996.09375
acce(mg)=	30	-5.859375	4.882812	-1000
acce(mg)=	31	-2.929688	5.371094	-986.816406
acce(mg)=	32	9.277344	-0.488281	-995.605469
acce(mg)=	33	5.371094	-2.441406	-1000.976562
acce(mg)=	34	1.464844	-0.976562	-988.28125
acce(mg)=	35	2.441406	3.90625	-1010.253906
acce(mg)=	36	5.859375	0.488281	-983.886719
acce(mg)=	37	6.347656	-0.976562	-989.257812
acce(mg)=	38	5.859375	3.90625	-1003.417969
acce(mg)=	39	8.300781	-7.8125	-1008.789062
acce(mg)=	40	7.324219	-2.441406	-999.023438
acce(mg)=	41	-4.394531	2.929688	-998.535156
acce(mg)=	42	0.976562	2.441406	-1009.277344
acce(mg)=	43	5.859375	-6.835938	-996.09375
acce(mg)=	44	-1.464844	3.417969	-1008.789062
acce(mg)=	45	-3.417969	6.347656	-1000.488281
acce(mg)=	46	-9.277344	0.976562	-999.511719
acce(mg)=	47	6.347656	1.953125	-993.652344
acce(mg)=	48	1.464844	7.8125	-1002.441406
acce(mg)=	49	5.859375	2.929688	-998.535156
acce(mg)=	50	6.347656	-7.324219	-996.09375
acce(mg)=	51	-10.253906	-2.441406	-1000
acce(mg)=	52	0.488281	-5.371094	-998.046875
acce(mg)=	53	2.929688	8.300781	-999.511719
acce(mg)=	54	2.929688	0.976562	-994.628906
acce(mg)=	55	1.464844	3.417969	-996.09375
acce(mg)=	56	-7.324219	-2.929688	-998.046875
acce(mg)=	57	-2.929688	0.976562	-1014.160156
acce(mg)=	58	0	2.441406	-1015.136719
acce(mg)=	59	1.953125	-6.347656	-996.09375
[... 299 lines cut ...]
acce(mg)=	359	0.488281	3.417969	-995.117188
acce(mg)=	360	-6.347656	0.976562	-993.652344
acce(mg)=	361	2.929688	4.394531	-1007.8125
acce(mg)=	362	2.929688	1.953125	-1000.488281
acce(mg)=	363	7.8125	-0.488281	-993.652344
acce(mg)=	364	1.953125	3.417969	-999.023438
acce(mg)=	365	3.90625	-0.976562	-991.210938
acce(mg)=	366	0.976562	-5.371094	-1013.183594
acce(mg)=	367	7.324219	7.324219	-1007.324219
acce(mg)=	368	5.371094	-5.371094	-1010.742188
acce(mg)=	369	4.882812	3.90625	-992.1875
acce(mg)=	370	-0.488281	0.976562	-1001.953125
acce(mg)=	371	5.859375	7.324219	-991.699219
acce(mg)=	372	8.300781	-6.347656	-998.046875
acce(mg)=	373	0.488281	0	-989.746094
acce(mg)=	374	0.488281	1.953125	-1005.859375
acce(mg)=	375	1.953125	9.277344	-1000.488281
acce(mg)=	376	-6.347656	-4.882812	-997.558594
acce(mg)=	377	0.488281	11.71875	-1001.953125
acce(mg)=	378	-1.464844	8.300781	-1002.441406
acce(mg)=	379	-5.859375	-7.8125	-1000.488281
acce(mg)=	380	-4.882812	2.441406	-998.535156
acce(mg)=	381	7.324219	3.90625	-1002.441406
acce(mg)=	382	-3.417969	-0.488281	-999.511719
acce(mg)=	383	18.554688	-1.953125	-1003.90625

kriswiner, 2017-06-30:
I wouldn't use the DMP, just use the FIFO if you need to store the data or
read the data continuously to the serial monitor.

The accel performance will depend on the sample rate, full scale, low pass
filter, etc. The data as you have presented it is impossible to diagnose.
Yes, it seems to be strange as you have described it, why? Who knows...

For slow motion, try 100 Hz sample rate, 2g full scale and 40 Hz bandwidth
or variations to check dependencies.

Did you calibrate the offset bias?

Lot's of things could be wrong...

sageblue, 2017-07-03:
Thanks a lot to your reply. Yes, offset is calibrated, otherwise reading while device is stationary would have larger deviation, (50+ mg ).

Is there any known issue with DMP ?

I am leveraging DMP for orientation estimation, which looks pretty good so far.

From FIFO I read out accelerometer, gyro reading plus quaternions at 40Hz , while register samples rate is set at 200Hz, gyro range set at 250 d/s, accelerometer at 2g, filter at 92 Hz.

Can try what you suggested, aside from not loading DMP firmware and described register setting, what is the 40Hz bandwidth referring to ? if sample rate is set at 100 Hz, does it mean I should get two or three sets of accelerometer value each time reading form FIFO  ?

Thanks.

kriswiner, 2017-07-03:
Not sure what to tell you here. If you want to use the DMP then you will
get what you get and there's not much I can do to help you.

The FIFO can hold 512 bytes so you can do the math about how many accel
samples can be held (hint: it is > 80).

Dead reckoning is not possible using the MPU9250 DMP just because it does
not include the magnetometer. In other words, it is a 6 DoF orientation
solution so will have pretty mean (~1 degree per minute) heading drift.
Nothing can be done about that except move to a 9 DoF solution which the
DMP cannot do.

https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion

Even with an accurate 9 DoF absolute orientation solution dead reckoning
accuracy can only be maintained for a very short time just due to the error
accumulation from the double integration of the linear acceleration.

For the best absolute orienation solution, you might consider this device:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/

Sorry I can't be of more help.

Kris

sageblue, 2017-07-03:
Thanks again
Just a quick update on the experiment without DMP.  100Hz sample rate, 2g accelerometer range, lpf 41Hz, I am also reading FIFO at 100Hz.  The results are not different, motion on Y axis result to reading on both X, Y  Does not seem to be related to DMP, I tend to believe this is the performance of MPU9250, or do you find it different  on your experience with the device ?

acce(mg)=	1	-4.882812	8.300781	-1006.835938
acce(mg)=	2	-14.648438	9.765625	-1015.136719
acce(mg)=	3	-3.417969	8.300781	-1004.882812
acce(mg)=	4	-7.324219	0.976562	-1025.878906
acce(mg)=	5	-6.835938	1.464844	-1005.371094
acce(mg)=	6	4.882812	11.230469	-1012.207031
acce(mg)=	7	-0.976562	11.230469	-992.675781
acce(mg)=	8	-2.441406	18.554688	-1007.324219
acce(mg)=	9	-2.441406	7.324219	-1012.207031
acce(mg)=	10	0.976562	10.742188	-1012.207031
acce(mg)=	11	-11.230469	5.859375	-1006.347656
acce(mg)=	12	-0.488281	1.464844	-1003.417969
acce(mg)=	13	-6.347656	2.929688	-995.605469
acce(mg)=	14	-10.253906	0.976562	-1008.300781
acce(mg)=	15	-9.765625	5.371094	-1000
acce(mg)=	16	-8.300781	4.882812	-1010.253906
acce(mg)=	17	-3.417969	6.835938	-1004.882812
acce(mg)=	18	0.976562	1.464844	-997.558594
acce(mg)=	19	-8.300781	11.71875	-1030.273438
acce(mg)=	20	-7.324219	13.183594	-1020.507812
acce(mg)=	21	2.929688	1.953125	-1015.136719
acce(mg)=	22	-2.929688	9.765625	-984.375
acce(mg)=	23	-15.136719	2.441406	-1001.464844
acce(mg)=	24	-1.464844	2.929688	-1009.765625
acce(mg)=	25	-0.976562	6.835938	-981.445312
acce(mg)=	26	-2.441406	16.113281	-1013.671875
acce(mg)=	27	-0.976562	6.347656	-1015.136719
acce(mg)=	28	-9.765625	10.742188	-1010.253906
acce(mg)=	29	-2.441406	4.394531	-1010.253906
acce(mg)=	30	-5.371094	7.324219	-998.046875
acce(mg)=	31	0.976562	5.859375	-999.023438
acce(mg)=	32	-9.765625	14.160156	-1005.859375
acce(mg)=	33	-6.835938	-5.371094	-995.117188
acce(mg)=	34	-8.300781	-4.394531	-1003.90625
acce(mg)=	35	-2.929688	-0.488281	-1011.230469
acce(mg)=	36	-4.394531	9.765625	-1003.417969
acce(mg)=	37	0	8.300781	-1005.371094
acce(mg)=	38	-0.976562	2.929688	-990.234375
acce(mg)=	39	2.441406	0.976562	-1015.625
acce(mg)=	40	-6.835938	4.882812	-1002.441406
acce(mg)=	41	5.859375	1.953125	-1005.371094
acce(mg)=	42	-2.441406	2.441406	-1016.113281
acce(mg)=	43	4.882812	2.929688	-999.511719
acce(mg)=	44	-1.953125	5.859375	-1007.8125
acce(mg)=	45	2.441406	6.835938	-1004.882812
acce(mg)=	46	-0.488281	10.253906	-997.070312
acce(mg)=	47	-13.183594	6.835938	-1000.976562
acce(mg)=	48	5.371094	2.441406	-1006.347656
acce(mg)=	49	-1.464844	7.8125	-1012.207031
acce(mg)=	50	5.859375	4.394531	-1003.417969
acce(mg)=	51	-6.347656	13.671875	-993.164062
acce(mg)=	52	-9.277344	7.8125	-1005.371094
acce(mg)=	53	-7.324219	1.464844	-995.117188
acce(mg)=	54	-2.441406	-1.464844	-1022.460938
acce(mg)=	55	-4.394531	-0.976562	-999.511719
acce(mg)=	56	-11.71875	5.371094	-1015.625
acce(mg)=	57	-3.90625	-0.488281	-1014.160156
acce(mg)=	58	-9.277344	10.253906	-1004.882812
acce(mg)=	59	-8.789062	1.464844	-1007.324219
acce(mg)=	60	-9.277344	2.929688	-994.140625
acce(mg)=	61	-0.976562	6.347656	-1011.71875
acce(mg)=	62	-0.976562	5.371094	-1002.929688
acce(mg)=	63	-6.835938	4.882812	-1010.742188
acce(mg)=	64	0	9.765625	-1001.953125
acce(mg)=	65	-0.976562	8.789062	-991.210938
acce(mg)=	66	-8.300781	2.929688	-1009.765625
acce(mg)=	67	5.371094	22.949219	-1005.371094
[... 1106 lines cut ...]
acce(mg)=	1174	0	18.066406	-998.535156
acce(mg)=	1175	3.417969	0	-1001.464844
acce(mg)=	1176	-6.835938	6.347656	-1015.136719
acce(mg)=	1177	-4.882812	9.765625	-1016.601562
acce(mg)=	1178	-6.835938	6.835938	-1009.765625
acce(mg)=	1179	-4.394531	7.8125	-1007.8125
acce(mg)=	1180	-3.417969	12.207031	-1011.230469
acce(mg)=	1181	-13.671875	2.929688	-1002.441406
acce(mg)=	1182	-11.230469	4.394531	-994.628906
acce(mg)=	1183	-1.953125	12.695312	-1006.347656
acce(mg)=	1184	-6.347656	5.371094	-1010.253906
acce(mg)=	1185	-9.277344	7.8125	-1009.765625
acce(mg)=	1186	-4.394531	3.90625	-1009.765625
acce(mg)=	1187	-14.648438	9.765625	-1004.882812
acce(mg)=	1188	2.441406	-2.929688	-1015.136719
acce(mg)=	1189	-2.441406	7.324219	-1009.765625
acce(mg)=	1190	-2.929688	7.324219	-1002.441406
acce(mg)=	1191	3.90625	-0.488281	-1021.484375
acce(mg)=	1192	-11.230469	1.464844	-1011.230469
acce(mg)=	1193	-1.953125	-3.90625	-999.511719
acce(mg)=	1194	-7.324219	3.90625	-995.605469
acce(mg)=	1195	-2.929688	-1.464844	-1018.066406
acce(mg)=	1196	-7.324219	11.230469	-1000
acce(mg)=	1197	-3.417969	2.441406	-1001.464844
acce(mg)=	1198	-9.277344	2.441406	-1000

kriswiner, 2017-07-03:
This could be your MPU9250, but I don;t see this on mine. It could also be
the setup you are using to test, don't know.