kriswiner/MPU9250 #157: WhoAmI showing 0x58

PratikSK, 2017-07-06:
Hello Sir,

I have used the library your MPU9250.h in a custom board designed using nRF52832. I am getting the WhoAmI value as 0x58 instead of 0x48 as mentioned in the comment. 
Can you please suggest what kind of issue this can be? I am not able to understand the significance of this value.

Thanks in advance,
Pratik

kriswiner, 2017-07-06:
This seems odd since the WHO_AM_I of the MPU9250 should be 0x71, while the WHO_AM_I of the MPU9255 should be 0x73. I am not sure what either 0x48 or 0x58 means. Are you sure the I2C bus is working? If you do an I2C scan, which devices show up on the I2C bus. I have never seen this problem on my own [custom nRF52+MPU9250 board](https://github.com/kriswiner/nRF52832DevBoard).

PratikSK, 2017-07-15:
thank you for your response. I went through your repository on mbed and have imported the same. However, it is not working for me. I think the issue is related to i2c configuration. 
I have connected OLED display to the same i2c of nrf52832. So in this case, is there any special configuration is required for using the same i2c for mpu9250?
Thanks for your help.

kriswiner, 2017-07-15:
The I2C bus cannot be connected to another device, especially a display.