kriswiner/MPU9250 #158: gyro calibration and bias

abstruse123, 2017-07-07:
Hi Kris!

I ran into your code and observed that the gyro bias values are not subtracted from the current gyro readings. On the other hand, accel bias values are subtracted. Is this done deliberately or a mistake?

`

1
--
// Now we'll calculate the accleration value into actual g's
  | ax = (float)MPU9250Data[0]*aRes - accelBias[0];  // get actual g value, this depends on scale being set
  | ay = (float)MPU9250Data[1]*aRes - accelBias[1];
  | az = (float)MPU9250Data[2]*aRes - accelBias[2];
  |  
  | //   readGyroData(gyroCount);  // Read the x/y/z adc values
  |  
  | // Calculate the gyro value into actual degrees per second
  | gx = (float)MPU9250Data[4]*gRes;  // get actual gyro value, this depends on scale being set
  | gy = (float)MPU9250Data[5]*gRes;
  | gz = (float)MPU9250Data[6]*gRes;

`

kriswiner, 2017-07-07:
Gyro offset bias is stored on the gyro offset bias registers, the accel
offset bias registers could also be used but the temperature compensation
on the LSB makes the results unreliable, so I subtract the acccel bias
manually.

abstruse123, 2017-07-07:
Yes. The gyro offset bias values are written to the respective registers. But, the gyro bias values are not subtracted from the gyro readings. Once written to the offset registers, do they get automatically subtracted from the sensor data registers?

kriswiner, 2017-07-07:
yes, check the data sheet.

abstruse123, 2017-07-07:
OK. Thanks for the clarification.
If that is the case, then portion of the code can be commented as the accel bias registers are not written to.
`
1
--
// Construct the accelerometer biases for push to the hardware accelerometer bias registers. These registers contain
  | // factory trim values which must be added to the calculated accelerometer biases; on boot up these registers will hold
  | // non-zero values. In addition, bit 0 of the lower byte must be preserved since it is used for temperature
  | // compensation calculations. Accelerometer bias registers expect bias input as 2048 LSB per g, so that
  | // the accelerometer biases calculated above must be divided by 8.
  |  
  | int32_t accel_bias_reg[3] = {0, 0, 0}; // A place to hold the factory accelerometer trim biases
  | readBytes(MPU9250_ADDRESS, XA_OFFSET_H, 2, &data[0]); // Read factory accelerometer trim values
  | accel_bias_reg[0] = (int32_t) (((int16_t)data[0] << 8) | data[1]);
  | readBytes(MPU9250_ADDRESS, YA_OFFSET_H, 2, &data[0]);
  | accel_bias_reg[1] = (int32_t) (((int16_t)data[0] << 8) | data[1]);
  | readBytes(MPU9250_ADDRESS, ZA_OFFSET_H, 2, &data[0]);
  | accel_bias_reg[2] = (int32_t) (((int16_t)data[0] << 8) | data[1]);
  |  
  | uint32_t mask = 1uL; // Define mask for temperature compensation bit 0 of lower byte of accelerometer bias registers
  | uint8_t mask_bit[3] = {0, 0, 0}; // Define array to hold mask bit for each accelerometer bias axis
  |  
  | for(ii = 0; ii < 3; ii++) {
  | if((accel_bias_reg[ii] & mask)) mask_bit[ii] = 0x01; // If temperature compensation bit is set, record that fact in mask_bit
  | }
  |  
  | // Construct total accelerometer bias, including calculated average accelerometer bias from above
  | accel_bias_reg[0] -= (accel_bias[0]/8); // Subtract calculated averaged accelerometer bias scaled to 2048 LSB/g (16 g full scale)
  | accel_bias_reg[1] -= (accel_bias[1]/8);
  | accel_bias_reg[2] -= (accel_bias[2]/8);
  |  
  | data[0] = (accel_bias_reg[0] >> 8) & 0xFF;
  | data[1] = (accel_bias_reg[0])      & 0xFF;
  | data[1] = data[1] | mask_bit[0]; // preserve temperature compensation bit when writing back to accelerometer bias registers
  | data[2] = (accel_bias_reg[1] >> 8) & 0xFF;
  | data[3] = (accel_bias_reg[1])      & 0xFF;
  | data[3] = data[3] | mask_bit[1]; // preserve temperature compensation bit when writing back to accelerometer bias registers
  | data[4] = (accel_bias_reg[2] >> 8) & 0xFF;
  | data[5] = (accel_bias_reg[2])      & 0xFF;
  | data[5] = data[5] | mask_bit[2]; // preserve temperature compensation bit when writing back to accelerometer bias registers
  |  
  | // Apparently this is not working for the acceleration biases in the MPU-9250
  | // Are we handling the temperature correction bit properly?
  | // Push accelerometer biases to hardware registers
  | /*  writeByte(MPU9250_ADDRESS, XA_OFFSET_H, data[0]);
  | writeByte(MPU9250_ADDRESS, XA_OFFSET_L, data[1]);
  | writeByte(MPU9250_ADDRESS, YA_OFFSET_H, data[2]);
  | writeByte(MPU9250_ADDRESS, YA_OFFSET_L, data[3]);
  | writeByte(MPU9250_ADDRESS, ZA_OFFSET_H, data[4]);
  | writeByte(MPU9250_ADDRESS, ZA_OFFSET_L, data[5]);
  | */

`
As per the MPU-9250 register map & description document, it is mentioned in the offset register contents are subtracted from the gyro sensor values before they are written to the sensor registers. 
But in your code, the gyro bias values are made negative. Please clarify.
`

1
--
// Construct the gyro biases for push to the hardware gyro bias registers, which are reset to zero upon device startup
  | data[0] = (-gyro_bias[0]/4  >> 8) & 0xFF; // Divide by 4 to get 32.9 LSB per deg/s to conform to expected bias input format
  | data[1] = (-gyro_bias[0]/4)       & 0xFF; // Biases are additive, so change sign on calculated average gyro biases
  | data[2] = (-gyro_bias[1]/4  >> 8) & 0xFF;
  | data[3] = (-gyro_bias[1]/4)       & 0xFF;
  | data[4] = (-gyro_bias[2]/4  >> 8) & 0xFF;
  | data[5] = (-gyro_bias[2]/4)       & 0xFF;

`

Also, is the same not applicable to accel bias registers? There is no note written near the accel offset registers description. Request you to kindly clarify as I am a bit confused.

As per your code comments, why it is that the accel offset/bias registers are non-zero at startup whereas the gyro offset/bias registers are zero at startup?

kriswiner, 2017-07-07:
Try it both ways

abstruse123, 2017-07-07:
Can you please elaborate.. I couldn't understand what you meant

kriswiner, 2017-07-07:
Try storing the accel offset bias in the offset bias register and compare
with subtracting in the main code.