kriswiner/MPU9250 #159: Sync of two identical MPU-9250

Tobertet, 2017-07-10:
Hello Kris, first of all congratulations on all your work, it is simply amazing.

I am trying to synchronize two MPU-9250 in order that at any time I can know what is the difference in angles between two different objects with these units attached. However, I can't achieve this. I tried all your sketches modified, but I always get like random values of yaw pitch and roll.

Theoretically, I have to:
     Calibrate accelerometer and gyroscope (accelgyrocalMPU9250)
     Calibrate magnetometer (magcalMPU9250)
     Read raw data from the accelerometer, gyroscope and magnetometer.
     Call the Magdwick algorithm with the raw data.
     
However, the pitch, roll and yaw I get from the Madgwick is far too slow to be estable and the angles are incorrect. Could I be failing at calibrating the sensors?

Thank you very much

kriswiner, 2017-07-10:
Three possibilities if I assum your MPU9250 is properly functioning:

1) insufficient calibration
2) wrong order of sensor data into Madgwick filter
3) wrong sample rate/bandwidth and/or too slow fusion rate.

For 3), what MCU are you using and what fusion rates can you achieve with
it?

markussen145, 2017-07-14:
I'm trying the same in the next couple of weeks. I'll keep an eye on this thread and I'll bring any outcome I can get.

In the meantime, Tobertet, did you try to get attitude from both MPU's individually first? The output, was it right?